Mange/rtl8192eu-linux-driver #3: IP: [<f8f86d70>] rtw_ndev_notifier_call+0x13/0x5d [8192eu]

waldyd, 2016-03-01:
Hello,

After building the driver, the following was prompted:

```bash
[  258.028254] usb 5-7: new high-speed USB device number 3 using ehci-pci
[  258.161671] usb 5-7: New USB device found, idVendor=0bda, idProduct=818b
[  258.161691] usb 5-7: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  258.161704] usb 5-7: Product: 802.11n NIC
[  258.161715] usb 5-7: Manufacturer: Realtek
[  258.161726] usb 5-7: SerialNumber: 40A5EF07E4B4
[  259.323530] RTL871X: module init start
[  259.323541] RTL871X: rtl8192eu v4.3.1.1_11320.20140505
[  259.323547] RTL871X: build time: Jul  3 2014 14:13:29
[  259.323636] BUG: unable to handle kernel NULL pointer dereference at 00000028
[  259.323893] IP: [<f8f86d70>] rtw_ndev_notifier_call+0x13/0x5d [8192eu]
[  259.324018] *pdpt = 0000000032e6e001 *pde = 0000000000000000 
[  259.324286] Oops: 0000 [#1] SMP 
[  259.324286] Modules linked in: 8192eu(O+) fuse ctr ccm binfmt_misc nfnetlink_queue nfnetlink_log nfnetlink bluetooth sha256_generic cbc joydev arc4 uvcvideo b43 videobuf2_vmalloc videobuf2_memops videobuf2_core v4l2_common bcma videodev mac80211 media iTCO_wdt iTCO_vendor_support hp_wmi cfg80211 sparse_keymap rfkill i915 coretemp lpc_ich snd_hda_codec_idt mfd_core psmouse snd_hda_codec_generic evdev pcspkr serio_raw snd_hda_intel snd_hda_controller rng_core snd_hda_codec drm_kms_helper snd_hwdep drm wmi snd_pcm snd_timer snd i2c_algo_bit battery soundcore ac shpchp video button acpi_cpufreq processor autofs4 ext4 crc16 mbcache jbd2 xts gf128mul algif_skcipher af_alg dm_crypt dm_mod sg sd_mod ata_generic ata_piix libata scsi_mod uhci_hcd ehci_pci ehci_hcd ssb usbcore atl1c usb_common mmc_core pcmcia
[  259.325995]  pcmcia_core thermal thermal_sys
[  259.325995] CPU: 1 PID: 1478 Comm: systemd-udevd Tainted: G           O    4.0.0-kali1-686-pae #1 Debian 4.0.4-1+kali2
[  259.325995] Hardware name: Hewlett-Packard HP Mini 110-1000/308F, BIOS 308F0 Ver. F.16 09/02/2009
[  259.325995] task: f364f2a0 ti: f67c4000 task.ti: f67c4000
[  259.325995] EIP: 0060:[<f8f86d70>] EFLAGS: 00010282 CPU: 1
[  259.325995] EIP is at rtw_ndev_notifier_call+0x13/0x5d [8192eu]
[  259.325995] EAX: 00000000 EBX: f8fe9a00 ECX: f67c5dc4 EDX: 00000005
[  259.325995] ESI: c16c4390 EDI: f5631800 EBP: f67c5db0 ESP: f67c5d98
[  259.325995]  DS: 007b ES: 007b FS: 00d8 GS: 00e0 SS: 0068
[  259.325995] CR0: 8005003b CR2: 00000028 CR3: 35e60000 CR4: 000007f0
[  259.325995] Stack:
[  259.325995]  f67c5db8 f3428393 f6786be8 f67c5db0 f8fe9a00 c16c4390 f67c5dd4 c13e64df
[  259.325995]  c11e389a c16c4300 00000000 f5631800 f4dc5c20 c165e740 f8789000 f67c5ddc
[  259.325995]  f8f870b9 f67c5e60 f878905d f4dc5c20 c1002136 000000d0 c15c25cd f8789000
[  259.325995] Call Trace:
[  259.325995]  [<c13e64df>] ? register_netdevice_notifier+0x7f/0x1a0
[  259.336610]  [<c11e389a>] ? proc_create_data+0x6a/0xc0
[  259.336610]  [<f8789000>] ? 0xf8789000
[  259.336610]  [<f8f870b9>] ? rtw_ndev_notifier_register+0x12/0x14 [8192eu]
[  259.336610]  [<f878905d>] ? rtw_drv_entry+0x5d/0x1000 [8192eu]
[  259.336610]  [<c1002136>] ? do_one_initcall+0xc6/0x200
[  259.336610]  [<f8789000>] ? 0xf8789000
[  259.336610]  [<c113309d>] ? free_pages_prepare+0x1ad/0x2c0
[  259.336610]  [<c116179c>] ? free_vmap_area_noflush+0x2c/0x70
[  259.336610]  [<c14cb53d>] ? do_init_module+0x21/0x198
[  259.336610]  [<c14cb56c>] ? do_init_module+0x50/0x198
[  259.336610]  [<c10d099a>] ? load_module+0x1b4a/0x2300
[  259.336610]  [<c10d130d>] ? SyS_finit_module+0x7d/0xc0
[  259.336610]  [<c11474ab>] ? vm_mmap_pgoff+0x7b/0xa0
[  259.336610]  [<c14d0e5f>] ? sysenter_do_call+0x12/0x12
[  259.336610] Code: c2 02 05 d8 32 00 00 b9 06 00 00 00 e8 e9 f5 ff ff b8 00 00 00 00 5d c3 55 89 e5 56 53 83 ec 10 3e 8d 74 26 00 8b 81 28 01 00 00 <81> 78 28 8e 60 f9 f8 75 35 89 cb 89 d6 83 3d 08 94 fe f8 07 76
[  259.336610] EIP: [<f8f86d70>] rtw_ndev_notifier_call+0x13/0x5d [8192eu] SS:ESP 0068:f67c5d98
[  259.336610] CR2: 0000000000000028
[  259.449971] ---[ end trace e3cc2cdff0eea544 ]---
```

Any ideas about how to solve this issue?

Mange, 2016-03-03:
Sorry, I do not. I'm not a driver or kernel developer.

This is just a mirror of the code and patches I could find spread around on
the Internet.

Let's hope someone looks into this and provides a fix.

mirh, 2017-06-10:
Still a thing?

CGarces, 2019-11-20:
Version  v5.6.4 merged at #149 
Please test again an reopen the issue if needed

closed 2019-11-20 by CGarces