Mange/rtl8192eu-linux-driver #6: Doesn't work with tl-wn822n v4

yoavst, 2016-06-29:
Ubuntu 16.04:
```
git clone https://github.com/Mange/rtl8192eu-linux-driver/
cd rtl8192eu-linux-driver
make # added an ignore to date error on makefile
sudo make install
reboot
```

lsusb:
```
bus 001: Device 003: ID 2357:0108
```

Have I missed a step?

Mange, 2016-06-29:
I think that's all. Let me know if you figure it out so I can integrate
your patch.

I am not a kernel developer. This is just a mirror for the official driver
with some patches that I've found laying around the Web that made it
compile on my Raspberry Pi 2.

yoavst, 2016-06-29:
Well, I'm Java developer, not C developer :(

I've contacted TP-Link and asked them for help. If there is something useful to this repo, I'll reopen the issue.

closed 2016-06-29 by yoavst

Mange, 2016-06-29:
Thank you. Good luck!

yoavst, 2016-06-30:
They've pointed that the driver should only work with `Linux (kernel 2.6.18 ~ 3.10.10)`. So I've checked their driver on Ubuntu 12.04 and it does work. I'll ask them to work on an update.

jeremyb31, 2016-07-11:
Hello Mange this commit should work for yoavst
https://github.com/jeremyb31/rtl8192eu-linux-driver/commit/56ccb6b05cff08de1beddbbd03efdb36eff1416d

yoavst, 2016-07-12:
I've installed your driver @jeremyb31 , 
`+	{USB_DEVICE(0x2357, 0x0108),.driver_info = RTL8192E}, /* TP-Link - Cameo */`
```lsusb
Bus 001 Device 003: ID 2357:0108
...
`
So it still doesn't recognize it as a wireless adapter.

Maybe it is because I've tried to install other drivers before?

jeremyb31, 2016-07-16:
Check
dkms status

There might be another version installed using dkms and causing issues.  You can also check
modinfo 8192eu | grep 0108

yoavst, 2016-07-16:
```
root@ubuntu:~# dkms status
root@ubuntu:~# lsusb
Bus 001 Device 002: ID 2357:0108  
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 002 Device 002: ID 0e0f:0003 VMware, Inc. Virtual Mouse
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
root@ubuntu:~# modinfo 8192eu | grep 0108
alias:          usb:v2357p0108d*dc*dsc*dp*ic*isc*ip*in*
root@ubuntu:~# lsmod 
Module                  Size  Used by
cfg80211              565248  0
8192eu                929792  0
vmw_vsock_vmci_transport    28672  1
vsock                  36864  2 vmw_vsock_vmci_transport
snd_ens1371            28672  2
snd_ac97_codec        131072  1 snd_ens1371
gameport               16384  1 snd_ens1371
ac97_bus               16384  1 snd_ac97_codec
coretemp               16384  0
crct10dif_pclmul       16384  0
crc32_pclmul           16384  0
aesni_intel           167936  0
snd_pcm               106496  2 snd_ac97_codec,snd_ens1371
aes_x86_64             20480  1 aesni_intel
snd_seq_midi           16384  0
snd_seq_midi_event     16384  1 snd_seq_midi
snd_rawmidi            32768  2 snd_ens1371,snd_seq_midi
snd_seq                69632  2 snd_seq_midi_event,snd_seq_midi
lrw                    16384  1 aesni_intel
snd_seq_device         16384  3 snd_seq,snd_rawmidi,snd_seq_midi
snd_timer              32768  2 snd_pcm,snd_seq
gf128mul               16384  1 lrw
snd                    81920  11 snd_ac97_codec,snd_timer,snd_pcm,snd_seq,snd_rawmidi,snd_ens1371,snd_seq_device
glue_helper            16384  1 aesni_intel
vmw_balloon            20480  0
ablk_helper            16384  1 aesni_intel
joydev                 20480  0
input_leds             16384  0
soundcore              16384  1 snd
cryptd                 20480  2 aesni_intel,ablk_helper
serio_raw              16384  0
nfit                   32768  0
shpchp                 36864  0
vmw_vmci               65536  2 vmw_vsock_vmci_transport,vmw_balloon
8250_fintek            16384  0
i2c_piix4              24576  0
mac_hid                16384  0
parport_pc             32768  0
ppdev                  20480  0
lp                     20480  0
parport                49152  3 lp,ppdev,parport_pc
autofs4                40960  2
hid_generic            16384  0
usbhid                 49152  0
hid                   118784  2 hid_generic,usbhid
vmwgfx                237568  3
ttm                    98304  1 vmwgfx
drm_kms_helper        139264  1 vmwgfx
syscopyarea            16384  1 drm_kms_helper
psmouse               126976  0
sysfillrect            16384  1 drm_kms_helper
sysimgblt              16384  1 drm_kms_helper
fb_sys_fops            16384  1 drm_kms_helper
drm                   360448  6 ttm,drm_kms_helper,vmwgfx
mptspi                 24576  2
mptscsih               40960  1 mptspi
ahci                   36864  0
libahci                32768  1 ahci
e1000                 135168  0
mptbase               102400  2 mptspi,mptscsih
scsi_transport_spi     32768  1 mptspi
pata_acpi              16384  0
fjes                   28672  0
```

jeremyb31, 2016-07-16:
rfkill list all; lshw -c net

It should work as the module has the correct alias line

yoavst, 2016-07-16:
Oh, well. It does work! Just not shown on `lsusb` as recognized. Thank you very much!

Mange, 2016-07-16:
> @Mange <https://github.com/Mange> I think it worth merging ;)

Do you have a link to the source code for it? I could check it out then.

yoavst, 2016-07-16:
@Mange https://github.com/jeremyb31/rtl8192eu-linux-driver

jeremyb31, 2016-07-17:
@Mange you just need to add https://github.com/jeremyb31/rtl8192eu-linux-driver/commit/56ccb6b05cff08de1beddbbd03efdb36eff1416d

yoavst, 2016-07-17:
@jeremyb31 I think it'll be easier if you'll make a pull request.

yoavst, 2016-07-17:
Sent a pull request from jermy's repo.

Mange, 2016-07-17:
Thanks. I'll get to it the next time I'm at my PC.

RattusSolarus, 2016-07-23:
Thanks jeremyb31. Works perfectly with TP-Link WN822 V4 on MX-15 with Debian kernel 4.6.0-0.bpo.1-amd64 . 
I was tearing my hair out trying to compile the download from TP-Link site.

dexter1989, 2016-09-05:
i have this tp link tl wn822n V4 iam using kali linux 2016.1 but wlan0 not showing up could any one give me any information to help me install this wn822 v4 i appreciate it

i used this driver but i have a make issues with compiling
http://www.tp-link.com/en/download/TL-WN822N.html#Driver

yoavst, 2016-09-05:
```
git clone https://github.com/jeremyb31/rtl8192eu-linux-driver
cd rtl8192eu-linux-driver
sudo make
sudo make install
```

dexter1989, 2016-09-06:
@yoavst  i installed this driver git clone https://github.com/jeremyb31/rtl8192eu-linux-driver
and i follow the instructions but nothing happend

dexter1989, 2016-09-06:
@yoavst i installed this driver git clone https://github.com/jeremyb31/rtl8192eu-linux-driver
 but the wireless is working so slow, any help about that please .

yoavst, 2016-09-08:
I used this driver only for vm and for kali Linux, so idk.

rbuzio, 2016-11-29:
Thank you guys! This driver worked perfectly with my TL-WN822N EU/4.0 on Mint 17.3 with kernel 3.19.
A life saver ;)

FilipeLaborde, 2016-11-29:
Works great iwth TL-WN822N on Ubuntu 16.04! Thanks guys!

MatthieuCotton, 2016-12-15:
Thanks you all. I used :

git clone https://github.com/jeremyb31/rtl8192eu-linux-driver
cd rtl8192eu-linux-driver
sudo make
sudo make install

It works perfectly on ubuntu 16.04
Thank you again

MatthieuCotton, 2016-12-15:
Thank you all ! 
It works great with : 

git clone https://github.com/jeremyb31/rtl8192eu-linux-driver
cd rtl8192eu-linux-driver
sudo make
sudo make install

Thank you again. Just a small problem, when I reboot my PC I have to unplug/replug my dongle to have the wifi ? Someone has an issue ? Thank you again

frankipl, 2017-05-23:
Hi
I have just compiled this driver for TL-WN822N Ver. 4.0 and it works.
However, I cannot start hostapd with this card as it says that it cant set the device into AP mode.
Also iw list does not shows any data (I was looking for Supported interface modes: AP)
Can this card work as AP ?

Best Regards
Marek

CGarces, 2017-05-23:
Hi @frankipl, can you please test [this](https://github.com/CGarces/rtl8192eu-linux-driver/tree/v4.4.1)?

gilcu3, 2017-05-23:
@frankipl I have exactly the same device, and the driver posted by @CGarces works like a charm for me in AP mode.

frankipl, 2017-05-23:
Hi
With this module it works as an AP.
However, I need to connect 80 devices to my AP, and it seems that this card is only able to connect 19, after connecting 19 devices it does not connect any more, hostapd freezes, even command strace does not show any data, I need to kill -9 hostapd - no... command doesn't work, reboot is needed.

With this config I have tested 6 different usb based wifi cards, so.... it is not easy ;)

Best Regards
Marek

Gourlae, 2017-07-25:
Hello,
I tried to install it, but when I enter the make command, this error is showing up:
`make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.9.0-kali4-amd64/build M=/root/rtl8192eu-linux-driver  modules
make[1]: *** /lib/modules/4.9.0-kali4-amd64/build: No such file or directory.  Stop.
Makefile:1335: recipe for target 'modules' failed
make: *** [modules] Error 2`
Anyone knows what to do?
Thank you

FFY00, 2017-07-28:
@frankipl you can use multiple APs with the same ESSID. Don't know what it's your goal but maybe this would acheive it.

jeremyb31, 2017-07-30:
@Gourlae You need the linux-header files to compile.  https://unix.stackexchange.com/a/328663/102332
Might be enough help

mdsadiqueinam, 2017-10-08:
@jeremyb31
I have kali linux 2017.2
I want use my Tp-link tl-wn822n v 4.0
rtl8192EU firmware I tried to install the driver using this
git clone https://github.com/jeremyb31/rtl8192eu-linux-driver
cd rtl8192eu-linux-driver
sudo make
sudo make install
But when I use make it shows some error
Like
Make:***[modules] Error 2
Please help

mdsadiqueinam, 2017-10-08:
<pre><font color="#CC0000"><b>root@kali</b></font>:<font color="#3465A4"><b>~</b></font># cd rtl8192eu-linux-driver
<font color="#CC0000"><b>root@kali</b></font>:<font color="#3465A4"><b>~/rtl8192eu-linux-driver</b></font># make
make ARCH=i386 CROSS_COMPILE= -C /lib/modules/4.12.0-kali2-686-pae/build M=/root/rtl8192eu-linux-driver  modules
make[1]: Entering directory &apos;/usr/src/linux-headers-4.12.0-kali2-686-pae&apos;
  CC [M]  /root/rtl8192eu-linux-driver/core/rtw_cmd.o
In file included from <b>/root/rtl8192eu-linux-driver/include/drv_types.h:32:0</b>,
                 from <b>/root/rtl8192eu-linux-driver/core/rtw_cmd.c:22</b>:
<b>/root/rtl8192eu-linux-driver/include/osdep_service.h:</b> In function ‘<b>thread_enter</b>’:
<b>/root/rtl8192eu-linux-driver/include/osdep_service.h:342:2:</b> <font color="#CC0000"><b>error: </b></font>implicit declaration of function ‘<b>allow_signal</b>’ [<font color="#CC0000"><b>-Werror=implicit-function-declaration</b></font>]
  <font color="#CC0000"><b>allow_signal</b></font>(SIGTERM);
  <font color="#CC0000"><b>^~~~~~~~~~~~</b></font>
<b>/root/rtl8192eu-linux-driver/include/osdep_service.h:</b> In function ‘<b>flush_signals_thread</b>’:
<b>/root/rtl8192eu-linux-driver/include/osdep_service.h:352:6:</b> <font color="#CC0000"><b>error: </b></font>implicit declaration of function ‘<b>signal_pending</b>’ [<font color="#CC0000"><b>-Werror=implicit-function-declaration</b></font>]
  if (<font color="#CC0000"><b>signal_pending</b></font> (current))
      <font color="#CC0000"><b>^~~~~~~~~~~~~~</b></font>
<b>/root/rtl8192eu-linux-driver/include/osdep_service.h:354:3:</b> <font color="#CC0000"><b>error: </b></font>implicit declaration of function ‘<b>flush_signals</b>’ [<font color="#CC0000"><b>-Werror=implicit-function-declaration</b></font>]
   <font color="#CC0000"><b>flush_signals</b></font>(current);
   <font color="#CC0000"><b>^~~~~~~~~~~~~</b></font>
cc1: some warnings being treated as errors
/usr/src/linux-headers-4.12.0-kali2-common/scripts/Makefile.build:307: recipe for target &apos;/root/rtl8192eu-linux-driver/core/rtw_cmd.o&apos; failed
make[4]: *** [/root/rtl8192eu-linux-driver/core/rtw_cmd.o] Error 1
/usr/src/linux-headers-4.12.0-kali2-common/Makefile:1532: recipe for target &apos;_module_/root/rtl8192eu-linux-driver&apos; failed
make[3]: *** [_module_/root/rtl8192eu-linux-driver] Error 2
Makefile:152: recipe for target &apos;sub-make&apos; failed
make[2]: *** [sub-make] Error 2
Makefile:8: recipe for target &apos;all&apos; failed
make[1]: *** [all] Error 2
make[1]: Leaving directory &apos;/usr/src/linux-headers-4.12.0-kali2-686-pae&apos;
Makefile:1335: recipe for target &apos;modules&apos; failed
make: *** [modules] Error 2
</pre>
please help

jeremyb31, 2017-10-10:
Delete the current rtl8192eu-linux-driver directory and use Mange's version as it appears to have the patches to fix those errors

mdsadiqueinam, 2017-10-10:
I want to know something about that Tp-link tl-wn822n which is rtl8192EU chipset does this device support moniter mode or packet injection

mdsadiqueinam, 2017-10-10:
A new issue arrived while installing mange's driver
<pre><font color="#CC0000"><b>root@kali</b></font>:<font color="#3465A4"><b>~/rtl8192eu-linux-driver</b></font># make install
install -p -m 644 8192eu.ko  /lib/modules/4.12.0-kali2-686-pae/kernel/drivers/net/wireless/
/sbin/depmod -a 4.12.0-kali2-686-pae
depmod: ERROR: fstatat(3, buil): No such file or directory
</pre>

jeremyb31, 2017-10-10:
I would try
`sudo depmod -a`
See if any errors show or you can just add 8192eu to /etc/modules

Neilpmcd, 2021-03-26:
Thanks guys for your effort in making this all doable.  I'm getting what looks like small errors in the makefile log.  This may be do to the RPI 4 B 64 OS? 

DKMS make.log for rtl8192eu-1.0 for kernel 5.10.17-v8+ (aarch64)
Fri 26 Mar 2021 04:41:39 PM CET
make ARCH=arm CROSS_COMPILE= -C /lib/modules/5.10.17-v8+/build M=/var/lib/dkms/rtl8192eu/1.0/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.10.17-v8+'
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_cmd.o
gcc: error: unrecognized argument in option ‘-mabi=apcs-gnu’
gcc: note: valid arguments to ‘-mabi=’ are: ilp32 lp64
gcc: error: unrecognized command line option ‘-mapcs’; did you mean ‘--specs’?
gcc: error: unrecognized command line option ‘-mno-sched-prolog’; did you mean ‘-Wno-sign-promo’?
gcc: error: unrecognized command line option ‘-msoft-float’
make[2]: *** [scripts/Makefile.build:279: /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1804: /var/lib/dkms/rtl8192eu/1.0/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.10.17-v8+'
make: *** [Makefile:1700: modules] Error 2

CGarces, 2021-03-27:
Hi @Neilpmcd

Looks like an issue with ARM64 compilation not something related to this bug on tl-wn822n from 5 years ago.

I'm not sure about the parameters of your make.

Please follow the  [README.md](https://github.com/Mange/rtl8192eu-linux-driver/blob/realtek-4.4.x/README.md#building-and-installing-using-dkms)

Also you can use the command line.

`make CONFIG_PLATFORM_ARM_AARCH64=y CONFIG_PLATFORM_I386_PC=n
`

If still failing, please open a new issue.

Neilpmcd, 2021-03-28:
Very much appreciated, will do now.

It looked like Mange was combining two directories into one clone command, I wasn't sure I had it right but will go with your take.

Thanks,

Neil.

Neilpmcd, 2021-03-28:
Makefile:681: arch/aarch64/Makefile: No such file or directory

Now, it really appears like one small steps.  The readme and the commands you shared seemed to download all the drivers (I hope you don't mind here, I'm happy to create new):

Building module:
cleaning build area...
'make' all KVER=5.10.17-v8+...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.10.17-v8+ (aarch64)
Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.

DKMS make.log for rtl8192eu-1.0 for kernel 5.10.17-v8+ (aarch64)
Sun 28 Mar 2021 02:42:27 PM CEST
make ARCH=aarch64 CROSS_COMPILE= -C /lib/modules/5.10.17-v8+/build M=/var/lib/dkms/rtl8192eu/1.0/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.10.17-v8+'
Makefile:681: arch/aarch64/Makefile: No such file or directory
make[1]: *** No rule to make target 'arch/aarch64/Makefile'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-5.10.17-v8+'
make: *** [Makefile:2293: modules] Error 2

PR #7 (2016-08-01): Add support for tp-links devices
Fixes #6