Mange/rtl8192eu-linux-driver #8: install: cannot stat ‘8192eu.ko’: No such file or directory

farazappy, 2016-08-01:
Using `make install` gives me the following : 
`install -p -m 644 8192eu.ko  /lib/modules/4.3.0-kali1-amd64/kernel/drivers/net/wireless/
install: cannot stat ‘8192eu.ko’: No such file or directory
Makefile:1341: recipe for target 'install' failed
make: *** [install] Error 1`

Help is appreciated!

Mange, 2016-08-01:
Did you run "make" first?

farazappy, 2016-08-09:
Well I solved that issue but now airmon-ng returns a tabular data and under `Driver` it says `?????`

Any fix?

Mange, 2016-08-10:
No. Let me know if you find anything. 😯

nomadmzl, 2017-05-14:
Maybe [comment #11 could be related](https://github.com/CGarces/rtl8192eu-linux-driver/issues/1).

daliboss, 2017-10-25:
How did you fix it

Aspect9, 2018-10-04:
the Linux Drivers for the Wn722n V3 beta version are not working! cannot "make" or "make install" it presents "Error-1" cannot locate - Does anyone have any ideas? tryna get the V3 working on linux - Kali failed miserably... -_-

CGarces, 2019-11-20:
Version  v5.6.4 merged at #149 
Please test again an reopen the issue if needed

closed 2019-11-20 by CGarces