Mange/rtl8192eu-linux-driver #10: TP-LINK TL-WN823N V2 WiFi adapter

morfikov, 2016-08-14:
I've been testing this driver for a few days, and I can tell that it works nice with TP-LINK TL-WN823N V2 WiFi adapter . I've got transfer around 170 mbit/s. There's a few problems with replugging/unplugging the device. I always get the following log:

```
Aug 14 12:14:07 morfikownia kernel: ------------[] cut here ]------------
Aug 14 12:14:07 morfikownia kernel: WARNING: CPU: 0 PID: 36955 at /build/linux-aOJbQa/linux-4.6.4/fs/proc/generic.c:564 remove_proc_entry+0x171/0x180
Aug 14 12:14:07 morfikownia kernel: name 'wlan20'
Aug 14 12:14:07 morfikownia kernel: Modules linked in: 8192eu(OE) veth(E) nls_utf8(E) nls_cp437(E) vfat(E) fat(E) uas(E) usb_storage(E) fuse(E) cpuid(E) cpufreq_conservative(E) cpufreq_userspace(E) cpufreq_powersave(E) cpufreq_stats(E) bridge(E) stp(E) llc(E) drbg(E) ansi_cprng(E) ctr(E) ccm(E) nf_log_ipv4(E) nf_log_common(E) xt_LOG(E) xt_limit(E) xt_pkttype(E) ipt_REJECT(E) nf_reject_ipv4(E) ipt_SYNPROXY(E) nf_synproxy_core(E) xt_conntrack(E) nf_conntrack_netlink(E) iptable_filter(E) ipt_MASQUERADE(E) nf_nat_masquerade_ipv4(E) iptable_nat(E) nf_nat_ipv4(E) nf_nat(E) xt_TCPMSS(E) xt_comment(E) xt_statistic(E) xt_owner(E) xt_mark(E) iptable_mangle(E) xt_tcpudp(E) xt_CT(E) xt_multiport(E) xt_set(E) iptable_raw(E) ip_tables(E) hid_a4tech(E) iTCO_wdt(E) iTCO_vendor_support(E) ip_set_hash_ip(E) ip_set_hash_net(E) acpi_cpufreq(E)
Aug 14 12:14:07 morfikownia kernel:  ip_set(E) nfnetlink(E) battery(E) ac(E) intel_powerclamp(E) lpc_ich(E) processor(E) hp_wmi(E) sparse_keymap(E) shpchp(E) sg(E) mfd_core(E) i2c_i801(E) mei_me(E) tpm_tis(E) tpm(E) mei(E) joydev(E) serio_raw(E) nfs(E) fscache(E) arc4(E) brcmsmac(E) cordic(E) brcmutil(E) bcma(E) mac80211(E) cfg80211(E) rfkill(E) evdev(E) eeprom(E) tun(E) xt_recent(E) snd_hda_codec_hdmi(E) snd_hda_codec_realtek(E) snd_hda_codec_generic(E) snd_hda_intel(E) snd_hda_codec(E) snd_hda_core(E) snd_hwdep(E) zram(E) snd_pcm(E) snd_timer(E) snd(E) soundcore(E) act_mirred(E) sch_fq_codel(E) xt_connmark(E) ifb(E) nf_conntrack_ipv6(E) nf_defrag_ipv6(E) nf_conntrack_ipv4(E) nf_defrag_ipv4(E) zsmalloc(E) nf_conntrack(E) cls_cgroup(E) xt_cgroup(E) x_tables(E) coretemp(E) lz4_compress(E) bonding(E) ecryptfs(E) cbc(E) hmac(E)
Aug 14 12:14:07 morfikownia kernel:  encrypted_keys(E) nfsd(E) auth_rpcgss(E) nfs_acl(E) lockd(E) grace(E) parport_pc(E) ppdev(E) sunrpc(E) lp(E) parport(E) loop(E) autofs4(E) ext4(E) ecb(E) lrw(E) glue_helper(E) ablk_helper(E) cryptd(E) aes_x86_64(E) crc16(E) jbd2(E) mbcache(E) btrfs(E) crc32c_generic(E) xor(E) raid6_pq(E) xts(E) gf128mul(E) algif_skcipher(E) af_alg(E) dm_crypt(E) dm_mod(E) sr_mod(E) sd_mod(E) cdrom(E) hid_generic(E) usbhid(E) hid(E) psmouse(E) ahci(E) libahci(E) r8169(E) mii(E) libata(E) scsi_mod(E) i915(E) fjes(E) video(E) i2c_algo_bit(E) thermal(E) wmi(E) drm_kms_helper(E) drm(E) ehci_pci(E) ehci_hcd(E) usbcore(E) usb_common(E) button(E) [last unloaded: 8192eu]
Aug 14 12:14:07 morfikownia kernel: CPU: 0 PID: 36955 Comm: kworker/0:2 Tainted: G        WC OE   4.6.0-1-amd64 #1 Debian 4.6.4-1
Aug 14 12:14:07 morfikownia kernel: Hardware name: Hewlett-Packard HP G62 Notebook PC              /1439, BIOS F.48 11/09/2011
Aug 14 12:14:07 morfikownia kernel: Workqueue: usb_hub_wq hub_event [usbcore]
Aug 14 12:14:07 morfikownia kernel:  0000000000000286 00000000071d7865 ffffffff81311505 ffff88006aa27a38
Aug 14 12:14:07 morfikownia kernel:  0000000000000000 ffffffff8107a50e ffff8800280cd000 ffff88006aa27a90
Aug 14 12:14:07 morfikownia kernel:  ffff8800637b4a80 ffffc900004c2000 ffff8800280cd000 ffff8800637b4a80
Aug 14 12:14:07 morfikownia kernel: Call Trace:
Aug 14 12:14:07 morfikownia kernel:  [<ffffffff81311505>] ? dump_stack+0x5c/0x77
Aug 14 12:14:07 morfikownia kernel:  [<ffffffff8107a50e>] ? __warn+0xbe/0xe0
Aug 14 12:14:07 morfikownia kernel:  [<ffffffff8107a58f>] ? warn_slowpath_fmt+0x5f/0x80
Aug 14 12:14:07 morfikownia kernel:  [<ffffffff8126001c>] ? __xlate_proc_name+0x5c/0xa0
Aug 14 12:14:07 morfikownia kernel:  [<ffffffff81260c31>] ? remove_proc_entry+0x171/0x180
Aug 14 12:14:07 morfikownia kernel:  [<ffffffffc0dcbf00>] ? rtw_adapter_proc_deinit+0x74/0x8a [8192eu]
Aug 14 12:14:07 morfikownia kernel:  [<ffffffffc0db774a>] ? rtw_ndev_uninit+0x3b/0x3d [8192eu]
Aug 14 12:14:07 morfikownia kernel:  [<ffffffff814c2ebb>] ? rollback_registered_many+0x18b/0x300
Aug 14 12:14:07 morfikownia kernel:  [<ffffffff814c3069>] ? rollback_registered+0x39/0x60
Aug 14 12:14:07 morfikownia kernel:  [<ffffffff814c4050>] ? unregister_netdevice_queue+0x40/0x90
Aug 14 12:14:07 morfikownia kernel:  [<ffffffff814c40b8>] ? unregister_netdev+0x18/0x20
Aug 14 12:14:07 morfikownia kernel:  [<ffffffffc0db7de1>] ? rtw_unregister_netdevs+0x40/0x5a [8192eu]
Aug 14 12:14:07 morfikownia kernel:  [<ffffffffc0db93b3>] ? rtw_dev_remove+0x32/0x88 [8192eu]
Aug 14 12:14:07 morfikownia kernel:  [<ffffffffc002272f>] ? usb_unbind_interface+0x7f/0x280 [usbcore]
Aug 14 12:14:07 morfikownia kernel:  [<ffffffff8143f7da>] ? __device_release_driver+0x9a/0x150
Aug 14 12:14:07 morfikownia kernel:  [<ffffffff8143f8ae>] ? device_release_driver+0x1e/0x30
Aug 14 12:14:07 morfikownia kernel:  [<ffffffff8143ee79>] ? bus_remove_device+0xf9/0x170
Aug 14 12:14:07 morfikownia kernel:  [<ffffffff8143b267>] ? device_del+0x127/0x250
Aug 14 12:14:07 morfikownia kernel:  [<ffffffffc0026eab>] ? usb_remove_ep_devs+0x1b/0x30 [usbcore]
Aug 14 12:14:07 morfikownia kernel:  [<ffffffffc00200ce>] ? usb_disable_device+0x7e/0x260 [usbcore]
Aug 14 12:14:07 morfikownia kernel:  [<ffffffffc0015e40>] ? usb_disconnect+0x90/0x280 [usbcore]
Aug 14 12:14:07 morfikownia kernel:  [<ffffffffc0011c64>] ? set_port_feature+0x44/0x50 [usbcore]
Aug 14 12:14:07 morfikownia kernel:  [<ffffffffc0018077>] ? hub_event+0x767/0x14f0 [usbcore]
Aug 14 12:14:07 morfikownia kernel:  [<ffffffff81092dcd>] ? process_one_work+0x15d/0x410
Aug 14 12:14:07 morfikownia kernel:  [<ffffffff810930cd>] ? worker_thread+0x4d/0x480
Aug 14 12:14:07 morfikownia kernel:  [<ffffffff81093080>] ? process_one_work+0x410/0x410
Aug 14 12:14:07 morfikownia kernel:  [<ffffffff81098f1d>] ? kthread+0xcd/0xf0
Aug 14 12:14:07 morfikownia kernel:  [<ffffffff815c69b2>] ? ret_from_fork+0x22/0x40
Aug 14 12:14:07 morfikownia kernel:  [<ffffffff81098e50>] ? kthread_create_on_node+0x190/0x190
Aug 14 12:14:07 morfikownia kernel: ---[] end trace 0a9b61bbcf52d0e9 ]---
```

Here's the kernel version:
```

$ uname -a
Linux morfikownia 4.6.0-1-amd64 #1 SMP Debian 4.6.4-1 (2016-07-18) x86_64 GNU/Linux
```

Mange, 2016-08-14:
Thank you for letting us know.

vherasme, 2016-09-26:
Hi, I am a newbie. I wish you could tell me how to make the wireless adapter work. Regards,

scdas141, 2017-01-28:
Even though it is a little late, I hope this tutorial I will help you and other, especially new users in compiling this driver package, including how to make it work in kernel 4.9.x..

[https://scdas141.wordpress.com/2017/01/28/how-to-compile-mangertl8192eu-linux-driver-driver/](url)

Fopper, 2017-04-06:
The next Ubuntu version will ship with 4.10.x kernel, so I needed the work around from scdas141 to get the driver working. Thanks for the helpfull howto!

Lenseth, 2017-04-06:
i did what scadas141 said with ArchLinux on kernel 4.10 and it didn't worked :/
did the same with KDE Neon on kernel 4.8 and worked

shashishailaj, 2017-04-09:
This works absolutely like a charm with TP-LINK TL-WN823N V2 WiFi adapter running on KDE neon 4.9.2 (based on Ubuntu 16 LTS) Kernel 4.8.

masterzorag, 2017-05-03:
Hi guys, I've bought this adapter.
After testing upstream new rtl8xxxu driver that locks me at 1MBit/s, I've started using the Realtek one;
I've found and [pushed a most recent version](https://github.com/masterzorag/RTL8192EU-linux), see mirrored [Release Notes](https://github.com/masterzorag/RTL8192EU-linux/blob/master/Linux_Release_notes.pdf).
Realtek supports linux-4.4.65 longterm, working stable at 130.0 MBit/s.
I've patched differently from this repo for [linux-4.7 to get compiled](https://github.com/masterzorag/RTL8192EU-linux/tree/linux-4.7).

nomadmzl, 2017-05-05:
Hi, 

I've downloaded this one: https://github.com/masterzorag/RTL8192EU-linux compiled and installed the folder "rtl8192eu", after restart I'm locked at 1MBit/s upload speed. 

Running lsmod I see the "rtl8xxxu" module loaded, so, how can I install and load the Realtek one that had worked for you to unlock the 1MBit/s upload limit?

(my kernel is 4.8.0-49-generic). Thanks

masterzorag, 2017-05-05:
@nomadmzl just
```sh
rmmod rtl8xxxu
modprobe 8192eu
```
Maybe interface name will increment, check `dmesg`
`iwconfig` if using WEXT, `iw wlan0 link` if using new CFG80211

Also, linux-4.8.y seems EndOfLine, making linux-4.4.y more updated than it...

nomadmzl, 2017-05-05:
@masterzorag 

Hi, inside the folder "rtl8192eu" I'd ran "make && make install", so I have unloaded the module rtl8xxxu module with rmmod, but when I try to load the 8192eu one I get: 

~$ modprobe 8192eu
modprobe: FATAL: Module 8192eu not found in directory /lib/modules/4.8.0-49-generic

What I'm doing wrong? I'm using the files in: https://github.com/masterzorag/RTL8192EU-linux

Bye!

nomadmzl, 2017-05-06:
@masterzorag 

I must forgeting someting, when compile I get this:

root@tron:/home/user/desktop/RTL8192EU-linux-master/rtl8192eu# make && make install
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.8.0-49-generic/build M=/home/user/desktop/RTL8192EU-linux-master/rtl8192eu  modules
make[1]: directory '/usr/src/linux-headers-4.8.0-49-generic'
  CC [M]  /home/user/desktop/RTL8192EU-linux-master/rtl8192eu/os_dep/linux/ioctl_linux.o
/home/user/desktop/RTL8192EU-linux-master/rtl8192eu/os_dep/linux/ioctl_linux.c: In function ‘rtw_ioctl_wext_private’:
/home/user/desktop/RTL8192EU-linux-master/rtl8192eu/os_dep/linux/ioctl_linux.c:14035:5: error: implicit declaration of function ‘is_compat_task’ [-Werror=implicit-function-declaration]
  if(is_compat_task())
     ^
cc1: some warnings being treated as errors
scripts/Makefile.build:289: failed in the instructions for the target '/home/user/desktop/RTL8192EU-linux-master/rtl8192eu/os_dep/linux/ioctl_linux.o'
make[2]: *** [/home/user/desktop/RTL8192EU-linux-master/rtl8192eu/os_dep/linux/ioctl_linux.o] Error 1
Makefile:1491: failed in the instructions for the target '_module_/home/user/desktop/RTL8192EU-linux-master/rtl8192eu'
make[1]: *** [_module_/home/user/desktop/RTL8192EU-linux-master/rtl8192eu] Error 2
make[1]: exit directory '/usr/src/linux-headers-4.8.0-49-generic'
Makefile:1686: failed in the instructions for the target 'modules'
make: *** [modules] Error 2
root@tron:/home/user/desktop/RTL8192EU-linux-master/rtl8192eu# 

Just to give it a chance, after restart I'm running rmmod rtl8xxxu and modprobe 8192eu, so I get: "modprobe: FATAL: Module 8192eu not found in directory /lib/modules/4.8.0-49-generic"

What am I doing wrong?

masterzorag, 2017-05-06:
@nomadmzl, you are running linux-4.8.y that is not officially supported by Realtek driver;
We have to patch the code to get it ported and I've patch for linux-4.7;
You should move to linux-4.4.y to get it working as it should, use you package manager to get it;
If you want to try my edits

CGarces, 2017-05-06:
@masterzorag, @nomadmzl maybe you should continue the discussion on a new issue 
The compilation problems with 4.8 are not related to this repo
Will be more easy to track for future references if someone open a issue directly on https://github.com/masterzorag/RTL8192EU-linux/issues

nomadmzl, 2017-05-07:
Solved for me:

it seems that kernels 4.8+ doesn't load the module 8192eu even if you've compiled and installed it, intead it loads anyway the generic one rtl8xxxu. 

As masterzorag said here [https://github.com/Mange/rtl8192eu-linux-driver/issues/10#issuecomment-298852261](https://github.com/Mange/rtl8192eu-linux-driver/issues/10#issuecomment-298852261) the Realtek driver works better (a lot better for me), but it seems that you need to force it to load in boot, you can identify if this is your problem running:

lshw -c network 

it will show that the wireless connection has this driver (rtl8xxxu) loaded instead the Realtek one. Check the solution that scdas141 did here: http://techpatterns.com/forums/about2574.html  worked for me like a charm.

Now I don't need to unplug and re-plug the dongle, the bit rate has upgraded from 1 Mb/s to 144 Mb/s, the tests in speedtest.net now shows the same download/upload speeds as a wired connection showed previously.

I'm running kernel 4.8.0-51.

easysid, 2017-07-02:
I have the same adapter TL-WN823N v2

`Kernel - 4.9.34-1-lts Arch Linux`

The dongle was detected by the default rtl8xxxu module, but the light didn't come on, and I was unable to actually connect to a network.

Using this repo, everything seems to work fine for now. I blacklisted the inbuilt 8xxxu module to force the 8192eu to load.

Initially, I got a lot of journal spam

> Jul 02 17:33:33 lucy kernel: RTL871X: rtl8192e_set_FwPwrMode_cmd: Mode=0 SmartPS=2 UAPSD=0
> Jul 02 17:33:35 lucy kernel: RTL871X: rtw_set_ps_mode(wifi1) Enter 802.11 power save - WIFI-TRAFFIC_IDLE
> Jul 02 17:33:35 lucy kernel: RTL871X: rtl8192e_set_FwPwrMode_cmd: Mode=1 SmartPS=2 UAPSD=0
> Jul 02 17:33:40 lucy kernel: RTL871X: rtw_set_ps_mode(wifi1) Leave 802.11 power save - WIFI-TRAFFIC_BUSY
> Jul 02 17:33:40 lucy kernel: RTL871X: rtl8192e_set_FwPwrMode_cmd: Mode=0 SmartPS=2 UAPSD=0
> Jul 02 17:33:42 lucy kernel: RTL871X: rtw_set_ps_mode(wifi1) Enter 802.11 power save - WIFI-TRAFFIC_IDLE
> Jul 02 17:33:42 lucy kernel: RTL871X: rtl8192e_set_FwPwrMode_cmd: Mode=1 SmartPS=2 UAPSD=0
> Jul 02 17:33:46 lucy kernel: RTL871X: rtw_set_ps_mode(wifi1) Leave 802.11 power save - WIFI-TRAFFIC_BUSY
> Jul 02 17:33:46 lucy kernel: RTL871X: rtl8192e_set_FwPwrMode_cmd: Mode=0 SmartPS=2 UAPSD=0
> Jul 02 17:33:48 lucy kernel: RTL871X: rtw_set_ps_mode(wifi1) Enter 802.11 power save - WIFI-TRAFFIC_IDLE
> Jul 02 17:33:48 lucy kernel: RTL871X: rtl8192e_set_FwPwrMode_cmd: Mode=1 SmartPS=2 UAPSD=0
> Jul 02 17:33:50 lucy kernel: RTL871X: rtw_set_ps_mode(wifi1) Leave 802.11 power save - WIFI-TRAFFIC_BUSY
> Jul 02 17:33:50 lucy kernel: RTL871X: rtl8192e_set_FwPwrMode_cmd: Mode=0 SmartPS=2 UAPSD=0
> 

adding the following to /etc/modprobe.d/8192.conf stopped those

`options 8192eu rtw_power_mgnt=0 rtw_enusbss=0`

(I don't thing rtl_enusbss=0 is required, because it already defaults to 0)

I do however get some more messages like the following

> Jul 02 21:50:23 lucy kernel: RTL871X: UpdateHalRAMask8192E => mac_id:0, networkType:0x03, mask:0x00000fff
> Jul 02 21:50:23 lucy kernel: RTL871X: rtl8192e_set_raid_cmd, bitmap=0xff0, mac_id=0x0, raid=0x6, shortGIrate=0
> Jul 02 21:50:38 lucy kernel: RTL871X: UpdateHalRAMask8192E => mac_id:0, networkType:0x03, mask:0x00000fff
> Jul 02 21:50:38 lucy kernel: RTL871X: rtl8192e_set_raid_cmd, bitmap=0xff5, mac_id=0x0, raid=0x6, shortGIrate=0
> Jul 02 21:50:48 lucy kernel: RTL871X: UpdateHalRAMask8192E => mac_id:0, networkType:0x03, mask:0x00000fff
> Jul 02 21:50:48 lucy kernel: RTL871X: rtl8192e_set_raid_cmd, bitmap=0xff0, mac_id=0x0, raid=0x6, shortGIrate=0
> Jul 02 21:51:48 lucy kernel: RTL871X: UpdateHalRAMask8192E => mac_id:0, networkType:0x03, mask:0x00000fff
> Jul 02 21:51:48 lucy kernel: RTL871X: rtl8192e_set_raid_cmd, bitmap=0xff5, mac_id=0x0, raid=0x6, shortGIrate=0
> Jul 02 21:52:34 lucy kernel: RTL871X: UpdateHalRAMask8192E => mac_id:0, networkType:0x03, mask:0x00000fff

Not sure how to get rid of them. 

Another issue is that iw does not detect the device. It must be a wext/cfg issue. I am connected to the internet, but iw lists no devices.

Thanks for this repo. :)

CGarces, 2017-07-06:
Fixed on realtek-4.4.x branch as commented by @easysid

closed 2017-07-06 by CGarces

dhruvasagar, 2018-11-21:
I have the same device TL-WN823N v2

I am on Kernel Version 4.18.0. Following is the output of trying to install this driver as a kernel module, the build seems to go through fine, but it fails to update initramfs for some reason (I am a noob at building kernel modules) and rollsback.

Any help would be appreciated.

```sh
$ sudo dkms install rtl8192eu/1.0;                                                                                                                                                                                                                                   node-1[60/92]
                                                                                                                                                                                                                                                                                  
Kernel preparation unnecessary for this kernel.  Skipping...                                                                                                                                                                                                                      
                                                                                                                                                                                                                                                                                  
Building module:                                                                                                                                                                                                                                                                  
cleaning build area...                                                                                                                                                                                                                                                            
'make' all KVER=4.18.0-11-generic..........................................                                                                                                                                                                                                       
cleaning build area...                                                                                                                                                                                                                                                            
                                                                                                                                                                                                                                                                                  
DKMS: build completed.                                                                                                                                                                                                                                                            
                                                                                                                                                                                                                                                                                  
8192eu.ko:                                                                                                                                                                                                                                                                        
Running module version sanity check.                                                                                                                                                                                                                                              
 - Original module                                                                                                                                                                                                                                                                
   - No original module exists within this kernel                                                                                                                                                                                                                                 
 - Installation                                                                                                                                                                                                                                                                   
   - Installing to /lib/modules/4.18.0-11-generic/updates/dkms/                                                                                                                                                                                                                   
                                                                                                                                                                                                                                                                                  
depmod....                                                                                                                                                                                                                        [... cut ...]
DKMS: uninstall completed.                                                                                                                                                                                                                                                        
Error! Problems with mkinitrd detected.  Automatically uninstalling this module.                                                                                                                                                                                                  
DKMS: Install Failed (mkinitrd problems).  Module rolled back to built state.
```

abhishekdanej, 2018-12-20:
Hi, I have the device TL-WN823N v2, and I was facing the same problem that everyone is facing with ubuntu 16.04 LTS. After googling, i followed steps on forums and reached Pilot6 page and got the 4.4 driver..which worked great. But after downloading the ubuntu updates, twice my driver broke. This time i searched further and found this link.. and followed the steps. It is working great now ..and with increased performance. The wifi signal shows increased strength network signal is increased too.

I have one query though, in future ..should i download standard updates or not? And since you blocked the 8xxxeu driver ..can this problem reappear? Thanks!

My unbutu details:
Linux ubuntu16 4.15.0-43-generic #46~16.04.1-Ubuntu SMP Fri Dec 7 13:30:54 UTC 2018 i686 i686 i686 GNU/Linux

dhruvasagar, 2018-12-20:
@abhishekdanej This problem can reappear. However, since you are compiling specific drivers for the hardware manually to get it working (and it does work), it would be advisable to disable the standard updates.