Mange/rtl8192eu-linux-driver #17: Need Help, Thank you

Lazy1997, 2016-12-31:
i still cant get my driver work!

root@kali:~/rtl8192eu-linux-driver# make install
install -p -m 644 8192eu.ko  /lib/modules/4.6.0-kali1-amd64/kernel/drivers/net/wireless/
/sbin/depmod -a 4.6.0-kali1-amd64

scdas141, 2017-01-25:
Please try 'make && make install'.. worked for me flawlessly many times..

michaelweghorn, 2017-02-18:
@Lazy1997: Is the problem still there or can the issue be closed?

Lazy1997, 2017-02-19:
Thank you for the help!
it still doesn't work on my wifi adapter(DWA-131), so i recently brought a TP-WN722N(2 day ago) it work!!! :D

closed 2017-02-19 by Lazy1997

Josue198s, 2017-06-24:
I cant even start the TL-Wn822N in Kali. 

root@NikolaTesla:~/Downloads# lsusb
Bus 002 Device 003: ID 8087:07da Intel Corp. 
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 004: ID 04f3:001a Elan Microelectronics Corp. 
Bus 001 Device 003: ID 2232:1035 Silicon Motion 
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 2357:0108  
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@NikolaTesla:~/Downloads# lspci
00:00.0 Host bridge: Intel Corporation 3rd Gen Core processor DRAM Controller (rev 09)
00:02.0 VGA compatible controller: Intel Corporation 3rd Gen Core processor Graphics Controller (rev 09)
00:14.0 USB controller: Intel Corporation 7 Series/C210 Series Chipset Family USB xHCI Host Controller (rev 04)
00:16.0 Communication controller: Intel Corporation 7 Series/C210 Series Chipset Family MEI Controller #1 (rev 04)
00:1a.0 USB controller: Intel Corporation 7 Series/C210 Series Chipset Family USB Enhanced Host Controller #2 (rev 04)
00:1b.0 Audio device: Intel Corporation 7 Series/C210 Series Chipset Family High Definition Audio Controller (rev 04)
00:1c.0 PCI bridge: Intel Corporation 7 Series/C210 Series Chipset Family PCI Express Root Port 1 (rev c4)
00:1c.3 PCI bridge: Intel Corporation 7 Series/C210 Series Chipset Family PCI Express Root Port 4 (rev c4)
00:1d.0 USB controller: Intel Corporation 7 Series/C210 Series Chipset Family USB Enhanced Host Controller #1 (rev 04)
00:1f.0 ISA bridge: Intel Corporation HM76 Express Chipset LPC Controller (rev 04)
00:1f.2 SATA controller: Intel Corporation 7 Series Chipset Family 6-port SATA Controller [AHCI mode] (rev 04)
00:1f.3 SMBus: Intel Corporation 7 Series/C210 Series Chipset Family SMBus Controller (rev 04)
01:00.0 Network controller: Intel Corporation Centrino Advanced-N 6235 (rev 24)
02:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 06)
root@NikolaTesla:~/Downloads# ifconfig
eth0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 50:b7:c3:0b:6a:ab  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1  (Local Loopback)
        RX packets 26  bytes 1610 (1.5 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 26  bytes 1610 (1.5 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.11  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::b504:d5fd:6b29:7537  prefixlen 64  scopeid 0x20<link>
        ether c8:f7:33:7a:70:5b  txqueuelen 1000  (Ethernet)
        RX packets 657  bytes 90484 (88.3 KiB)
        RX errors 0  dropped 1  overruns 0  frame 0
        TX packets 241  bytes 37790 (36.9 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

[   35.011093] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   35.684318] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   42.138856] wlan0: authenticate with b0:c5:54:a6:23:fc
[   42.141431] wlan0: send auth to b0:c5:54:a6:23:fc (try 1/3)
[   42.143729] wlan0: authenticated
[   42.144807] wlan0: associate with b0:c5:54:a6:23:fc (try 1/3)
[   42.152917] wlan0: RX AssocResp from b0:c5:54:a6:23:fc (capab=0x411 status=0 aid=2)
[   42.178270] wlan0: associated
[   42.178394] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   42.539564] alg: No test for fips(ansi_cprng) (fips_ansi_cprng)
[   82.846603] fuse init (API version 7.24)
[  102.947308] iwlwifi 0000:01:00.0: RF_KILL bit toggled to disable radio.
[  102.947501] wlan0: deauthenticating from b0:c5:54:a6:23:fc by local choice (Reason: 3=DEAUTH_LEAVING)
[  102.947577] iwlwifi 0000:01:00.0: Not sending command - RF KILL
[  102.947583] wlan0: HW problem - can not stop rx aggregation for b0:c5:54:a6:23:fc tid 0
[  102.947593] iwlwifi 0000:01:00.0: Not sending command - RF KILL
[... 3 lines cut ...]
[  102.947717] iwlwifi 0000:01:00.0: Not sending command - RF KILL
[  102.947726] iwlwifi 0000:01:00.0: Not sending command - RF KILL
[  102.947736] iwlwifi 0000:01:00.0: Not sending command - RF KILL
[  102.947749] iwlwifi 0000:01:00.0: Not sending command - RF KILL
[  102.947763] iwlwifi 0000:01:00.0: Not sending command - RF KILL
[  102.948898] iwlwifi 0000:01:00.0: Not sending command - RF KILL
[  102.948906] wlan0: failed to remove key (1, ff:ff:ff:ff:ff:ff) from hardware (-5)
[  102.949048] iwlwifi 0000:01:00.0: Not sending command - RF KILL
[  102.949061] iwlwifi 0000:01:00.0: Not sending command - RF KILL
[  143.370884] iwlwifi 0000:01:00.0: RF_KILL bit toggled to enable radio.
[  143.371312] iwlwifi 0000:01:00.0: L1 Disabled - LTR Disabled
[  143.378067] iwlwifi 0000:01:00.0: L1 Disabled - LTR Disabled
[  143.378158] iwlwifi 0000:01:00.0: Radio type=0x2-0x1-0x0
[  143.650926] iwlwifi 0000:01:00.0: L1 Disabled - LTR Disabled
[... cut ...]
[  143.741254] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  143.799640] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  150.203294] wlan0: authenticate with b0:c5:54:a6:23:fc
[  150.207475] wlan0: send auth to b0:c5:54:a6:23:fc (try 1/3)
[  150.218477] wlan0: authenticated
[  150.220953] wlan0: associate with b0:c5:54:a6:23:fc (try 1/3)
[  150.244183] wlan0: RX AssocResp from b0:c5:54:a6:23:fc (capab=0x411 status=0 aid=2)
[  150.275547] wlan0: associated
[  150.275683] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready