Mange/rtl8192eu-linux-driver #18: [Mange/rtl8192eu-linux-driver] unable to compile after Kernel 4.9.x upgrade..

scdas141, 2017-01-25:
Hi,

Many thanks first of all, I have been using this [Mange/rtl8192eu-linux-driver] package since kernel 4.6.x..  Infact, this has been the only working driver package for my TP-LINK TL-WN823N V2 adapter..

But since I upgraded to kernel 4.9.x, I have been unable to compile this package.. I am enclosing the error Root-terminal output for **"make && make install"**:

`root@ik--pc:/home/ik/Downloads/lm2/rtl8192eu-linux-driver-master# make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.9.0-5.1-liquorix-amd64/build M=/home/ik/Downloads/lm2/rtl8192eu-linux-driver-master  modules
make[1]: Entering directory '/usr/src/linux-headers-4.9.0-5.1-liquorix-amd64'
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/core/rtw_cmd.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/core/rtw_security.o
/home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/core/rtw_security.c: In function ‘aes_cipher’:
/home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/core/rtw_security.c:1540:5: warning: this ‘for’ clause does not guard... [-Wmisleading-indentation]
     for (j = 0; j < 8; j++)
     ^~~
/home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/core/rtw_security.c:1543:2: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the ‘for’
  payload_index = hdrlen + 8;
  ^~~~~~~~~~~~~
/home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/core/rtw_security.c: In function ‘aes_decipher’:
/home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/core/rtw_security.c:1925:5: warning: this ‘for’ clause does not guard... [-Wmisleading-indentation]
     for (j = 0; j < 8; j++)
     ^~~
/home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/core/rtw_security.c:1928:2: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the ‘for’
  payload_index = hdrlen + 8;
  ^~~~~~~~~~~~~
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/core/rtw_debug.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/core/rtw_io.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/core/rtw_ioctl_query.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/core/rtw_ioctl_set.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/core/rtw_ieee80211.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/core/rtw_mlme.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/core/rtw_mlme_ext.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/core/rtw_wlan_util.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/core/rtw_vht.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/core/rtw_pwrctrl.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/core/rtw_rf.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/core/rtw_recv.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/core/rtw_sta_mgt.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/core/rtw_ap.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/core/rtw_xmit.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/core/rtw_p2p.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/core/rtw_tdls.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/core/rtw_br_ext.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/core/rtw_iol.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/core/rtw_sreset.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/core/rtw_btcoex.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/core/rtw_beamforming.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/core/rtw_odm.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/core/efuse/rtw_efuse.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/os_dep/osdep_service.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/os_dep/linux/os_intfs.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/os_dep/linux/usb_intf.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/os_dep/linux/usb_ops_linux.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/os_dep/linux/ioctl_linux.o
/home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/os_dep/linux/ioctl_linux.c: In function ‘rtw_mp_read_reg’:
/home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/os_dep/linux/ioctl_linux.c:10395:8: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
        if ( data[i] != '\0' )
        ^~
/home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/os_dep/linux/ioctl_linux.c:10398:10: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the ‘if’
          j++;
          ^
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/os_dep/linux/xmit_linux.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/os_dep/linux/mlme_linux.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/os_dep/linux/recv_linux.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/os_dep/linux/wifi_regd.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/os_dep/linux/rtw_android.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/os_dep/linux/rtw_proc.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/hal/hal_intf.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/hal/hal_com.o
[... 25 lines cut ...]
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/hal/OUTSRC/odm.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/hal/OUTSRC/HalPhyRf.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/hal/OUTSRC/rtl8192e/HalHWImg8192E_MAC.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/hal/OUTSRC/rtl8192e/HalHWImg8192E_BB.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/hal/OUTSRC/rtl8192e/HalHWImg8192E_RF.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/hal/OUTSRC/rtl8192e/HalHWImg8192E_FW.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/hal/OUTSRC/rtl8192e/HalPhyRf_8192e.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/hal/OUTSRC/rtl8192e/odm_RegConfig8192E.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/hal/OUTSRC/rtl8192e/odm_RTL8192E.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/platform/platform_ops.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/core/rtw_mp.o
  CC [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/core/rtw_mp_ioctl.o
  LD [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/8192eu.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/8192eu.mod.o
  LD [M]  /home/ik/Downloads/lm2/rtl8192eu-linux-driver-master/8192eu.ko
make[1]: Leaving directory '/usr/src/linux-headers-4.9.0-5.1-liquorix-amd64'
root@ik--pc:/home/ik/Downloads/lm2/rtl8192eu-linux-driver-master# make install
install -p -m 644 8192eu.ko  /lib/modules/4.9.0-5.1-liquorix-amd64/kernel/drivers/net/wireless/
/sbin/depmod -a 4.9.0-5.1-liquorix-amd64
root@ik--pc:/home/ik/Downloads/lm2/rtl8192eu-linux-driver-master# 
`

I hope the gurus look into it.. Thanks in Advance!!

michaelweghorn, 2017-01-25:
Could you possibly give more details about what is going wrong during the build?

Maybe I am missing something obvious, but for me, the output itself looks OK so far. As far as I can see, there is no error message in it, just a few warnings. The build itself seems to have succeeded and I can see no error message of the "make" or "make install" command.

Did you possibly get any other error messages or something else?

michaelweghorn, 2017-01-25:
I just tried this on Debian stretch/testing which also has Kernel 4.9 at the moment and the above commands worked OK there; the output is similar to the one you posted.

Kernel information:
~~~
$ uname -a
Linux debian 4.9.0-1-amd64 #1 SMP Debian 4.9.2-2 (2017-01-12) x86_64 GNU/Linux
~~~

scdas141, 2017-01-25:
Hi..

Pls see the error messages in Pink/blue in the attached compile screen-shots:

[img]http://i.imgur.com/vZKlbB8.png[/img]

[img]http://i.imgur.com/MQJYvRu.png[/img]

[image]
[image]

michaelweghorn, 2017-01-25:
Thank you for the quick reply.

A distinction between errors and other messages has to be made at this place. The messages by themselves are actually warnings (pink ones) or notes (blue one). I got them as well when compiling.

However, other than errors, warnings and notes do not abort the build. Therefore the compilation itself should have been successful and I would expect that the kernel module was installed in your case. Can you confirm that?

(Certainly, it makes sense to have a closer look why those warnings appear, but according to my understanding, they should not hinder you from building and installing the module so far...)

scdas141, 2017-01-25:
ok.. Here is the output of a successful compile on 4.8.x & older kernels...

`root@ik--pc:/home/ik/Downloads/rtl8192eu-linux-driver-master# make && make install
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.6.0-1-amd64/build M=/home/ik/Downloads/rtl8192eu-linux-driver-master  modules
make[1]: Entering directory '/usr/src/linux-headers-4.6.0-1-amd64'
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/core/rtw_cmd.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/core/rtw_security.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/core/rtw_debug.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/core/rtw_io.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/core/rtw_ioctl_query.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/core/rtw_ioctl_set.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/core/rtw_ieee80211.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/core/rtw_mlme.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/core/rtw_mlme_ext.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/core/rtw_wlan_util.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/core/rtw_vht.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/core/rtw_pwrctrl.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/core/rtw_rf.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/core/rtw_recv.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/core/rtw_sta_mgt.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/core/rtw_ap.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/core/rtw_xmit.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/core/rtw_p2p.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/core/rtw_tdls.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/core/rtw_br_ext.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/core/rtw_iol.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/core/rtw_sreset.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/core/rtw_btcoex.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/core/rtw_beamforming.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/core/rtw_odm.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/core/efuse/rtw_efuse.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/os_dep/osdep_service.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/os_dep/linux/os_intfs.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/os_dep/linux/usb_intf.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/os_dep/linux/usb_ops_linux.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/os_dep/linux/ioctl_linux.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/os_dep/linux/xmit_linux.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/os_dep/linux/mlme_linux.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/os_dep/linux/recv_linux.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/os_dep/linux/wifi_regd.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/os_dep/linux/rtw_android.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/os_dep/linux/rtw_proc.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/hal/hal_intf.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/hal/hal_com.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/hal/hal_com_phycfg.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/hal/hal_phy.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/hal/hal_btcoex.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/hal/hal_hci/hal_usb.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/hal/led/hal_usb_led.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/hal/HalPwrSeqCmd.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/hal/rtl8192e/Hal8192EPwrSeq.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/hal/rtl8192e/rtl8192e_xmit.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/hal/rtl8192e/rtl8192e_sreset.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/hal/rtl8192e/rtl8192e_hal_init.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/hal/rtl8192e/rtl8192e_phycfg.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/hal/rtl8192e/rtl8192e_rf6052.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/hal/rtl8192e/rtl8192e_dm.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/hal/rtl8192e/rtl8192e_rxdesc.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/hal/rtl8192e/rtl8192e_cmd.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/hal/rtl8192e/usb/usb_halinit.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/hal/rtl8192e/usb/rtl8192eu_led.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/hal/rtl8192e/usb/rtl8192eu_xmit.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/hal/rtl8192e/usb/rtl8192eu_recv.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/hal/rtl8192e/usb/usb_ops_linux.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/hal/rtl8192e/rtl8192e_mp.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/hal/OUTSRC/odm_debug.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/hal/OUTSRC/odm_AntDiv.o
[... cut ...]
  LD [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/8192eu.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/ik/Downloads/rtl8192eu-linux-driver-master/8192eu.mod.o
  LD [M]  /home/ik/Downloads/rtl8192eu-linux-driver-master/8192eu.ko
make[1]: Leaving directory '/usr/src/linux-headers-4.6.0-1-amd64'
install -p -m 644 8192eu.ko  /lib/modules/4.6.0-1-amd64/kernel/drivers/net/wireless/
/sbin/depmod -a 4.6.0-1-amd64
root@ik--pc:/home/ik/Downloads/rtl8192eu-linux-driver-master# 
`

michaelweghorn, 2017-01-25:
Thank you for posting that output. 

What is the output of the following command when run in the source directory?

~~~
make && make install ; echo "RETURN VALUE: $?"
~~~

This prints the return value in a line at the end. A return value of 0 means "success", everything else means that something went wrong.

Am I understanding it correctly that the error you are talking about are the purple and blue output produced? What I meant is that those are no real errors, but only warnings (and a note). Errors mean that something went wrong. Warnings are no errors. When (only) a warning is shown, the compilation should still finish successfully. As far as I understand your output, this is what happens in your case. It is what happens in my case as well. However, the module was built successfully and was installed fine.

What is the output of the following command:

~~~
find /lib/modules -name "8192eu.ko"
~~~

This searches for all "installed" kernel modules with the name "8192eu.ko" and it should return a match in a subdirectory for your respective kernel version if everything went well. (As far as I understand the output, this might be `/lib/modules/4.9.0-5.1-liquorix-amd64/kernel/drivers/net/wireless/8192eu.ko` in your case.)

Does your wireless adapter work or do you experience problems with it with kernel 4.9?

While I get the same warnings as you wrote about, the adapter works fine after running "make && make install" - which is what I would expect in case "only" warnings are shown and no "real" error occurs.

(Sorry in case I misunderstand you; I am not sure whether I have really understood the actual problem you are facing...)

PS: The output you posted is a bit hard to read as everything is shown in a single line. Putting the output into code blocks would simplify reading. You can create such a code block e.g. by using `~~~` before posting the actual output and `~~~` at the end again, e.g.

```
~~~
# First line by itself
# second line
# ...
~~~
```

(In order for those `~~~` delimiters to be shown here, I nested them in other code block delimiters.)

scdas141, 2017-01-25:
Hi Michael..

Thanks for helping me out ..

The terminal-output for "make && make install ; echo "RETURN VALUE: $?" got hung for so long that I had to close the terminal window manually.. after waiting for about 10 mins

Also, the find /lib/modules -name "8192eu.ko" lists all three kernels, 4.8.0-11, 4.8.0-15 and 4.9.0-5.. Here's the output:
```
root@ik--pc:/home/ik# find /lib/modules -name "8192eu.ko"
/lib/modules/4.8.0-15.1-liquorix-amd64/kernel/drivers/net/wireless/8192eu.ko
/lib/modules/4.8.0-11.1-liquorix-amd64/kernel/drivers/net/wireless/8192eu.ko
/lib/modules/4.9.0-5.1-liquorix-amd64/kernel/drivers/net/wireless/8192eu.ko
```

Infact, I can see my home network and all other networks but the wifi connection just seems to go on spinning-endlessly without actually connecting..

I really think this is a 4.9.x compatibility issue as every instance of kernel upgrade before 4.9.x, I was able to compile flawlessly and connect succesfully without a single hitch...

Thanks again but no luck for me..

michaelweghorn, 2017-01-25:
Thanks for your reply.

As far as I understand, there is basically no problem during the compilation of the module.

It seems to me like the problem was somewhere else (might be a compatibility issue with the 4.9.x kernel or something else).

I wonder a bit that the `make && make install ; echo "RETURN VALUE: $?"` hung for so long. Did you get the output that you got before for calling `make && make install`? One reason that I could imagine is that a `"` might have been forgotten in the command (in which case the string is not terminated and the shell "waits" for the closing `"`. However, that is probably not relevant here as the compilation itself seems to be OK.

When testing the driver on different machines and in virtual machines, I had the situation that connecting to a  network worked well with the same adapter (and same kernel version, at that point 4.8) on some machines, but not on the others, but I did not really find out what was the relevant difference... (I was using a TP-Link TL-WN821N v5). This may or may not be the same problem you have.

The output of the command `dmesg` or the log file `/var/log/syslog` might contain helpful information for further examining that problem.

Which tool are you using in order to connect to the WLAN (e.g. NetworkManager, wicd,...)?

Unfortunately, I do not now much about the details of the connection management. Maybe somebody with more knowledge in that area can help out here...

scdas141, 2017-01-25:
Hi Michael, Thks again..

You were right... the last " quotation mark was missing, I got a succesful install but same as before, spinning-endlessly without actually connecting..

Here's the output:

```
ik@ik--pc:~/Downloads/lm2/rtl8192eu-linux-driver-master$ su
Password: 
root@ik--pc:/home/ik/Downloads/lm2/rtl8192eu-linux-driver-master# make && make install ; echo "RETURN VALUE: $?"
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.9.0-5.1-liquorix-amd64/build M=/home/ik/Downloads/lm2/rtl8192eu-linux-driver-master  modules
make[1]: Entering directory '/usr/src/linux-headers-4.9.0-5.1-liquorix-amd64'
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: Leaving directory '/usr/src/linux-headers-4.9.0-5.1-liquorix-amd64'
install -p -m 644 8192eu.ko  /lib/modules/4.9.0-5.1-liquorix-amd64/kernel/drivers/net/wireless/
/sbin/depmod -a 4.9.0-5.1-liquorix-amd64
RETURN VALUE: 0
```
Also, var/syslog:

```
Dec 22 23:04:56 ik--pc kernel: RTL871X: +871x_drv - drv_open, bup=1
Dec 22 23:04:56 ik--pc kernel: RTL871X: -871x_drv - drv_open, bup=1
Dec 22 23:04:56 ik--pc kernel: IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Dec 22 23:04:56 ik--pc NetworkManager[543]: <info>  [1482428096.0063] device (wlan0): supplicant interface state: inactive -> disabled
Dec 22 23:04:56 ik--pc NetworkManager[543]: <info>  [1482428096.0178] device (wlan0): supplicant interface state: disabled -> inactive
Dec 22 23:04:56 ik--pc kernel: RTL871X: (2)871x_drv - drv_close, bup=1, hw_init_completed=1
Dec 22 23:04:56 ik--pc wpa_supplicant[664]: ioctl[SIOCGIWSCAN]: Resource temporarily unavailable
Dec 22 23:04:56 ik--pc kernel: RTL871X: indicate disassoc
Dec 22 23:04:56 ik--pc kernel: RTL871X: +rtl8192e_set_FwJoinBssReport_cmd(wlan0): iface_type=0 mstatus(0)
Dec 22 23:04:56 ik--pc kernel: RTL871X: SetHwReg8192E(wlan0) HW_VAR_BASIC_RATE: BrateCfg(0x15d)
Dec 22 23:04:56 ik--pc kernel: RTL871X: -871x_drv - drv_close, bup=1
Dec 22 23:04:56 ik--pc kernel: RTL871X: +871x_drv - drv_open, bup=1
Dec 22 23:04:56 ik--pc kernel: RTL871X: -871x_drv - drv_open, bup=1
Dec 22 23:04:56 ik--pc kernel: IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Dec 22 23:04:56 ik--pc NetworkManager[543]: <warn>  [1482428096.1419] device (wlan0): set-hw-addr: new MAC address 72:AE:02:9F:BE:40 not successfully set (scanning)
Dec 22 23:04:56 ik--pc NetworkManager[543]: <info>  [1482428096.1426] device (wlan0): supplicant interface state: inactive -> disabled
Dec 22 23:04:56 ik--pc wpa_supplicant[664]: ioctl[SIOCGIWSCAN]: Resource temporarily unavailable
Dec 22 23:04:56 ik--pc NetworkManager[543]: <info>  [1482428096.1548] device (wlan0): supplicant interface state: disabled -> inactive
Dec 22 23:04:56 ik--pc kernel: RTL871X: (2)871x_drv - drv_close, bup=1, hw_init_completed=1
Dec 22 23:04:56 ik--pc kernel: RTL871X: indicate disassoc
Dec 22 23:04:56 ik--pc kernel: RTL871X: SetHwReg8192E(wlan0) HW_VAR_BASIC_RATE: BrateCfg(0x15d)
Dec 22 23:04:56 ik--pc kernel: RTL871X: +rtl8192e_set_FwJoinBssReport_cmd(wlan0): iface_type=0 mstatus(0)
Dec 22 23:04:56 ik--pc kernel: RTL871X: -871x_drv - drv_close, bup=1
Dec 22 23:04:56 ik--pc NetworkManager[543]: <warn>  [1482428096.2814] device (wlan0): set-hw-addr: new MAC address 72:AE:02:9F:BE:40 not successfully set (scanning)
Dec 22 23:04:56 ik--pc kernel: RTL871X: +871x_drv - drv_open, bup=1
Dec 22 23:04:56 ik--pc kernel: RTL871X: -871x_drv - drv_open, bup=1
Dec 22 23:04:56 ik--pc kernel: IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Dec 22 23:04:56 ik--pc NetworkManager[543]: <info>  [1482428096.2820] device (wlan0): supplicant interface state: inactive -> disabled
Dec 22 23:04:56 ik--pc wpa_supplicant[664]: ioctl[SIOCGIWSCAN]: Resource temporarily unavailable
Dec 22 23:04:56 ik--pc NetworkManager[543]: <info>  [1482428096.2989] device (wlan0): supplicant interface state: disabled -> inactive
Dec 22 23:04:56 ik--pc kernel: RTL871X: (2)871x_drv - drv_close, bup=1, hw_init_completed=1
Dec 22 23:04:56 ik--pc kernel: RTL871X: indicate disassoc
Dec 22 23:04:56 ik--pc kernel: RTL871X: SetHwReg8192E(wlan0) HW_VAR_BASIC_RATE: BrateCfg(0x15d)
Dec 22 23:04:56 ik--pc kernel: RTL871X: -871x_drv - drv_close, bup=1
Dec 22 23:04:56 ik--pc kernel: RTL871X: +rtl8192e_set_FwJoinBssReport_cmd(wlan0): iface_type=0 mstatus(0)
Dec 22 23:04:56 ik--pc NetworkManager[543]: <warn>  [1482428096.4245] device (wlan0): set-hw-addr: new MAC address 72:AE:02:9F:BE:40 not successfully set (scanning)
Dec 22 23:04:56 ik--pc kernel: RTL871X: +871x_drv - drv_open, bup=1
Dec 22 23:04:56 ik--pc kernel: RTL871X: -871x_drv - drv_open, bup=1
Dec 22 23:04:56 ik--pc kernel: IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Dec 22 23:04:56 ik--pc NetworkManager[543]: <info>  [1482428096.4251] device (wlan0): supplicant interface state: inactive -> disabled
Dec 22 23:04:56 ik--pc wpa_supplicant[664]: ioctl[SIOCGIWSCAN]: Resource temporarily unavailable
Dec 22 23:04:56 ik--pc NetworkManager[543]: <info>  [1482428096.4397] device (wlan0): supplicant interface state: disabled -> inactive
Dec 22 23:04:56 ik--pc kernel: RTL871X: (2)871x_drv - drv_close, bup=1, hw_init_completed=1
Dec 22 23:04:56 ik--pc kernel: RTL871X: indicate disassoc
Dec 22 23:04:56 ik--pc kernel: RTL871X: SetHwReg8192E(wlan0) HW_VAR_BASIC_RATE: BrateCfg(0x15d)
Dec 22 23:04:56 ik--pc kernel: RTL871X: -871x_drv - drv_close, bup=1
Dec 22 23:04:56 ik--pc kernel: RTL871X: +rtl8192e_set_FwJoinBssReport_cmd(wlan0): iface_type=0 mstatus(0)
[... cut ...]
Dec 22 23:05:02 ik--pc kernel: RTL871X: indicate disassoc
Dec 22 23:05:02 ik--pc kernel: RTL871X: SetHwReg8192E(wlan0) HW_VAR_BASIC_RATE: BrateCfg(0x15d)
Dec 22 23:05:02 ik--pc kernel: RTL871X: -871x_drv - drv_close, bup=1
Dec 22 23:05:02 ik--pc kernel: RTL871X: +rtl8192e_set_FwJoinBssReport_cmd(wlan0): iface_type=0 mstatus(0)
```
I am using NetworkManager "nm-connection-editor", as linux has by tradition very well supported on older hardware, I hope the developer and other gurus will look into it..

Thanks again..

thamer, 2017-01-27:
This might be related, but I noticed after booting kernel 4.9 that 8192eu.ko is not being used by default. Instead, the kernel was using rtl8xxxu (see the output of `lshw -C network`). My USB adapter (2357:0109) partially worked using rtl8xxxu, but with bad perfomance and required manually reattaching the adpater after reboot. 

As a workaround, I added "8192eu" to `/etc/modules` and now its working fine. 

hth,
Thamer

scdas141, 2017-01-27:
oh-my-god.. Thanks a million to thamer..

Just adding "8192eu" via text editor to "/etc/modules" did the trick..

Thanks again!!

closed 2017-01-27 by scdas141

jagmeethanspal, 2019-06-30:
Old thread, but can you tell why does it download firmware?
"RTW: FirmwareDownload8192E fw:NIC, size:xxx", and it happens every boot and sometimes while the system is running too thereby rebooting the TPLink device. Isnt the TPLink having its firmware in the ROM, and this is just a linuc driver?