Mange/rtl8192eu-linux-driver #19: Please add kernel 4.9 compatibility

zackyjack, 2017-02-07:
I just upgrade my kernel to 4.9, and my wifi adapter can't work. please add compatibility for kernel 4.9.

Many thanks

Mange, 2017-02-07:
Patches welcome. Open a PR with the necessary code changes and it will most
likely be merged right away.

michaelweghorn, 2017-02-07:
In general, this should work with kernel 4.9.
Can you give more details about the problem you are encountering?

Is this possibly the same as described in #18 ?

In that issue, the problem was that the kernel module was not used. Everything worked fine after explicitly adding it to `/etc/modules`, s. https://github.com/Mange/rtl8192eu-linux-driver/issues/18#issuecomment-275705282. This was also with kernel 4.9.

CillianCritchley, 2017-02-10:
i upgraded to 4.9 this morning and my wn822n v4EU didn't work either, already had 8192eu in /etc/modules 

back on 4.4 now and all working well.

i removed 4.9 after but I'll happily install it again if anyone knowledgeable wants me to test some stuff out to see why it isn't working

scdas141, 2017-02-15:
Probably many ppl forgetting that they need to recompile - `"sudo make && make install"` after upgrading the kernel..

I have had no issues, flawless performance since adding 8192eu to `/etc/modules`.. Also any major software install like wine etc needs a drive re-compile..

Hope-this-helps..

CillianCritchley, 2017-02-18:
but I didn't do that the first time I install it, I just used dkms
and the main page says that the module will automatically be rebuilt from source when a kernel is upgraded if I use dkms

scdas141, 2017-02-18:
Hey Cillian..

Every time you upgrade the kernel, you will definitely have to re-compile & re-install as the compilation itself is for a particular "uname -r" lib/bin/module..

The ONLY instance where you don't need to re-compile the drivers are if you yourself compiled the kernel after downloading a RAW kernel-snapshot from kernel.org.. You may be able to specify any custom driver modules you want installed.. That way all the drivers get built into the kernel itself..

That is probably why Linux Mint has the "Don't Break My PC" option built-in, which freezes all automatic updates..

[http://cdn.ghacks.net/wp-content/uploads/2016/06/linux-mint-18-update-manager.png](url)

michaelweghorn, 2017-02-18:
If DKMS works correctly, it should not be necessary to rebuild the kernel module manually but DKMS should do that automatically.

You can check whether the module has been built by using the following command:

    find /lib -name 8192eu.ko

In case it has been built, there should be a result for the module in a directory for your kernel version, e.g. `/lib/modules/4.9.0-1-amd64/updates/dkms/8192eu.ko`.

The following command helps to find out whether the module is actually loaded:

    lsmod | grep 8192

This could help to find out whether the kernel module itself has been built and is loaded. Of course, trying to build the module manually, as suggested in https://github.com/Mange/rtl8192eu-linux-driver/issues/19#issuecomment-280839583 is also a good way to narrow down the cause of the issue. (But in case everything with DKMS works as intended, this should not be necessary...)

CillianCritchley, 2017-02-18:
if I remember right, there was an 8192eu.ko in the 4.9 kernel directory but it was a week ago now so maybe I'm misremembering.  I'll give it another shot later today or tomorrow morning and will update. thanks all.

CillianCritchley, 2017-02-18:
well when you're wrong you're wrong. 
haven't a notion what was different this time than the last but it's working great now in 4.9
thanks again lads

michaelweghorn, 2017-02-18:
Thank you for that information. Since the problem is gone, I'll close this issue.

closed 2017-02-18 by michaelweghorn