Mange/rtl8192eu-linux-driver #21: tp-link usb wifi installation on Kali linux 4.6.0

etamaro, 2017-03-02:
Hi Gurus,

I was searching for hours and found ton of how-to's to compile and install driver for TL-WN821N v5 on Debian Kali linux. The problem is that tp-link source is supporting linux up to kernel 3.10.10 and my kali version is 4.6.0
http://www.tp-link.com/us/download/TL-WN821N.html#Driver
I couldn't find anything on how to get around it and install the drivers.
Can anyone give me a hand?

Cheers

michaelweghorn, 2017-03-02:
Hi,

installing the firmware as described in the README of this repository should work with kernel 4.6 as well, as modifications to the original TP-Link source have been made to provide compatibility with newer kernel versions.

Please let us know in case you are encountering problems.

michaelweghorn, 2017-03-20:
As the original question has been answered and no further problems have been reported, I am closing this issue. Please reopen or report a new one in case you are still having difficulties compiling and installing this driver.

closed 2017-03-20 by michaelweghorn

4te3, 2018-05-05:
hi please someone can help please cuz i got a tp link tl-wn8200nd v2.0 with driver rtl8192eu v 4.4.1.1 i want to know how to installing on kali linux please