Mange/rtl8192eu-linux-driver #22: can't connect to a any network

munkaka, 2017-03-05:
Hello, I've just installed your driver on kali 4.6.0. It recognizes the wireless adapter (tp-link tl-wn823n v2) and finds the networks. But it fails to connect to all of them. I've tried the same on ubuntu 16.10 and have exactly the same issue. Both, Kali and Ubuntu are virtual box machines, is that the problem? If not, do you know any solution? Thanks

michaelweghorn, 2017-03-05:
> Both, Kali and Ubuntu are virtual box machines, is that the problem? If not, do you know any solution?

In general, using the driver in a virtual machine should work when the USB device is properly redirected to the virtual machine (which seems to be the case as wireless networks are being detected).

Can you verify that this driver is actually used by the device? The output of the command `lshw -C network` should show this.

The output of the command `dmesg` or the log file `/var/log/syslog` might contain helpful information for further examining that problem. Can you possibly attach (all or part of) them?

Which tool are you using in order to connect to the WLAN (e.g. NetworkManager, wicd,...)?

munkaka, 2017-03-06:
Hi,
it has connected a couple of times today, but lost the connection in a couple of minutes. And even when connected it was extremely slow. I have to say that my host machine is quite old, its a MacBook Pro mid 2009.  But I don't think its that the problem, since its working fine with a couple of upgrades during the years. In any case, attached you find the outputs of the commands you asked for. These are from kali. In order to connect I'm using the default tool of Kali, in the upper bar.
Thanks :)

[lshw.txt](https://github.com/Mange/rtl8192eu-linux-driver/files/822516/lshw.txt)
[out_dmesg.txt](https://github.com/Mange/rtl8192eu-linux-driver/files/822492/out_dmesg.txt)
[syslog.txt](https://github.com/Mange/rtl8192eu-linux-driver/files/822493/syslog.txt)

michaelweghorn, 2017-03-13:
Thank you very much for providing further information.
I tried to reproduce the issue with Ubuntu 16.10 in a virt-manager (KVM) virtual machine on a Debian stretch/testing host, but connecting to WLAN networks worked find there with a TP-LINK TL-WN821N v5.

I am currently pretty much out of ideas but maybe somebody else has an idea?

michaelweghorn, 2017-03-13:
I also tested with a virtual machine running Kali Linux now and it worked well for me there as well. As far as I can see, you were running kernel 4.6 when you collected the above attachments, while Kali now has kernel 4.9. Could you possibly check whether the problem is still there after upgrading (`sudo apt-get update && sudo apt-get dist-upgrade`) and rebooting? Depending on the method you chose to install the kernel modules for this driver, you might have to rebuild them after kernel upgrade.

Note: `uname -a` shows the version of the kernel currently running.

munkaka, 2017-03-14:
Hi, thanks very much for the help. I've tried with kali 4.9 as well but nothing....
Forget it, I've bought a tp-link tl-wn722n and it works very well without installing anything.
Thank you in any case and sorry for taking your time in vain.

michaelweghorn, 2017-03-14:
You're welcome. It is sad that it is not working with the other adapter.
However, as far as I understand your message, this issue can be closed now.

closed 2017-03-14 by michaelweghorn