Mange/rtl8192eu-linux-driver #24: Support Linux 4.10

rudra123, 2017-03-31:
I have been using this patch since Linux 4.7 and it has worked flawlessly ever since. I was wondering if support for Linux 4.10 would be added as it is the default kernel in the upcoming Ubuntu 17.04?

Mange, 2017-04-02:
If you have the necessary code changes on hand, they would be very welcome as a Pull Request.

michaelweghorn, 2017-04-02:
@rudra123: Can you please describe more precisely what your problems with kernel 4.10 are? I tested compiling the kernel module on a beta version of Ubuntu 17.04 (which has kernel 4.10) and the compilation was working OK for me. (I did not test beyond the build itself.)

From your initial post, I am not exactly sure whether that was just a question whether kernel 4.10 is supported or you are currently experiencing a specific problem, so it would be nice if you could clarify that.

rudra123, 2017-04-02:
Sorry for the ambiguity. It was just a question regarding the support. I have not tried it on 17.04 myself yet. Anyway, thanks a lot for the reply.

michaelweghorn, 2017-04-02:
Thank you for your quick reply and for clarifying this. From what I know so far, this should work, s. my previous comment. Please open a new issue in case you should encounter any problems.

closed 2017-04-02 by michaelweghorn

Lenseth, 2017-04-04:
4.10 kernel the module compiles but the driver doesn't work for some reason
4.8 kernel the module compiles and the driver works