Mange/rtl8192eu-linux-driver #29: Wifi disconnects every few minutes on Linux 4.8.0-48

rudra123, 2017-04-20:
Ever since the upgrade to Linux 4.8.0-48, the USB Wifi adapter disconnects every few minutes. So I have to re-plug in the device for it to work.
Note: There is no problem with the network itself as the in-built adapter and even other devices stay connected to the network.

nomadmzl, 2017-04-23:
Check if there was any kernel update: 

uname -r
sudo dkms status
Compare the versions for rtl8192eu driver, if the driver version is different from the kernel one, recompile the driver for the new kernel:

sudo dkms install -m rtl8192eu -v 1.0
Mount your device: modprobe 8192eu

Restart linux, unplug and plug your dongle (just one time, you won't need it again).

rudra123, 2017-04-24:
The versions are the same. Still i recompiled it but to no effect.
[image]

nomadmzl, 2017-04-24:
It seems that your device is the CU versión, I don't know if it is compatible. In example, for the 8192EU device you can't use the CU drivers, (I've tried, it doesn't load). Try the CU drivers posted here: [8192cu](https://github.com/search?utf8=%E2%9C%93&q=8192cu&type=)

nomadmzl, 2017-05-14:
If that solved your issue, you could considere to close this post.

PanderMusubi, 2018-06-04:
@rudra123 is this still an issue for you? Otherwise please close this issue.

closed 2018-06-04 by CGarces