Mange/rtl8192eu-linux-driver #30: Upload speed limited to 1 Mb/s

nomadmzl, 2017-04-23:
Perhaps RTL8192eu supports bit rates up to 54 Mb/s, when connected it only shows 1 Mb/s in LAN connection.

Using Internet using www.speedtest.net I've checked connecting via ethernet and I got 2 Mb/s upload (accoding with my Internet plan), but when I change to the wireless connection (USB dongle RTL8192eu), I only get 0,8 Mb/s.

Download speed has no issues, it can go up to 10Mb/s so far (may be more with a better internet service).

nomadmzl, 2017-05-07:
SOLVED!

The issue is that perhaps you've installed the 8192eu driver (from any source), the kernel 4.8+ doesn't  load it, instead it use a generic rtl8xxxu (that came with it). This generic driver has issues with connection stability, slow speed and other things reported in the internet. I've solved it in this way:

After installing the driver from Mange's git:

1. Using any file manager and in a terminal (with sudo), open a window and browse to "etc/modules-load.d"  
2. Look for the file "modules.conf", add the line: 8192eu
3. Restart

Done! now the kernel will load the 8192eu modele at boot instead the rtl8xxxu, you can check that running lshw -c network in a terminal and checking something like this:
[image]

The 8192eu driver is being used in the wireless conection. Now I don't to unplug and re-plug the driver to recover my connection, also the bit rate boosted from 1 Mb/s to 144 Mb/s (I've checked in speedtest.net and the upload speed was recovered at normal levels as a previously wired connection test showed it could go).

closed 2017-05-07 by nomadmzl

NhlalukoG, 2017-09-24:
Well here is the funny about all that...
From Mange's source and basically all other sources for 8192eu state that monitor mode and other custom modes are not supported except for Managed,Master and Ad-Hoc. That is entirely false. On the RTL8xxxu all those modes work. I'm on an antenna based dongle so RLT8xxxu isnt configured for that.

On os_dep/linux/ioctl_linux.c those modes were intentionally removed and on the rest of the source, for what good reason i dont know. If u modify one of the cases from line 1465 and rename it to monitor instead and use iwconfig **** mode monitor, it does recognise just not able to apply since the mode is missing entirely.

bottom line, the chipset supports all these fancy modes but the source does not.