Mange/rtl8192eu-linux-driver #31: Updated driver for RTL8192eu: v4.3.8_12406.20140929

nomadmzl, 2017-04-23:
I've found an updated driver newer than v4.3.1.1_11320.20140505: 

The driver  [rtl8192EU_linux_v4.3.8_12406.20140929.tar.gz](https://github.com/Mange/rtl8192eu-linux-driver/files/950085/rtl8192EU_linux_v4.3.8_12406.20140929.tar.gz) features:

1. Support Adaptivity (v7.1) 
2. Fix efuse write failed problem.
3. Improve performance of Soft-AP mode.
4. Adjust driver procedure to eliminate RF interference.
5. Clear up all compile warning.
6. Fix IOT issue with some AP’s RA mechanism

I've followed this guide to compile it, but I couldn't get it done: https://scdas141.wordpress.com/2017/01/28/how-to-compile-mangertl8192eu-linux-driver-driver/

if someone (with more compiling skills than me) could give it a try...

CGarces, 2017-04-24:
Where is the changelog? and the source of that file?

nomadmzl, 2017-04-25:
You're right, here it goes. I've found the driver in a Comfast installation CD, there is also the WPA_supplicant folder, I suppose that is is already included in the driver that you've compiled.

[ReleaseNotes.pdf](https://github.com/Mange/rtl8192eu-linux-driver/files/953528/ReleaseNotes.pdf)

CGarces, 2017-04-30:
Please @nomadmzl try to compile from my git and confirm if is working fine.
Maybe @Mange will be interested on add a new branch if users confirm that works fine, but I don't have the proper hardware to test.

nomadmzl, 2017-05-01:
Hi, I'm trying to compile it, but I get:

se entra en el directorio '/usr/src/linux-headers-4.8.0-49-generic'
  CC [M]  /rtl8192eu-linux-driver/core/rtw_cmd.o
In file included from /rtl8192eu-linux-driver/include/drv_types.h:95:0,
                 from /rtl8192eu-linux-driver/core/rtw_cmd.c:22:
/rtl8192eu-linux-driver/include/hal_com.h:385:13: error: ‘file_path’ redeclared as different kind of symbol
 extern char file_path[PATH_LENGTH_MAX];
             ^
In file included from ./include/linux/seq_file.h:10:0,
                 from ./include/linux/pinctrl/consumer.h:17,
                 from ./include/linux/pinctrl/devinfo.h:21,
                 from ./include/linux/device.h:24,
                 from ./include/linux/dmaengine.h:20,
                 from ./include/linux/netdevice.h:38,
                 from /rtl8192eu-linux-driver/include/osdep_service_linux.h:35,
                 from /rtl8192eu-linux-driverinclude/osdep_service.h:41,
                 from /rtl8192eu-linux-driver/include/drv_types.h:32,
                 from /rtl8192eu-linux-driver/core/rtw_cmd.c:22:
./include/linux/fs.h:2712:14: note: previous declaration of ‘file_path’ was here
 extern char *file_path(struct file *, char *, int);
              ^
scripts/Makefile.build:289: fallo en las instrucciones para el objetivo '/rtl8192eu-linux-driver/core/rtw_cmd.o'
make[2]: *** [/rtl8192eu-linux-driver/core/rtw_cmd.o] Error 1
Makefile:1491: fallo en las instrucciones para el objetivo '_module_/rtl8192eu-linux-driver'
make[1]: *** [_module_/rtl8192eu-linux-driver] Error 2
make[1]: se sale del directorio '/usr/src/linux-headers-4.8.0-49-generic'
Makefile:1468: fallo en las instrucciones para el objetivo 'modules'
make: *** [modules] Error 2

CGarces, 2017-05-01:
Anda... estas compilando en español :+1: 
I have tested the compilation with the Libreelec toolchaim but obviously I have miss something... probably [this ](https://github.com/Mange/rtl8192eu-linux-driver/commit/89e79bf30ac03d905fbfbb9d4ea0d79f8f9f60fc)

I'll try to reproduce your error and fix it, but will be delayed some time (3 weeks), I'm outside home

nomadmzl, 2017-05-02:
Si, soy de Suramérica.

I'll test the code again when checked, let me know if I need any module installed in linux in order to compile the code properly. Bye!

CGarces, 2017-05-03:
@nomadmzl Please ignore this thread, looks like there is a newer version, so not made sense spend time with my v4.3.8_12406.20140929 branch if you can test directly  v.4.4.1.
Please go to https://github.com/Mange/rtl8192eu-linux-driver/issues/10#issuecomment-298852261 and consider close this issue.

nomadmzl, 2017-05-03:
Ok

closed 2017-05-03 by nomadmzl