Mange/rtl8192eu-linux-driver #33: not working under ubuntu 17.04 desktop x64

psndna88, 2017-05-06:
compiles fine for 4.10.20 kernel of ubuntu desktop gnome 17.04 x86_64, its module is loaded and usb device detected by gnome network manager, but i see no light and it cannot connect to any network,,

nomadmzl, 2017-05-07:
Hi, there are some things to check:

1. Try to unplug and re-plug the device. 
2. Check if the module is loaded: sudo dkms status
3. If you don't see the module there, so run: sudo depmod -a
4. Reboot or reload the service: sudo service network-manager restart

You may see two drivers: rtl8xxxu or 8192eu or both. The rtl8xxxu is the generic driver compatible with some realtek chips that comes inside the kernel, it has stability, speed issues and other problems so we have to force the kernel to load instead the 8192eu driver, lets check which one is using your wireless connection: 

lshw -c network

And past here the results.

psndna88, 2017-05-09:
*-network
       description: Wireless interface
       physical id: 2
       bus info: usb@1:1.4
       logical name: wlx18d6c71aff12
       capabilities: ethernet physical wireless
       configuration: broadcast=yes driver=**rtl8xxxu** driverversion=4.10.0-20-generic firmware=N/A link=no multicast=yes wireless=IEEE 802.11

rtl8xxxu is being used.... instead of 8192eu

nomadmzl, 2017-05-09:
Try opening this from a terminal using sudo:

`etc/modules-load.d`, look for `modules.conf` and add the line: `8192eu`

So reboot, run again lshw -c network and you should see `driver=8192eu`

psndna88, 2017-05-09:
I got it working by adding rtl8xxxu to modprobe blacklist and restarting network manager service
Thanks...

  *-network
       description: Wireless interface
       physical id: 2
       bus info: usb@1:1.4
       logical name: wlx18d6c71aff12
       capabilities: ethernet physical wireless
       configuration: broadcast=yes driver=rtl8192eu ip=192.168.100.3 multicast=yes wireless=IEEE 802.11bgn

closed 2017-05-09 by psndna88