Mange/rtl8192eu-linux-driver #36: Create new branch for v4.4.1

CGarces, 2017-05-23:
The master branch is a little bit outdated, based on v4.3.1.1_11320.20140505

I have merged all your improvements into the v4.4.1 version from realtek.

Can you please create a new branch in order to send a PR from [my branch](https://github.com/CGarces/rtl8192eu-linux-driver/tree/v4.4.1)?

The code comes from [this tarball](https://github.com/masterzorag/RTL8192EU-linux/blob/master/tarballs/rtl8192EU_linux_v4.4.1_17696.20160509.tar.xz)

This is the changelog based on the [release notes](https://github.com/masterzorag/RTL8192EU-linux/blob/master/Linux_Release_notes.pdf):

+ **v4.3.8_12406.20140929**
 1. Support Adaptivity (v7.1)
 2. Fix efuse write failed problem.
 3. Improve performance of Soft-AP mode.
 4. Adjust driver procedure to eliminate RF interference.
 5. Clear up all compile warning.
 6. Fix IOT issue with some AP’s RA mechanism.

+ **v4.3.15_14701.20150715_BTCOEX20150615-41**
1. This driver can support 8192EU and 8192EU+8761AW
2. Support BT coexist (version BTCOEX20150615-41)
3. Support Linux kernel 4.0
4. Support Android 5.0
5. Support High power feature
6. Support Sniffer feature (linux)
7. Update Adaptivity (v8.4)
8. Fix legacy AP connect issue
9. Fix P2P listen state setting listen channel issue

+ **v4.3.15.2_16504.20160106_BTCOEX20150615-41**
1. Support efuse 5.7.3
2. Support monitor mode
3. Fix Adaptivity IPS issue
4. Fix WEXT concurrent mode issue

+ **v4.4.1_17696.20160509_BTCOEX20160412-0042**
1. Support Linux Kernel 4.4.6
2. Support Android 6.0
3. Support TDLS
4. Support proc current_tx_rate
5. Support proc tx_info_msg /rx_info_msg
6. Update BT coexist (version BTCOEX20160412-0042)
7. Update Adaptivity (v9.0)
8. Fix wpa_supplicant disconnect ignore issue
9. Fix P2P logo issue

Compilation (I don't have real hardware) was tested by me with

- Debian 8 stable with kernel 3.16 and GCC 4.8
- LibreElec 7 kernel 3.14 and GCC 5.3.1
- LibreElec 9 kernel 4.10 and 4.11. GCC 6.2.0

Note that LibreElec has one [patch for GCC](https://github.com/LibreELEC/LibreELEC.tv/blob/master/packages/linux-drivers/RTL8192EU/patches/RTL8192EU-gcc-5.patch) not included in the repository.

The code has been tested with real hardware by @nomadmzl, @masterzorag, @gilcu3  with several kernels, like 4.4.9, 4.8 or 4.11

Please see https://github.com/CGarces/rtl8192eu-linux-driver/issues/1 for details

Mange, 2017-05-24:
I don't need to create a branch in order for you to send a PR. Just browse to your branch in your fork and press "Pull Request". 

Sounds like a really good patch. I look forward to it. I don't have the hardware either, so I'll have to trust your testers too.

CGarces, 2017-05-24:
I can send a PR over your master, but not over a non existing branch.

Mange, 2017-05-27:
Is there a reason not to merge it to master? Does it break the driver for some users?

CGarces, 2017-05-27:
I have tested the compilation with old kernels and old GCC versions, looks backguard compatible.
There is no specific reason to not merge to master.
Maintain 2 branches during some time maybe is more safe while we receive feedback from users with the real hardware.
In any case, I have send the PR  #37 right now to master.

CGarces, 2017-07-06:
Fixed by https://github.com/Mange/rtl8192eu-linux-driver/pull/37

closed 2017-07-06 by CGarces