Mange/rtl8192eu-linux-driver #40: Fix build regression due to netdev destructor fix.

CGarces, 2017-07-06:
I need some free time to check this regresion bug.
https://github.com/masterzorag/RTL8192EU-linux/pull/2/commits/0a294fae65b3505d2a4e298c7d80254b9dd3c24b

More info at 
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git/commit/?h=v4.11.9&id=ed66e50d9587fc0bb032e276a2563c0068a5b63a
https://patchwork.kernel.org/patch/9773757/

masterzorag, 2017-07-06:
Seems just a struct change in 4.11.9, so we just include the right definition by conditioning.

Currently testing on gentoo x86_64, no problem in a few hours...

CGarces, 2017-07-06:
Fixed by 7203537add3407738cb69fef625dd68214aee927

masterzorag, 2017-07-07:
hmmm, not!
@CGarces, please recheck [my commit](https://github.com/masterzorag/RTL8192EU-linux/commit/0a294fae65b3505d2a4e298c7d80254b9dd3c24b)

CGarces, 2017-07-07:
I have tested the compilation and looks ok.

Is based in the similar patch done by [Libreelec 9](https://github.com/LibreELEC/LibreELEC.tv/pull/1629) for all out-of-the-tree WiFi drivers., for example https://github.com/lwfinger/rtl8192du/pull/48/files

Change at `include/osdep_intf.h` is just cosmetic so I have ignored it.

About os_dep/linux/ioctl_cfg80211.c, I read [this](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git/commit/include/linux/netdevice.h?id=cf124db566e6b036b8bcbe8decbed740bdfac8c6
):
> netdev->priv_destructor() performs all actions to free up the private
resources that used to be freed by netdev->destructor(), except for
free_netdev().

So I assume, **but maybe I'm wrong**, that  the old `mon_ndev->destructor = rtw_ndev_destructor` must be replaced by `mon_ndev->priv_destructor = rtw_ndev_destructor` and `mon_ndev->needs_free_netdev` to works exactly in the same way and the old code.

But maybe the old code is incorrect and your code is right, `priv_destructor` is not really needed at that point, only `needs_free_netdev=true` fixing some bug, but that is out my knowledge. I just made the change just to keep the code working in the same way.

Can you clarify why `priv_destructor` is not needed at that point of the code?

masterzorag, 2017-07-07:
Ack, thanks for pointing out, need further testing.

closed 2017-07-07 by CGarces