Mange/rtl8192eu-linux-driver #41: D-Link DWA-131 rev E1 is not supported in default branch (realtek-4.4.x)

serg-z, 2017-07-10:
> Drivers for the rtl8192eu chipset for wireless adapters **(D-Link DWA-131 rev E1 included!)**

Turns out USB_DEVICE(0x2001, 0x3319) was removed in 9dde4572b40e248bb41b56d67857bcb01552eeea

Mange, 2017-07-10:
If someone could either backport whatever added support for it that would
be excellent. Otherwise we'll have to rewrite the README to point to master
for that device instead.

CGarces, 2017-07-10:
I'll work on that

rnavene, 2017-07-17:
Patch provided by @CGarces worked for me. I used it for installing 8192eu driver for D-link DWA131 in linux mint 18.2 (Kernel version - 4.8.0-53-generic)

CGarces, 2017-07-17:
Thanks @rnavene for your feedback, patch merged.

closed 2017-07-17 by CGarces

ADITYASHAH2310, 2020-03-05:
@navene and @CGarces  sir i have D-Link DWA-131 Wireless N Nano USB Adapter(rev.E) RTL8192eu and i want to use it in virtual box under linux environment.i am able to all stuffs like monitor mode ,airomon-ng and deauthentiaation but when i need to connect to wifi under linux environment (not using NAT network),then its not connecting to network.i have not downloaded any driver for it.can you guide me the steps for correcting it. As i am electrical student ,i am new to linux and its terminal command ,so if you can guide me for the commadns then it will be a great hel to me

PR #42 (2017-07-17): Add D-Link DWA-131
Reintroduce USB_DEVICE(0x2001, 0x3319)  removed in 9dde457
Close #41