Mange/rtl8192eu-linux-driver #43: Cannot install rtl8192-eu-linux-driver-master on Kernel  4.9.35

GillesZo, 2017-07-31:
Hello, 

After some reading and check my issue with existing post, unfortunatly i cannot install it.

Here is the output:

```
make && make install ; echo "RETURN VALUE: $?"
make: Warning: File 'Makefile' has modification time 64283 s in the future
make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/4.9.35-v7+/build M=/home/pi/rlt/rtl8192eu-linux-driver-master  modules
make[1]: *** /lib/modules/4.9.35-v7+/build: No such file or directory.  Stop.
Makefile:1335: recipe for target 'modules' failed
make: *** [modules] Error 2
RETURN VALUE: 2

```

Mange, 2017-07-31:
Is the clock on your machine set up correctly? Seems like it might be stuck
in the past, when looking at that warning.

GillesZo, 2017-07-31:
Thank's for your quick answer, i will test it tomorrow and will let you know the result

zaidiasyraf, 2017-08-01:
Greeting sir, I am facing the same problem, I tried changing the clock but it still produce the same output.

PanderMusubi, 2017-08-31:
I have a similar problem on Raspberry Pi with Raspbian stretch

    root@XXX:/home/pi/rtl8192eu-linux-driver# dkms add .
    Error! DKMS tree already contains: rtl8192eu-1.0
    You cannot add the same module/version combo more than once.
    root@XXX:/home/pi/rtl8192eu-linux-driver# dkms install rtl8192eu/1.0
    
    Kernel preparation unnecessary for this kernel.  Skipping...
    
    Building module:
    cleaning build area...
    'make' all KVER=4.9.41-v7+....(bad exit status: 2)
    Error! Bad return status for module build on kernel: 4.9.41-v7+ (armv7l)
    Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.
    root@XXX:/home/pi/rtl8192eu-linux-driver# git branch
    * realtek-4.4.x
    root@XXX:/home/pi/rtl8192eu-linux-driver# uname -a
    Linux XXX 4.9.41-v7+ #1023 SMP Tue Aug 8 16:00:15 BST 2017 armv7l GNU/Linux
    root@XXX:/home/pi/rtl8192eu-linux-driver# dkms install rtl8192eu/1.0
    
    Kernel preparation unnecessary for this kernel.  Skipping...
    
    Building module:
    cleaning build area...
    'make' all KVER=4.9.41-v7+....(bad exit status: 2)
    Error! Bad return status for module build on kernel: 4.9.41-v7+ (armv7l)
    Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.

PanderMusubi, 2017-08-31:
Workaround to get the build going is the following:

    cd /usr/src/linux-headers-4.9.41-v7+
    sudo ln -s arm armv7l

At the moment looking into this if that results in a proper build.

PanderMusubi, 2017-08-31:
$ sudo dkms install rtl8192eu/1.0
    
    Kernel preparation unnecessary for this kernel.  Skipping...
    
    Building module:
    cleaning build area...
    'make' all KVER=4.9.41-v7+.........................................................................................................................................................
    cleaning build area...
    
    DKMS: build completed.
    
    8192eu.ko:
    Running module version sanity check.
     - Original module
       - No original module exists within this kernel
     - Installation
       - Installing to /lib/modules/4.9.41-v7+/kernel/drivers/net/wireless//
    
    depmod....
    
    Warning: Unable to find an initial ram disk that I know how to handle.
    Will not try to make an initrd.
    
    DKMS: install completed.

PanderMusubi, 2017-08-31:
the symbolic link seems to fix it. Could someone please make patch for this so that linking is not needed?

CGarces, 2017-08-31:
Hi @PanderMusubi
I need to understand and reproduce the error the problem first. 
In the mean time can you please, change the Makefile and set `CONFIG_PLATFORM_I386_PC= n` and `CONFIG_PLATFORM_ARM_RP = y`?
Please compile and install dkms thing.

PanderMusubi, 2017-09-01:
That works, the result is:

    pi@raspberrypi:~/rtl8192eu-linux-driver $ sudo dkms add .
    
    Creating symlink /var/lib/dkms/rtl8192eu/1.0/source ->
                     /usr/src/rtl8192eu-1.0
    
    DKMS: add completed.
    pi@raspberrypi:~/rtl8192eu-linux-driver $ sudo dkms install rtl8192eu/1.0
    
    Kernel preparation unnecessary for this kernel.  Skipping...
    
    Building module:
    cleaning build area...
    'make' all KVER=4.9.41-v7+..........................................................................................................................................................................
    cleaning build area...
    
    DKMS: build completed.
    
    8192eu.ko:
    Running module version sanity check.
     - Original module
       - No original module exists within this kernel
     - Installation
       - Installing to /lib/modules/4.9.41-v7+/kernel/drivers/net/wireless//
    
    depmod....
    
    Warning: Unable to find an initial ram disk that I know how to handle.
    Will not try to make an initrd.
    
    DKMS: install completed.

I did this on another system with a clean installation.

What is way forward? Add this to the documentation or improve the Makefile so it will work out of the box?

CGarces, 2017-09-04:
Hi @PanderMusubi 

I'm not and expert, but I have check other MakeFiles from similar projects and all looks very similar.

Other README.md files just document the required changes to build it for PRI

https://github.com/gnab/rtl8812au/blob/master/README.md
https://github.com/abperiasamy/rtl8812AU_8821AU_linux/blob/master/README.md

So I have done the same at #54, don't hesitate to ping me if you find a better solution.

closed 2017-09-04 by CGarces

PanderMusubi, 2017-09-04:
No idea how to do that automatically Could you please add the needed information to the README.md to build on RPi?

PanderMusubi, 2017-09-04:
No idea how to do that automatically Could you please add the needed information to the README.md to build on RPi?

Ah, see you have already added it. Thanks.