Mange/rtl8192eu-linux-driver #44: Can'

Andrew3019, 2017-08-02:


Mange, 2017-08-03:
Find the specific package name for the kernel headers for your distribution and install them instead of the "linux-headers-generic" package.

Andrew3019, 2017-08-04:
Sorry, I'm a noob at Linux. How exactly do you do that? 
I'm trying to install the driver for my TP-LINK WN823N, which uses the rtl8192eu Linux driver.
It's a raspberry pi 2b, that's connected to Ethernet right now.

Thanks

nomadmzl, 2017-08-04:
Try with it: 

sudo apt-get install linux-headers-$(uname -r) build-essential dkms git
sudo git clone https://github.com/Mange/rtl8192eu-linux-driver.git
sudo dkms add ./rtl8192eu-linux-driver
sudo dkms install rtl8192eu/1.0
sudo depmod -a

It usually works for me.

Andrew3019, 2017-08-10:
`pi@rpi:~ $ sudo apt-get install linux-headers-$(uname -r) build-essential dkms git
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package linux-headers-4.9.35-v7
E: Couldn't find any package by regex 'linux-headers-4.9.35-v7'
pi@rpi:~ $ sudo git clone https://github.com/Mange/rtl8192eu-linux-driver.git
Cloning into 'rtl8192eu-linux-driver'...
remote: Counting objects: 1135, done.
remote: Total 1135 (delta 0), reused 0 (delta 0), pack-reused 1135
Receiving objects: 100% (1135/1135), 3.90 MiB | 251.00 KiB/s, done.
Resolving deltas: 100% (606/606), done.
Checking connectivity... done.
pi@rpi:~ $ sudo dkms add ./rtl8192eu-linux-driver
sudo: dkms: command not found
pi@rpi:~ $ sudo dkms add
sudo: dkms: command not found
pi@rpi:~ $ dkms
-bash: dkms: command not found`

Not working :(

nomadmzl, 2017-08-10:
Usually is a depencies issue, try running: `sudo apt-get -f install` and then try again sudo `apt-get install dkms`.

Look here: https://stackoverflow.com/questions/22165929/install-linux-headers-on-debian-unable-to-locate-package

PanderMusubi, 2017-09-10:
The package is called raspberrypi-kernel-headers. This is now also documented in the README.md. This issue can be closed.

closed 2017-09-10 by CGarces

rahulm-gdb, 2018-06-04:
Does it works on ubuntu 18? i'm tyrying to install it but it's failing.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=4.15.0-22-generic...(bad exit status: 2)
ERROR (dkms apport): binary package for rtl8192eu: 1.0 not found
Error! Bad return status for module build on kernel: 4.15.0-22-generic (x86_64)
Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.

PanderMusubi, 2018-06-04:
No problem here building on 18.04 on Intel.

    $ sudo dkms add .
    
    Creating symlink /var/lib/dkms/rtl8192eu/1.0/source ->
                     /usr/src/rtl8192eu-1.0
    
    DKMS: add completed.

and

    $ sudo dkms install rtl8192eu/1.0
    
    Kernel preparation unnecessary for this kernel.  Skipping...
    
    Building module:
    cleaning build area...
    'make' all KVER=4.15.0-22-generic.........................
    cleaning build area...
    
    DKMS: build completed.
    
    8192eu.ko:
    Running module version sanity check.
     - Original module
       - No original module exists within this kernel
     - Installation
       - Installing to /lib/modules/4.15.0-22-generic/updates/dkms/
    
    depmod...
    
    Backing up initrd.img-4.15.0-22-generic to /boot/initrd.img-4.15.0-22-generic.old-dkms
    Making new initrd.img-4.15.0-22-generic
    (If next boot fails, revert to initrd.img-4.15.0-22-generic.old-dkms image)
    update-initramfs.......
    
    DKMS: install completed.

and

    $ sudo lshw -c network
      *-network                 
           description: Ethernet interface
           product: RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller
           vendor: Realtek Semiconductor Co., Ltd.
           physical id: 0
           bus info: pci@0000:05:00.0
           logical name: enp5s0
           version: 03
           serial: 20:cf:30:ea:3e:76
           size: 100Mbit/s
           capacity: 1Gbit/s
           width: 64 bits
           clock: 33MHz
           capabilities: pm msi pciexpress msix vpd bus_master cap_list rom ethernet physical tp mii 10bt 10bt-fd 100bt 100bt-fd 1000bt 1000bt-fd autonegotiation
           configuration: autonegotiation=on broadcast=yes driver=r8169 driverversion=2.3LK-NAPI duplex=full firmware=rtl_nic/rtl8168d-2.fw ip=192.168.12.63 latency=0 link=yes multicast=yes port=MII speed=100Mbit/s
           resources: irq:29 ioport:d800(size=256) memory:f6fff000-f6ffffff memory:f6ff8000-f6ffbfff memory:f7ef0000-f7efffff