Mange/rtl8192eu-linux-driver #45: Can't Build on Linux Kernel 4.12.3!!!

Syntist, 2017-08-03:
I am using Antergos, can't get it work!! Output log give below:

`DKMS make.log for rtl8192eu-1.0 for kernel 4.12.3-1-ARCH (x86_64)
Fri  4 Aug 00:33:53 PKT 2017
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.12.3-1-ARCH/build M=/var/lib/dkms/rtl8192eu/1.0/build  modules
make[1]: Entering directory '/usr/lib/modules/4.12.3-1-ARCH/build'
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_security.o
/var/lib/dkms/rtl8192eu/1.0/build/core/rtw_security.c: In function ‘aes_cipher’:
/var/lib/dkms/rtl8192eu/1.0/build/core/rtw_security.c:1598:5: warning: this ‘for’ clause does not guard... [-Wmisleading-indentation]
     for (j = 0; j < 8; j++)
     ^~~
/var/lib/dkms/rtl8192eu/1.0/build/core/rtw_security.c:1601:2: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘for’
  payload_index = hdrlen + 8;
  ^~~~~~~~~~~~~
/var/lib/dkms/rtl8192eu/1.0/build/core/rtw_security.c: In function ‘aes_decipher’:
/var/lib/dkms/rtl8192eu/1.0/build/core/rtw_security.c:1984:5: warning: this ‘for’ clause does not guard... [-Wmisleading-indentation]
     for (j = 0; j < 8; j++)
     ^~~
/var/lib/dkms/rtl8192eu/1.0/build/core/rtw_security.c:1987:2: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘for’
  payload_index = hdrlen + 8;
  ^~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mlme_ext.o
/var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mlme_ext.c: In function ‘rtw_delba_check’:
/var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mlme_ext.c:12427:7: warning: this ‘else’ clause does not guard... [-Wmisleading-indentation]
       else
       ^~~~
/var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mlme_ext.c:12429:8: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘else’
        psta->recvreorder_ctrl[i].enable = _FALSE;
        ^~~~
/var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mlme_ext.c:12430:7: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
       if (ret != _FAIL)
       ^~
/var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mlme_ext.c:12432:8: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
        rtw_reset_continual_no_rx_packet(psta, i);
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/os_dep/linux/os_intfs.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/os_dep/linux/usb_intf.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/os_dep/linux/usb_ops_linux.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/os_dep/linux/ioctl_linux.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/os_dep/linux/xmit_linux.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/os_dep/linux/mlme_linux.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/os_dep/linux/recv_linux.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/os_dep/linux/ioctl_cfg80211.o
/var/lib/dkms/rtl8192eu/1.0/build/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_spt_band_free’:
/var/lib/dkms/rtl8192eu/1.0/build/os_dep/linux/ioctl_cfg80211.c:289:20: warning: comparison between ‘enum nl80211_band’ and ‘enum ieee80211_band’ [-Wenum-compare]
[... 53 lines cut ...]
make[1]: *** [Makefile:1512: _module_/var/lib/dkms/rtl8192eu/1.0/build] Error 2
make[1]: Leaving directory '/usr/lib/modules/4.12.3-1-ARCH/build'
make: *** [Makefile:1697: modules] Error 2`

About My PC:

               `.-/::/-``                
            .-/osssssssso/.               cnyster@Cnyster-PC
           :osyysssssssyyys+-             OS: Antergos 
        `.+yyyysssssssssyyyyy+.           Kernel: x86_64 Linux 4.12.3-1-ARCH
       `/syyyyyssssssssssyyyyys-`         Uptime: 1h 1m
      `/yhyyyyysss++ssosyyyyhhy/`         Packages: 977
     .ohhhyyyyso++/+oso+syy+shhhho.       Shell: bash 4.4.12
    .shhhhysoo++//+sss+++yyy+shhhhs.      Resolution: 1366x768
   -yhhhhs+++++++ossso+++yyys+ohhddy:     DE: GNOME 
  -yddhhyo+++++osyyss++++yyyyooyhdddy-    WM: GNOME Shell
 .yddddhso++osyyyyys+++++yyhhsoshddddy`   WM Theme: Arc-Dark
`odddddhyosyhyyyyyy++++++yhhhyosddddddo   GTK Theme: Arc-Darker [GTK2/3]
.dmdddddhhhhhhhyyyo+++++shhhhhohddddmmh.  Icon Theme: Papirus-Dark
ddmmdddddhhhhhhhso++++++yhhhhhhdddddmmdy  Font: Cantarell 11
dmmmdddddddhhhyso++++++shhhhhddddddmmmmh  CPU: Intel Core i5-3230M @ 4x 3.2GHz [54.0°C]
-dmmmdddddddhhyso++++oshhhhdddddddmmmmd-  GPU: intel
 .smmmmddddddddhhhhhhhhhdddddddddmmmms.   RAM: 2211MiB / 3845MiB
   `+ydmmmdddddddddddddddddddmmmmdy/.    
      `.:+ooyyddddddddddddyyso+:.`

scdas141, 2017-08-04:
Same issue here, was working real fine upto kernel 4.11 but as soon as I upgraded to 4.12 -- received error!!

`# make && make install -j2
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.12.4-towo.2-siduction-amd64/build M=/home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x  modules
make[1]: Entering directory '/usr/src/linux-headers-4.12.4-towo.2-siduction-amd64'
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/rtw_cmd.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/rtw_security.o
/home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/rtw_security.c: In function ‘aes_cipher’:
/home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/rtw_security.c:1598:5: warning: this ‘for’ clause does not guard... [-Wmisleading-indentation]
     for (j = 0; j < 8; j++)
     ^~~
/home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/rtw_security.c:1601:2: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘for’
  payload_index = hdrlen + 8;
  ^~~~~~~~~~~~~
/home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/rtw_security.c: In function ‘aes_decipher’:
/home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/rtw_security.c:1984:5: warning: this ‘for’ clause does not guard... [-Wmisleading-indentation]
     for (j = 0; j < 8; j++)
     ^~~
/home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/rtw_security.c:1987:2: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘for’
  payload_index = hdrlen + 8;
  ^~~~~~~~~~~~~
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/rtw_debug.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/rtw_io.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/rtw_ioctl_query.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/rtw_ioctl_set.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/rtw_ieee80211.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/rtw_mlme.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/rtw_mlme_ext.o
/home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/rtw_mlme_ext.c: In function ‘rtw_delba_check’:
/home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/rtw_mlme_ext.c:12427:7: warning: this ‘else’ clause does not guard... [-Wmisleading-indentation]
       else
       ^~~~
/home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/rtw_mlme_ext.c:12429:8: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘else’
        psta->recvreorder_ctrl[i].enable = _FALSE;
        ^~~~
/home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/rtw_mlme_ext.c:12430:7: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
       if (ret != _FAIL)
       ^~
/home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/rtw_mlme_ext.c:12432:8: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
        rtw_reset_continual_no_rx_packet(psta, i);
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/rtw_wlan_util.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/rtw_vht.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/rtw_pwrctrl.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/rtw_rf.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/rtw_recv.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/rtw_sta_mgt.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/rtw_ap.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/rtw_xmit.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/rtw_p2p.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/rtw_tdls.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/rtw_br_ext.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/rtw_iol.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/rtw_sreset.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/rtw_btcoex.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/rtw_beamforming.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/rtw_odm.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/core/efuse/rtw_efuse.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/os_dep/osdep_service.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/os_dep/linux/os_intfs.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/os_dep/linux/usb_intf.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/os_dep/linux/usb_ops_linux.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/os_dep/linux/ioctl_linux.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/os_dep/linux/xmit_linux.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/os_dep/linux/mlme_linux.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/os_dep/linux/recv_linux.o
  CC [M]  /home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/os_dep/linux/ioctl_cfg80211.o
[... cut ...]
scripts/Makefile.build:302: recipe for target '/home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/os_dep/linux/ioctl_cfg80211.o' failed
make[2]: *** [/home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x/os_dep/linux/ioctl_cfg80211.o] Error 1
Makefile:1516: recipe for target '_module_/home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x' failed
make[1]: *** [_module_/home/ik/Downloads/rtl8192eu-linux-driver-realtek-4.4.x] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.12.4-towo.2-siduction-amd64'
Makefile:1699: recipe for target 'modules' failed
make: *** [modules] Error 2
`
Hope it is resolved!!

masterzorag, 2017-08-05:
Yes, known issues; I can confirm linux-4.12.y needs some patching.
Actually driver is working with 4.11.12 [EOL]

scdas141, 2017-08-11:
Maybe someone can get this working on kernel 4.12.x ??

masterzorag, 2017-08-11:
tracking first build error
```sh
ioctl_cfg80211.c:816:6: error: passing argument 2 of ‘cfg80211_roamed’ from incompatible pointer type [-Werror=incompatible-pointer-types]
    , notify_channel
note: expected ‘struct cfg80211_roam_info *’ but argument is of type ‘struct ieee80211_channel *’
 void cfg80211_roamed(struct net_device *dev, struct cfg80211_roam_info *info,
...
then obviously:
error: too many arguments to function ‘cfg80211_roamed’
```
lands me [here](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git/commit/?h=v4.12.6&id=29ce6ecbb83c9185d76e3a7c340c9702d2a54961)

scdas141, 2017-08-11:
@masterzorag 
Thanks buddy.. I'm sure a solution will crop up.. 

I think New Kernels, especially like, 4.12 with major changes -- [15,000 source code changes & More Than One Million New Lines](http://www.phoronix.com/scan.php?page=news_item&px=Linux-4.12-Post-Merge-Git-Stats) might take a while to complete debugging.. :-))

masterzorag, 2017-08-11:
just managed to build fine on linux-4.12.2, I will push patches on my branch

scdas141, 2017-08-12:
wow.. good news after all..

CGarces, 2017-08-17:
@scdas141 we try to do our best, but we have a private life outside Github. Fell free to be pro-active and send a PR.

PR #47 (2017-08-23): Fix build with Linux Kernel 4.12
Based on the work done by @masterzorag, fix the issues compiling with 4.12

Ping @Syntist, @scdas141, and @masterzorag for feedback

Close #45
closed 2017-08-23 by CGarces