Mange/rtl8192eu-linux-driver #46: I already installed the driver, but I can not connect to a wifi network

MSFHacker, 2017-08-03:
Hello, i have a TP-LINK adapter WN823N connect it to kali linux ande install this driver https://github.com/Mange/rtl8192eu-linux-driver.

Before installing the driver i only had one or two networks to "reach" because i did not have a good signal and it was impossible to connect to a network, after installing the driver mentioned above restart my kali linux and when i select, select networks , more networks came out and with good signal then i proceeded to connect to my wifi network, i put the key and it did not connect i stayed there loading, i tried to navigate and i did not have access.

someone help me please

nomadmzl, 2017-08-04:
Hi, try with `sudo depmod -a` if it doesn't works, so it will sound dumb but try to unplug and plug again your dongle.

scdas141, 2017-08-07:
Open “/etc/modules” from Root File Manager by a text-editor (Leafpad, Gedit. etc..)

It will be something like this..

[http://i.imgur.com/1KCEjB4.png](url)

Edit and Add an entry 8192eu as shown below:

[http://i.imgur.com/1KCEjB4.png](url)

Restart ur pc & it should work fine.. for details:

[https://scdas141.wordpress.com/2017/01/28/how-to-compile-mangertl8192eu-linux-driver-driver/](url)

PS. Remember to copy & paste the links as github does not allow clickable links:

nomadmzl, 2017-08-10:
@scdas141 the links are down.

scdas141, 2017-08-11:
@nomadmzl 

Hi.. The links are not down, I have tested them myself.. 

The only thing is github does not allow clickable links. You will have to copy & paste the links.. Select the link, right click copy - not copy link - and paste into a new tab..

xxd3vin, 2017-08-11:
@scdas141 Markdown syntax for a link should be:

```
[http://i.imgur.com/1KCEjB4.png](http://i.imgur.com/1KCEjB4.png)
```

OR

```
[Open “/etc/modules” from Root File Manager by a text-editor](http://i.imgur.com/1KCEjB4.png)
```

scdas141, 2017-08-11:
@xxd3vin 
Thx bro for the formatting help..

@nomadmzl 
Links fixed.. hope it helps..

xxd3vin, 2017-08-11:
@scdas141 I try your method on one of my workstations (Ubuntu 17.04, Linux kernel 4.10.0-30), it worked, thanks.

scdas141, 2017-08-11:
Glad if it helped.. Thx to u too.. :-))

milomak, 2017-08-16:
where can i go to get help with the following (DWA-131 rev E1 v5.02 running rtl8192eu)

debian sid and a backports 4.9 kernel

# nmcli dev wifi list
shows the list of wifi connections. and shows mine (WPA2-PSK-AES)

# nmcli device wifi connect SSID-Name password wireless-password
tells me "Secrets were required, but not provided"

the password is of course correct (disconnected my phone and reestablished a connection to the router)

scdas141, 2017-08-17:
@milomak 

[https://scdas141.wordpress.com/2017/01/28/how-to-compile-mangertl8192eu-linux-driver-driver/](https://scdas141.wordpress.com/2017/01/28/how-to-compile-mangertl8192eu-linux-driver-driver/)

milomak, 2017-08-17:
@scdas141 

I used the dkms method to install and the driver is loaded as when I run lsmod I can see 8192eu module. I ran the dkms module in the 4.9 kernel I am running.

Are you saying run make and make install will have a different outcome to building and installing using dkms? 

I can see the wireless card if I run ifconfig. I am able to run the nmcli command. It sees my wifi and those around. It's connect to the router that's the issue.

Or have I missed something in the link you provided?

scdas141, 2017-08-21:
@milomak 

run `lshw -c network` as root to verify the wireless driver your kernel is using.. 

Even if your install is succesful, whether by make or dkms (you can use either), the linux kernel will use the in-built default rtl8xxxu driver instead of the 8192eu..

To disable the rtl8xxxu driver, after installation open `etc/modprobe.d` in a root folder and create a new file `rtl8xxxu-blacklist.conf` -- edit it with a text editor (leafpad, mousepad etc..) and add these lines:

```
# Do not load the 'rtl8xxxu' module on boot.
blacklist rtl8xxxu
```
Screenshots:
[etc/modprobe.d](http://i.imgur.com/0Cr1aQG.png)

[rtl8xxxu-blacklist.conf](http://i.imgur.com/9CeQx9Y.png)

Save it and restart the system and your connection will work just fine..

Verify that the wireless module loaded is 8192eu by running `lshw -c network`

shvchk, 2017-08-22:
Having a similar problem: driver installs just fine, I can connect to open WiFi network (i.e. without password), but for password protected network (WPA2-PSK/AES) Network Manager just keeps asking password (system log says "4-Way Handshake failed - pre-shared key may be incorrect", but I'm 100% sure it is correct, I can connect with this password from other devices, I even tried copy-pasting it to rule typos or other input problems). Driver was installed with DKMS, then added to `/etc/modules`, other driver was blacklisted, then system was rebooted. I'm using KDE, but this problem probably isn't DE related, as I've also tried connecting from console with `nmtui-connect` with the same result (it just keeps asking password again and again). Any ideas what could be wrong?

milomak, 2017-08-24:
i don't think it's a case of the driver not loading properly. I haven't
blacklisted anything but yet this is the only driver running

lshw -c network
  *-network
       description: Wireless interface
       physical id: 1
       bus info: usb@2:3
       logical name: wlx542aa2b82afb
       serial: 54:2a:a2:b8:2a:fb
       capabilities: ethernet physical wireless
       configuration: broadcast=yes driver=rtl8192eu ip=192.168.1.115
multicast=yes wireless=IEEE 802.11bgn

lsmod | grep rtl returns nothing

lsmod | grep 8192
8192eu               1093632  0
cfg80211              598016  1 8192eu
snd_hda_core           81920  5
snd_hda_intel,snd_hda_codec,snd_hda_codec_hdmi,snd_hda_codec_generic,snd_hda_codec_via
ehci_hcd               81920  1 ehci_pci
usbcore               249856  6
usbhid,ehci_hcd,8192eu,ohci_pci,ohci_hcd,ehci_pci

karl6994, 2017-08-25:
I encountered the same issue - I just edited /etc/modules & am rebooting as we speak. 
Hoping it works! I'm running Debian 9 Stretch, but its upgraded to the Debian Buster repo.

karl6994, 2017-08-25:
Yeah... This isn't working... Same problem, it shows the networks & appropriately has the ranges displayed, but I still can't make it work :( I've tried everything listed here and on the links that were listed here also..

CGarces, 2017-08-30:
Hi.
Can any one confirm if the solution proposed by @IIETP at #48 fix the issue?

> To fix this you need edit NetworkManager config file:
> 
> `sudo nano /etc/NetworkManager/NetworkManager.conf`
> 
> and add below 2 lines:
> 
> `[device]`
> `wifi.scan-rand-mac-address=no`

shvchk, 2017-08-30:
Yeah, this fixed my issue indeed! Thanks!

CGarces, 2017-08-30:
Close the issue with the solution proposed at #48

closed 2017-08-30 by CGarces

juniorhero, 2017-11-27:
@scdas141  thanks man
you saved me

scdas141, 2017-11-27:
@juniorhero Really dude, it's nothing--

its-wednesday, 2018-03-21:
@scdas141 Thank you so much!
For search engine users on Arch/Antergos, the following command should set you:
`su -c ' echo "8192eu" > /etc/modules-load.d/wifi-driver.conf'`

leesei, 2018-07-26:
I'm using TP-LINK TN-WN826N, installed the rtl8192eu driver as DKMS module. The driver loads fine on boot but I'm getting authentication timeout when connecting to access point with WPA2-PSK.

It turns out the kernel built-in `rtl8xxxu` module was interfering with `rtl8192eu` (both are loaded when I plug in the dongle).

Here's a one-liner to blacklist `rtl8xxxu`:
```sh
sudo su
mkdir -p /etc/modprobe.d
cat << EOF > /etc/modprobe.d/rtl8xxxu-blacklist.conf
# Do not load the 'rtl8xxxu' module on boot.
blacklist rtl8xxxu
EOF
```

LautaroLobo12, 2019-07-01:
@scdas141 sorry to bother you. I installed the new driver, addedit to the modules file, also blacklisted the rtl8xxxu driver... and yet nothing works. With the rtl8xxxu driver I can't connect to wifi, it keeps trying to connect to any wifi network, and without it (blacklisted) there is no wireless connection available, lshw does not show it, just shows the ethernet.

Any clue? What may be causing this issue? I would love a solution but knowing what to look at would be also lovely.

Regards,
Lautaro.-

CGarces, 2019-07-01:
@LautaroLobo12 please try this:

Edit NetworkManager config file:

`sudo nano /etc/NetworkManager/NetworkManager.conf
`
and add below 2 lines:

```
[device]
wifi.scan-rand-mac-address=no
```

LautaroLobo12, 2019-07-01:
Thanks for answering men.

Well, those lines are already there... there is no need to add nothing because those lines came by default it seems.