Mange/rtl8192eu-linux-driver #48: Can't connect to any networks

karl6994, 2017-08-25:
Hello everyone, I can't make the wifi dongle connect to either my extender or my router.. 
I got the driver installed right, it displays the networks & appropriately shows the ranges, but it keeps prompting me for the network password & won't connect.. I have tried the solutions on the other post, the /etc/modules thing, and that doesn't seem to work either :/ 
I've tried rebooting, I've tried modprobe -r and modprobing, I've tried installing it normally & via DKMS.. not sure why it won't connect...

IIETP, 2017-08-25:
I had this issue too. And I've tried modprobe, blacklist, etc. But when i tried to connect with wpa_supplicant in terminal, it has connected without any problem.
In my Debian 9 wicd solved this problem:
`sudo apt install wicd`

CGarces, 2017-08-30:
Thanks @IIETP 
You have any link to the NetworkManager bug?

closed 2017-08-30 by CGarces

IIETP, 2017-08-30:
Here it is: https://bugs.launchpad.net/ubuntu/+source/network-manager/+bug/1681513

shvchk, 2017-08-30:
Relevant Network Manager bug: https://bugzilla.gnome.org/show_bug.cgi?id=777523

karl6994, 2017-09-01:
If not for the github notifications, I would've never been able to find this... Odd I can't find it in the list of issues... Just trying the fix now, and it worked!!!! Hurray!!! Thank you!!