Mange/rtl8192eu-linux-driver #49: Build not working if cross compile

Alex66955, 2017-08-30:
When I try to cross compile this driver for an IMX6.
The build process stops without any error.

Is cross compile supported by this driver?

```
make ARCH=arm CROSS_COMPILE=arm-v7a-linux-gnueabihf- -C /data/TQ-DEV/TQMa6x.BSP.SW.SRC.Snapshot-20161219/platform-ngc/build-target/kernel-header-4.1/ M=/data/TQ-DEV/TQMa6x.BSP.SW.SRC.Snapshot-20161219/local_src/rtl_module modules 
make: Verzeichnis „/data/TQ-DEV/TQMa6x.BSP.SW.SRC.Snapshot-20161219/platform-ngc/build-target/kernel-header-4.1“ wird betreten

  WARNING: Symbol version dump ./Module.symvers
           is missing; modules will have no dependencies and modversions.

  Building modules, stage 2.
  MODPOST 0 modules
make: Verzeichnis „/data/TQ-DEV/TQMa6x.BSP.SW.SRC.Snapshot-20161219/platform-ngc/build-target/kernel-header-4.1“ wird verlassen
```

CGarces, 2017-08-30:
Hi.

Which branch are you using ? `master` or `realtek-4.4`?

Cross compile must work, this driver is used by projects like OpenElec and LibreElec for all architectures, including IMX6.

Based on the [code from LibreELEC](https://github.com/LibreELEC/LibreELEC.tv/blob/master/projects/imx6/options#L51-L73) `master` branch compiles with 3.14, 4,4, 4.8 and 4.10.

masterzorag, 2017-09-06:
@Alex66955 cross-compile is supported, as build out-of-tree is supported;
Your Warning is just a warning, but you are building against kernel-headers instead of full kernel-sources;
Example:
```sh
blackbox RTL8192EU-linux # pwd
/Storage/RTL8192EU-linux

blackbox RTL8192EU-linux # make KSRC=/Archive/linux-4.12.10
make ARCH=x86_64 CROSS_COMPILE= -C /Archive/linux-4.12.10 M=/Storage/RTL8192EU-linux  modules
make[1]: Entering directory '/Archive/linux-4.12.10'

  WARNING: Symbol version dump ./Module.symvers
           is missing; modules will have no dependencies and modversions.
...

blackbox RTL8192EU-linux # ls -l 8192eu.ko 
-rw-r--r-- 1 root root 2358944 Sep  6 10:15 8192eu.ko
```
So, try just appending KSRC=/path/to/kernel-source to your `make` invocation, like this:

`make ARCH=arm CROSS_COMPILE=arm-v7a-linux-gnueabihf- KSRC=/path/to/kernel-source`

Note that:
- **_at least_** `make modules_prepare` has to be called **_before_** building Realtek module!
- you have to change target into module source too

Also, consider moving to linux-4.4 if you get any more issues.

Alex66955, 2017-09-08:
Thank you very much.
It is working now and I could test the driver in Client and AP-Mode!

closed 2017-09-08 by Alex66955

rohitsay89, 2018-04-02:
Thanks for the explanation above, just wanted to know how to locate the kernel folder in the beaglebone black. As I am trying to build the drivers for the same beaglebone black, do I need to mention the kernel files ?