Mange/rtl8192eu-linux-driver #55: Shipped with main kernel

PanderMusubi, 2017-09-10:
Maybe a strange question, is it possible to get this driver shipped with the main kernel so it is by default available to all?

FFY00, 2017-09-10:
I think so, but it would need some work.

PanderMusubi, 2017-09-10:
Such as? Think might help a bit:

     clang-format -style=file -i `find . -name '*rtl8192*.[ch]'`

and

    cppcheck --enable=all --suppress=missingIncludeSystem -Iinclude `find . -name '*rtl8192*.[ch]'`

gilcu3, 2017-12-08:
In the main kernel they are working in their own driver, rtl8xxxu, which currently works for most devices, but still lacks some functionality...

PanderMusubi, 2018-06-04:
So what is the most effective path? Port the lacking functionality from there to the driver in the kernel on continue work there?

PanderMusubi, 2019-06-17:
Ah, found https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/net/wireless/realtek/rtl8xxxu which is shipped in kernel version 5. What is the gap in functionality at the moment?

CGarces, 2019-06-17:
It's a good question @PanderMusubi 
rtl8xxu was not working fine in the past (2017) but now I don't have the hardware to test it.

closed 2019-12-09 by CGarces