Mange/rtl8192eu-linux-driver #57: Support of kernel 4.10 in Ubuntu 16.04.3

sargyrop, 2017-09-12:
I have installed the driver from the main repository and I can see a list of networks in the area. However, it keeps asking for the password (even if I set the WiFi to open access without password) and does not connect.

Is there a version that supports kernel 4.10?

OS: Ubuntu 16.04.3, kernel: 4.10.0-28-generic

$ lsmod |grep 8192 
'8192eu               1101824  0
cfg80211              602112  1 8192eu'

  *-network
       description: Wireless interface
       physical id: 1
       bus info: usb@3:9
       logical name: wlx18d6c71af449
       serial: 18:d6:c7:1a:f4:49
       capabilities: ethernet physical wireless
       configuration: broadcast=yes driver=rtl8192eu multicast=yes wireless=unassociated

$ rfkill list
0: phy0: Wireless LAN
    Soft blocked: no
    Hard blocked: no

CGarces, 2017-09-12:
Hi,

Looks like your system is affected by the bug on network-manager try to update it to 1.8.2-1ubuntu3 or check the solution proposed at #48

sargyrop, 2017-09-13:
Thanks for the prompt reply. It seems I have messed a  bit my system and after installing several other drivers, the usb dongle is not recognized now. What is the recommended way to completely uninstall previous versions and start from scratch?

sargyrop, 2017-09-13:
I am still stuck after applying the fix. Here's the output of the syslog
<pre>
Sep 13 09:01:56 savvas-Desktop kernel: [    7.520610] 8192eu: loading out-of-tree module taints kernel.
Sep 13 09:01:56 savvas-Desktop kernel: [    7.520858] 8192eu: module verification failed: signature and/or required key missing - tainting kernel
Sep 13 09:01:56 savvas-Desktop kernel: [    7.522006] RTL871X: rtl8192eu v4.4.1_17696.20160509_BTCOEX20160412-0042
Sep 13 09:01:56 savvas-Desktop kernel: [    7.522006] RTL871X: rtl8192eu BT-Coex version = BTCOEX20160412-0042
Sep 13 09:01:56 savvas-Desktop kernel: [    7.582044] usbcore: registered new interface driver rtl8192eu
Sep 13 09:01:56 savvas-Desktop kernel: [    7.627098] rtl8192eu 3-14:1.0 wlx18d6c71af449: renamed from wlan0
Sep 13 09:01:56 savvas-Desktop NetworkManager[901]: <info>  [1505282516.8846] rfkill0: found WiFi radio killswitch (at /sys/devices/pci0000:00/0000:00:14.0/usb3/3-14/3-14:1.0/ieee80211/phy0/rfkill0) (driver rtl8192eu)
<pre>

CGarces, 2017-09-13:
That's really strange, the description of your error is exactly the same as #46, #48 and #51

Have you edited `/etc/NetworkManager/NetworkManager.conf` ?
and added this lines?

```
[device]
wifi.scan-rand-mac-address=no
```

sargyrop, 2017-09-13:
Yes, the <pre> /etc/NetworkManager/NetworkManager.conf</pre> now looks like this:
<pre>
[main]
plugins=ifupdown,keyfile,ofono
dns=dnsmasq

[ifupdown]
managed=false

[device]
wifi.scan-rand-mac-address=no
</pre>

What is the recommended way to completely uninstall anything related to rtl8192 and start over?

karl6994, 2017-09-13:
Try doing a sudo make uninstall in the rtl8192eu-linux-driver folder.

sargyrop, 2017-09-13:
Thanks again for the prompt replies but they do seem to work for me.
I cannot switch to Debian because my whole system is setup in Ubuntu.

I tried uninstalling using <pre>sudo make uninstall</pre> and re-install but the problem persists. Also, even though I uninstalled, I get the following:
<pre>
$ lsmod | grep 8192
snd_hda_core           81920  5 8192eu               1105920  0
cfg80211              602112  2 mac80211,8192eu
</pre>
Why isn't 8192eu removed when I uninstall?

CGarces, 2017-09-18:
Looks like you have a conflict with the rtl8xxxu module shipped with the kernel.
https://wireless.wiki.kernel.org/en/users/drivers/rtl819x

Try to Blacklistist the module rtl8xxxu
https://wiki.debian.org/KernelModuleBlacklisting

ShaolinFu, 2017-10-09:
I have met the same issue.  anyone has a solution or workaround?

van100j, 2017-11-17:
Hey @ShaolinFu it worked for me after blacklisting the rtl8xxxu module, basically followed the instructions [in the link](https://wiki.debian.org/KernelModuleBlacklisting) @CGarces provided in the previous comment.

Here are the steps that did the job for me:

1. [Build and install using DKMS](https://github.com/Mange/rtl8192eu-linux-driver#building-and-installing-using-dkms)
2. Create a file `/etc/modprobe.d/rtl8xxxu.conf` containing containing `blacklist rtl8xxxu`
3. Run `sudo depmod -ae` [as root]
4. Recreate your initrd with `sudo update-initramfs -u`
5. Reboot

closed 2018-07-11 by CGarces