Mange/rtl8192eu-linux-driver #60: Works on my Ubuntu (studio) 16.04.3 but only after plug out and in

dirk-vanda, 2017-10-14:
I installed this driver on my Ubuntu (studio) 16.04.3 and rebooted. When I plugged in the wifi dongle, it works. But when I reboot with the dongle plugged in, it doesn't work (it doesn't detect any wifi signals) until I remove it and plug it back in. Then it recognizes wifi signals and connects to mine.
Can anyone help? How can the dongle work without plugging it out and in?

akana, 2017-11-08:
u should try

> sudo -i
> echo 8192eu >> /etc/modules
> exit

dirk-vanda, 2017-11-09:
Thanks a lot for this suggestion @akana . I works perfect now. I rebooted several times and the 8192eu module now automatically detects the Wifi signals. It isn't necessary to plug the dongle in and out any more.
I really appreciate your help.
Dirk

akana, 2017-11-10:
@dirk-vanda you're welcome ;)

closed 2017-11-10 by dirk-vanda