Mange/rtl8192eu-linux-driver #64: Network-Manager prompt for password repeatedly on linux  4.9

xkos, 2017-11-29:
it works well on debian 8
yesterday i upgrade to debian 9 and it didn't work
and i reinstall it and it prompt for password
this is msg show:
[image]

it seems auth success but i don't  know why it can't connect

xkos, 2017-11-29:
i remove driver and install and this is the new message:

```
[    3.848486] EXT4-fs (sdb5): re-mounted. Opts: errors=remount-ro
[    3.860764] systemd-journald[225]: Received request to flush runtime journal from PID 1
[    4.279748] random: crng init done
[    5.083290] input: Power Button as /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0C:00/input/input6
[    5.083293] ACPI: Power Button [PWRB]
[    5.083329] input: Power Button as /devices/LNXSYSTM:00/LNXPWRBN:00/input/input7
[    5.083330] ACPI: Power Button [PWRF]
[    5.125134] input: PC Speaker as /devices/platform/pcspkr/input/input8
[    5.128261] shpchp: Standard Hot Plug PCI Controller Driver version: 0.4
[    5.200662] RAPL PMU: API unit is 2^-32 Joules, 4 fixed counters, 655360 ms ovfl timer
[    5.200663] RAPL PMU: hw unit of domain pp0-core 2^-14 Joules
[    5.200663] RAPL PMU: hw unit of domain package 2^-14 Joules
[    5.200663] RAPL PMU: hw unit of domain dram 2^-14 Joules
[    5.200664] RAPL PMU: hw unit of domain pp1-gpu 2^-14 Joules
[    5.201529] sd 0:0:0:0: Attached scsi generic sg0 type 0
[    5.201548] sd 1:0:0:0: Attached scsi generic sg1 type 0
[    5.322911] [drm] Initialized
[    5.419575] iTCO_vendor_support: vendor-support=0
[    5.684496] [drm] Memory usable by graphics device = 2048M
[    5.684498] [drm] Replacing VGA console driver
[    5.685086] Console: switching to colour dummy device 80x25
[    5.691125] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[    5.691126] [drm] Driver supports precise vblank timestamp query.
[    5.693341] vgaarb: device changed decodes: PCI:0000:00:02.0,olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    5.734264] ACPI: Video Device [GFX0] (multi-head: yes  rom: no  post: no)
[    5.734587] acpi device:63: registered as cooling_device9
[    5.734639] input: Video Bus as /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/LNXVIDEO:00/input/input9
[    5.734694] [drm] Initialized i915 1.6.0 20160919 for 0000:00:02.0 on minor 0
[    5.762472] fbcon: inteldrmfb (fb0) is primary device
[    5.786096] Console: switching to colour frame buffer device 240x67
[    5.805461] i915 0000:00:02.0: fb0: inteldrmfb frame buffer device
[    5.816996] iTCO_wdt: Intel TCO WatchDog Timer Driver v1.11
[    5.817027] iTCO_wdt: Found a Lynx Point TCO device (Version=2, TCOBASE=0x1860)
[    5.817103] iTCO_wdt: initialized. heartbeat=30 sec (nowayout=0)
[    6.042482] Adding 8289276k swap on /dev/sdb6.  Priority:-1 extents:1 across:8289276k FS
[    6.143400] snd_hda_intel 0000:00:03.0: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    6.230806] input: HDA Intel HDMI HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:03.0/sound/card0/input10
[    6.230850] input: HDA Intel HDMI HDMI/DP,pcm=7 as /devices/pci0000:00/0000:00:03.0/sound/card0/input11
[    6.230890] input: HDA Intel HDMI HDMI/DP,pcm=8 as /devices/pci0000:00/0000:00:03.0/sound/card0/input12
[    6.252772] snd_hda_codec_realtek hdaudioC1D2: autoconfig for ALC892: line_outs=1 (0x14/0x0/0x0/0x0/0x0) type:line
[    6.252773] snd_hda_codec_realtek hdaudioC1D2:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[    6.252774] snd_hda_codec_realtek hdaudioC1D2:    hp_outs=1 (0x1b/0x0/0x0/0x0/0x0)
[    6.252775] snd_hda_codec_realtek hdaudioC1D2:    mono: mono_out=0x0
[    6.252775] snd_hda_codec_realtek hdaudioC1D2:    dig-out=0x1e/0x0
[    6.252776] snd_hda_codec_realtek hdaudioC1D2:    inputs:
[    6.252777] snd_hda_codec_realtek hdaudioC1D2:      Front Mic=0x19
[    6.252778] snd_hda_codec_realtek hdaudioC1D2:      Rear Mic=0x18
[    6.252779] snd_hda_codec_realtek hdaudioC1D2:      Line=0x1a
[    6.264131] input: HDA Intel PCH Front Mic as /devices/pci0000:00/0000:00:1b.0/sound/card1/input13
[    6.264172] input: HDA Intel PCH Rear Mic as /devices/pci0000:00/0000:00:1b.0/sound/card1/input14
[    6.264206] input: HDA Intel PCH Line as /devices/pci0000:00/0000:00:1b.0/sound/card1/input15
[    6.264238] input: HDA Intel PCH Line Out as /devices/pci0000:00/0000:00:1b.0/sound/card1/input16
[    6.264276] input: HDA Intel PCH Front Headphone as /devices/pci0000:00/0000:00:1b.0/sound/card1/input17
[    6.703460] usb 3-5: This Realtek USB WiFi dongle (0x0bda:0x818b) is untested!
[    6.703463] usb 3-5: Please report results to Jes.Sorensen@gmail.com
[    6.765759] usb 3-5: Vendor: Realtek
[    6.765760] usb 3-5: Product: 802.11n NI
[    6.765760] usb 3-5: Serial: 
[    6.765761] usb 3-5: rtl8192eu_parse_efuse: dumping efuse (0x200 bytes):
[    6.765762] usb 3-5: 00: 29 81 00 7c 01 40 03 00
[    6.765763] usb 3-5: 08: 40 74 04 50 14 00 00 00
[    6.765763] usb 3-5: 10: 22 22 22 22 22 22 23 23
[    6.765764] usb 3-5: 18: 23 23 23 02 ef ef ff ff
[    6.765764] usb 3-5: 20: ff ff ff ff ff ff ff ff
[    6.765765] usb 3-5: 28: ff ff ff ff ff ff ff ff
[    6.765765] usb 3-5: 30: ff ff ff ff ff ff ff ff
[    6.765766] usb 3-5: 38: ff ff 1f 1f 1e 1e 1e 1e
[... 92 lines cut ...]
[   59.099453] wlx3c46d8c536e6: authentication with 90:03:25:b9:a4:c0 timed out
[   59.355736] wlx3c46d8c536e6: authenticate with 74:5a:aa:c1:63:80
[   59.363072] wlx3c46d8c536e6: send auth to 74:5a:aa:c1:63:80 (try 1/3)
[   59.563471] wlx3c46d8c536e6: send auth to 74:5a:aa:c1:63:80 (try 2/3)
[   59.767503] wlx3c46d8c536e6: send auth to 74:5a:aa:c1:63:80 (try 3/3)
[   59.971519] wlx3c46d8c536e6: authentication with 74:5a:aa:c1:63:80 timed out
[   60.159828] wlx3c46d8c536e6: authenticate with 90:03:25:b9:a6:e0
[   60.167658] wlx3c46d8c536e6: send auth to 90:03:25:b9:a6:e0 (try 1/3)
[... cut ...]
[   63.751551] wlx3c46d8c536e6: send auth to 90:03:25:b9:a4:c0 (try 1/3)
[   63.951852] wlx3c46d8c536e6: send auth to 90:03:25:b9:a4:c0 (try 2/3)
[   64.155857] wlx3c46d8c536e6: send auth to 90:03:25:b9:a4:c0 (try 3/3)
[   64.359870] wlx3c46d8c536e6: authentication with 90:03:25:b9:a4:c0 timed out
[   64.675745] IPv6: ADDRCONF(NETDEV_UP): wlx3c46d8c536e6: link is not ready
[   64.685702] IPv6: ADDRCONF(NETDEV_UP): wlx3c46d8c536e6: link is not ready
[   64.703400] IPv6: ADDRCONF(NETDEV_UP): wlx3c46d8c536e6: link is not ready
```

xkos, 2017-11-29:
i add 8192eu to /etc/modules-load.d/module.conf
and it also  prompt for password repeatedly

xkos, 2017-11-29:
```
  *-network
       description: Wireless interface
       physical id: 1
       bus info: usb@3:5
       logical name: wlx3c46d8c536e6
       serial: 3c:46:d8:c5:36:e6
       capabilities: ethernet physical wireless
       configuration: broadcast=yes driver=rtl8192eu multicast=yes wireless=unassociated
```

xkos, 2017-11-29:
dmesg

```
[  117.900919] RTL871X: start auth
[  117.903989] RTL871X: auth success, start assoc
[  117.939147] RTL871X: rtw_cfg80211_indicate_connect(wlx3c46d8c536e6) BSS not found !!
[  117.939152] RTL871X: assoc success
[  117.954406] RTL871X: recv eapol packet
[  117.954560] RTL871X: send eapol packet
[  117.980124] RTL871X: OnDisassoc(wlx3c46d8c536e6) reason=8, ta=90:03:25:b9:a4:c0
[  121.905256] RTL871X: nolinked power save enter
[  125.282188] RTL871X: nolinked power save leave
[  125.940767] RTL871X: nolinked power save enter
[  127.302208] RTL871X: nolinked power save leave
[  127.302282] RTL871X: rtw_set_802_11_connect(wlx3c46d8c536e6)  fw_state=0x00000008
```

michaelweghorn, 2017-11-29:
This might be the issue desribed at https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=%23839605

Does it work if you disable MAC address randomization in NetworkManager, e.g. by creating a file `/etc/NetworkManager/conf.d/30-mac-randomization.conf` with the following content?

~~~
# disable MAC randomization during Wifi scan
# as this leads to (some) USB WLAN adapters not
# being able to connect to a WLAN network,
# s. Debian bug #839605.
[device-mac-randomization]
wifi.scan-rand-mac-address=no
~~~

xkos, 2017-11-29:
i work
thank you !!

closed 2017-11-29 by xkos

hsnhrn, 2021-08-09:
Can anyone help me in understanding what does log 
[  117.939147] RTL871X: rtw_cfg80211_indicate_connect(wlx3c46d8c536e6) BSS not found !!

means?

Thank you!