Mange/rtl8192eu-linux-driver #65: Problems on making file....

Creepim, 2017-11-29:
Hi, how are you?
Im kinda fine.

Hey, anyone know what TF is going on? 

```
> make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.13.0-kali1-amd64/build M=/root/rtl8192eu-linux-driver  modules
make[1]: Entering directory '/usr/src/linux-headers-4.13.0-kali1-amd64'
  CC [M]  /root/rtl8192eu-linux-driver/core/rtw_cmd.o
In file included from /root/rtl8192eu-linux-driver/include/drv_types.h:32:0,
                 from /root/rtl8192eu-linux-driver/core/rtw_cmd.c:22:
/root/rtl8192eu-linux-driver/include/osdep_service.h: In function ‘thread_enter’:
/root/rtl8192eu-linux-driver/include/osdep_service.h:342:2: error: implicit declaration of function ‘allow_signal’ [-Werror=implicit-function-declaration]
  allow_signal(SIGTERM);
  ^~~~~~~~~~~~
/root/rtl8192eu-linux-driver/include/osdep_service.h: In function ‘flush_signals_thread’:
/root/rtl8192eu-linux-driver/include/osdep_service.h:352:6: error: implicit declaration of function ‘signal_pending’ [-Werror=implicit-function-declaration]
  if (signal_pending (current))
      ^~~~~~~~~~~~~~
/root/rtl8192eu-linux-driver/include/osdep_service.h:354:3: error: implicit declaration of function ‘flush_signals’ [-Werror=implicit-function-declaration]
   flush_signals(current);
   ^~~~~~~~~~~~~
cc1: some warnings being treated as errors
/usr/src/linux-headers-4.13.0-kali1-common/scripts/Makefile.build:307: recipe for target '/root/rtl8192eu-linux-driver/core/rtw_cmd.o' failed
make[4]: *** [/root/rtl8192eu-linux-driver/core/rtw_cmd.o] Error 1
/usr/src/linux-headers-4.13.0-kali1-common/Makefile:1529: recipe for target '_module_/root/rtl8192eu-linux-driver' failed
make[3]: *** [_module_/root/rtl8192eu-linux-driver] Error 2
Makefile:145: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
Makefile:8: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.13.0-kali1-amd64'
Makefile:1335: recipe for target 'modules' failed
make: *** [modules] Error 2

```

thats my uname -r
`4.13.0-kali1-amd64`

thats my iwconfig

```
lo        no wireless extensions.

eth0      no wireless extensions.

wlan0     IEEE 802.11  ESSID:off/any  
          Mode:Managed  Access Point: Not-Associated   Tx-Power=20 dBm   
          Retry short limit:7   RTS thr=2347 B   Fragment thr:off
          Encryption key:off
          Power Management:off
```

already tried to put ifconfig wlan0 up
no results....

Anyone?

CGarces, 2017-11-29:
The error that I see in the log [was fixed](https://github.com/Mange/rtl8192eu-linux-driver/commit/c898434525948c47fed47dca555f7db87e6a24b1) on 19 of April.
Are you compiling the last version from branch `realtek-4.4.x`?

Also note that kernel 4.13.x has reached the End Of Life, consider migrate to 4.14 that is LTS.

Creepim, 2017-11-29:
THank you for your reply! No, i didn't, also, how can i migrate? New to linux, i made it work in ubuntu, but i cant fix it here :(

CGarces, 2017-11-30:
Migrate to 4.14 is just an advice, will not solve your compilation problem.
Looks like you are trying to compile a old version of the source code.
Try to get the  last version from branch realtek-4.4.x

closed 2018-10-21 by CGarces