Mange/rtl8192eu-linux-driver #66: How to make driver work on Arch variants?

kenny-wettlaufer, 2017-12-02:
Hey everyone. I am running Artix Linux, the OpenRC arch variant, I'm trying to get my USB dongle to work but I'm not having any luck, I also didn't have luck when I was running vanilla Arch or Manjaro. 
Any ideas on how to make this thing work with Arch variants?

kenny-wettlaufer, 2017-12-02:
Oh wow speak of the devil, I just tried this thing again & now it works! That's odd... It never worked on any of my other Arch variants... maybe the new kernel 4.14 has something to do with it?

[device]
wifi.scan-rand-mac-address=no

CGarces, 2017-12-02:
I need to document the issue in the readme.md, ist's a very frequent problem.

> To fix this you need edit NetworkManager config file:
> 
> `sudo nano /etc/NetworkManager/NetworkManager.conf`
> 
> and add below 2 lines:
> 
> ```
> [device]
> wifi.scan-rand-mac-address=no
> 
> ```
> 

See
https://bugs.launchpad.net/ubuntu/+source/network-manager/+bug/1681513
And #46 #57 

Feel free to send a PR enhancing the readme.md file.

closed 2017-12-02 by CGarces