Mange/rtl8192eu-linux-driver #67: Can't connect with secured wifi networks after installing a desktop-environment

SElroy, 2017-12-06:
Hi guys, i have a little problem with the wireless adapter TP-Link TL-WN821N v5.
I tried it with this driver, and it worked at the beginning. But when I installed a desktop-environment, the adapter doesn't connect with my wifi network.

Running a VM with Debian 9, Kernel 4.9-0-4

lsusb

> **Bus 001 Device 002: ID 2357:0107**  
> Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
> Bus 002 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
> Bus 002 Device 002: ID 0e0f:0003 VMware, Inc. Virtual Mouse
> Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

I tried to connect manually with wpa_passphrase and wpa_supplicant. Before the desktop-environment it works, after the install it doens't work anymore.

> Successfully initialized wpa_supplicant
> wlan0: Trying to associate with xx:xx:xx:xx:xx:xx (SSID='WLAN-XXXX' freq=2412 MHz)
> wlan0: Associated with xx:xx:xx:xx:xx:xx
> wlan0: CTRL-EVENT-DISCONNECTED bssid=xx:xx:xx:xx:xx:xx reason=0 locally_generated=1
> wlan0: WPA: 4-Way Handshake failed - pre-shared key may be incorrect
> wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="WLAN-XXXX" auth_failures=1 duration=10 reason=WRONG_KEY
> wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
> wlan0: CTRL-EVENT-ASSOC-REJECT status_code=1
> wlan0: Reject scan trigger since one is already pending
> wlan0: Failed to initiate AP scan
> wlan0: Reject scan trigger since one is already pending
> wlan0: Failed to initiate AP scan
> wlan0: Reject scan trigger since one is already pending
> wlan0: Failed to initiate AP scan
> wlan0: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
> wlan0: CTRL-EVENT-SSID-REENABLED id=0 ssid="WLAN-XXXX"
> wlan0: Trying to associate with xx:xx:xx:xx:xx:xx (SSID='WLAN-XXXX' freq=2412 MHz)
> wlan0: Associated with xx:xx:xx:xx:xx:xx
> wlan0: CTRL-EVENT-DISCONNECTED bssid=xx:xx:xx:xx:xx:xx reason=0 locally_generated=1
> wlan0: WPA: 4-Way Handshake failed - pre-shared key may be incorrect
> wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="WLAN-XXXX" auth_failures=2 duration=23 reason=WRONG_KEY
> wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
> wlan0: CTRL-EVENT-ASSOC-REJECT status_code=1
> wlan0: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
> wlan0: CTRL-EVENT-SSID-REENABLED id=0 ssid="WLAN-XXXX"
> wlan0: Trying to associate with xx:xx:xx:xx:xx:xx (SSID='WLAN-XXXX' freq=2412 MHz)
> wlan0: Associated with xx:xx:xx:xx:xx:xx
> wlan0: CTRL-EVENT-DISCONNECTED bssid=xx:xx:xx:xx:xx:xx reason=0 locally_generated=1
> wlan0: WPA: 4-Way Handshake failed - pre-shared key may be incorrect
> wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="WLAN-XXXX" auth_failures=3 duration=46 reason=WRONG_KEY
> wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="WLAN-XXXX" auth_failures=4 duration=77 reason=CONN_FAILED
> wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
> wlan0: CTRL-EVENT-ASSOC-REJECT status_code=1
> nl80211: deinit ifname=wlan0 disabled_11b_rates=0
> wlan0: CTRL-EVENT-TERMINATING 

I recognized with ifconfig, that the MAC-address is sometimes changing.

I would be very happy, if you can help me and solve the problem.

CGarces, 2017-12-06:
I need to document the issue in the readme.md, it's a very frequent problem.

> To fix this you need edit NetworkManager config file:
> 
> `sudo nano /etc/NetworkManager/NetworkManager.conf`
> 
> and add below 2 lines:
> 
> ```
> [device]
> wifi.scan-rand-mac-address=no
> 
> ```
> 

See
https://bugs.launchpad.net/ubuntu/+source/network-manager/+bug/1681513
And #46 #57 

Feel free to send a PR enhancing the readme.md file.

SElroy, 2017-12-06:
It works, thank you very much. I wasted so much time for this problem.

closed 2017-12-06 by SElroy