Mange/rtl8192eu-linux-driver #69: new driver found

gilcu3, 2017-12-08:
Hey guys, look what I have found here:
http://tendacn.com/en/product/download/U1.html
When you download the drivers for linux, you get a folder named:
RTL8192EU_linux_v5.2.7_22739_COEX20170113-0046.20170615
with the whole driver and updated docs by realtek up to 20170615.
@masterzorag @CGarces @Mange @nomadmzl I think we have some work to do :D

CGarces, 2017-12-10:
Hi @gilcu3 
Thanks, I'll try to take a look on Christmas, but now my priority is make the current code work with kernel 4.15.

Currently is failing and I don't have time to check the solution.

See PR #70  and [compilation logs](https://travis-ci.org/Mange/rtl8192eu-linux-driver/builds/314394517?utm_source=github_status&utm_medium=notification).

Mange, 2017-12-12:
Cool. Good find!

CGarces, 2019-11-20:
Fixed by #149

closed 2019-11-20 by CGarces