Mange/rtl8192eu-linux-driver #74: Always claims that wifi password is incorrect

zaibacu, 2018-02-14:
Wifi device is being listed. It shows correct networks. However, when connecting to any of those, it claims that password is incorrect.

Some info:

Actual device: `TP-LINK TL-WN821N v6`

Kernel: `4.9.0-4-amd64`
```
  *-network:3
       description: Wireless interface
       physical id: 4
       bus info: usb@1:4
       logical name: wlx7c8bca0588c3
       serial: a2:36:97:c1:55:48
       capabilities: ethernet physical wireless
       configuration: broadcast=yes driver=rtl8192eu multicast=yes wireless=unassociated
```

shaygover, 2018-02-17:
Search this section. You need to add some line in the Network Manager config.

zaibacu, 2018-02-28:
Yes, it was solved by this issue: https://github.com/Mange/rtl8192eu-linux-driver/issues/64

closed 2018-02-28 by zaibacu