Mange/rtl8192eu-linux-driver #79: driver is noisy

aleqx, 2018-03-28:
By default, the driver fills up the log with tons of messages. Here's a 1m capture from syslog

```
Mar 28 07:14:47 felix kernel: [36720.478330] RTL871X: rtw_set_ps_mode(enx98ded01a7873) Enter 802.11 power save - WIFI-TRAFFIC_IDLE
Mar 28 07:14:47 felix kernel: [36720.480249] RTL871X: rtl8192e_set_FwPwrMode_cmd: Mode=2 SmartPS=2 UAPSD=0
Mar 28 07:14:48 felix kernel: [36721.084268] RTL871X: rtw_set_ps_mode(enx98ded01a7873) Leave 802.11 power save - WIFI-LPS_CTRL_LEAVE
Mar 28 07:14:48 felix kernel: [36721.084272] RTL871X: rtl8192e_set_FwPwrMode_cmd: Mode=0 SmartPS=2 UAPSD=0
Mar 28 07:14:48 felix kernel: [36721.085136] RTL871X: SetHwReg8192E(enx98ded01a7873): [HW_VAR_MACID_SLEEP] macid=0, org reg_0x4d4=0x00000000
Mar 28 07:14:49 felix kernel: [36722.510485] RTL871X: survey done event(29) band:0 for enx98ded01a7873
Mar 28 07:14:49 felix kernel: [36722.510511] RTL871X: rtw_indicate_scan_done(enx98ded01a7873)
Mar 28 07:14:51 felix kernel: [36724.487026] RTL871X: rtw_set_ps_mode(enx98ded01a7873) Enter 802.11 power save - WIFI-TRAFFIC_IDLE
Mar 28 07:14:51 felix kernel: [36724.487037] RTL871X: rtl8192e_set_FwPwrMode_cmd: Mode=2 SmartPS=2 UAPSD=0
Mar 28 07:15:03 felix kernel: [36736.514801] RTL871X: rtw_set_ps_mode(enx98ded01a7873) Leave 802.11 power save - WIFI-TRAFFIC_BUSY
Mar 28 07:15:03 felix kernel: [36736.514807] RTL871X: rtl8192e_set_FwPwrMode_cmd: Mode=0 SmartPS=2 UAPSD=0
Mar 28 07:15:05 felix kernel: [36738.518796] RTL871X: rtw_set_ps_mode(enx98ded01a7873) Enter 802.11 power save - WIFI-TRAFFIC_IDLE
Mar 28 07:15:05 felix kernel: [36738.518805] RTL871X: rtl8192e_set_FwPwrMode_cmd: Mode=2 SmartPS=2 UAPSD=0
Mar 28 07:15:07 felix kernel: [36740.522856] RTL871X: rtw_set_ps_mode(enx98ded01a7873) Leave 802.11 power save - WIFI-TRAFFIC_BUSY
Mar 28 07:15:07 felix kernel: [36740.522862] RTL871X: rtl8192e_set_FwPwrMode_cmd: Mode=0 SmartPS=2 UAPSD=0
Mar 28 07:15:09 felix kernel: [36742.526740] RTL871X: rtw_set_ps_mode(enx98ded01a7873) Enter 802.11 power save - WIFI-TRAFFIC_IDLE
Mar 28 07:15:09 felix kernel: [36742.526745] RTL871X: rtl8192e_set_FwPwrMode_cmd: Mode=2 SmartPS=2 UAPSD=0
Mar 28 07:15:29 felix kernel: [36762.566316] RTL871X: rtw_set_ps_mode(enx98ded01a7873) Leave 802.11 power save - WIFI-TRAFFIC_BUSY
Mar 28 07:15:29 felix kernel: [36762.566335] RTL871X: rtl8192e_set_FwPwrMode_cmd: Mode=0 SmartPS=2 UAPSD=0
Mar 28 07:15:31 felix kernel: [36764.570029] RTL871X: rtw_set_ps_mode(enx98ded01a7873) Enter 802.11 power save - WIFI-TRAFFIC_IDLE
Mar 28 07:15:31 felix kernel: [36764.570035] RTL871X: rtl8192e_set_FwPwrMode_cmd: Mode=2 SmartPS=2 UAPSD=0
Mar 28 07:15:37 felix kernel: [36770.581885] RTL871X: rtw_set_ps_mode(enx98ded01a7873) Leave 802.11 power save - WIFI-TRAFFIC_BUSY
Mar 28 07:15:37 felix kernel: [36770.581890] RTL871X: rtl8192e_set_FwPwrMode_cmd: Mode=0 SmartPS=2 UAPSD=0
Mar 28 07:15:39 felix kernel: [36772.585861] RTL871X: rtw_set_ps_mode(enx98ded01a7873) Enter 802.11 power save - WIFI-TRAFFIC_IDLE
Mar 28 07:15:39 felix kernel: [36772.585871] RTL871X: rtl8192e_set_FwPwrMode_cmd: Mode=2 SmartPS=2 UAPSD=0
Mar 28 07:15:49 felix kernel: [36782.605596] RTL871X: rtw_set_ps_mode(enx98ded01a7873) Leave 802.11 power save - WIFI-TRAFFIC_BUSY
Mar 28 07:15:49 felix kernel: [36782.605603] RTL871X: rtl8192e_set_FwPwrMode_cmd: Mode=0 SmartPS=2 UAPSD=0
```

I managed to silence the `power save` messages by loading the driver with `options 8192eu rtw_power_mgnt=0 rtw_enusbss=0 rtw_ips_mode=0` but I still get the `SetHwReg8192E(enx98ded01a7873): [HW_VAR_MACID_SLEEP]` every minute, even though it has already connected:

```
Mar 28 07:41:07 foo kernel: [ 1159.892182] RTL871X: SetHwReg8192E(enx98ded01a7873): [HW_VAR_MACID_SLEEP] macid=0, org reg_0x4d4=0x00000000
Mar 28 07:41:08 foo kernel: [ 1161.307079] RTL871X: survey done event(2d) band:0 for enx98ded01a7873
Mar 28 07:41:08 foo kernel: [ 1161.307100] RTL871X: rtw_indicate_scan_done(enx98ded01a7873)
Mar 28 07:43:07 foo kernel: [ 1279.884516] RTL871X: SetHwReg8192E(enx98ded01a7873): [HW_VAR_MACID_SLEEP] macid=0, org reg_0x4d4=0x00000000
Mar 28 07:43:08 foo kernel: [ 1281.309074] RTL871X: survey done event(2a) band:0 for enx98ded01a7873
Mar 28 07:43:08 foo kernel: [ 1281.309078] RTL871X: rtw_indicate_scan_done(enx98ded01a7873)
Mar 28 07:45:07 foo kernel: [ 1399.872746] RTL871X: SetHwReg8192E(enx98ded01a7873): [HW_VAR_MACID_SLEEP] macid=0, org reg_0x4d4=0x00000000
Mar 28 07:45:08 foo kernel: [ 1401.302292] RTL871X: survey done event(2e) band:0 for enx98ded01a7873
Mar 28 07:45:08 foo kernel: [ 1401.302296] RTL871X: rtw_indicate_scan_done(enx98ded01a7873)
```

How can I silence that too? I don't need the driver to still scan while it's already connected.

CGarces, 2019-09-23:
Hi @udartsev  and @aleqx 

Please check [this document](https://github.com/bsdfirst/rtl8814au_EW7833UAC/blob/master/Document/How_to_set_driver_debug_log_level.pdf) and give me some feedback

log level can be changed on run-time so there is no need to compile a new version

CGarces, 2019-12-09:
Should be fixed by #152

closed 2019-12-09 by CGarces