Mange/rtl8192eu-linux-driver #80: TxPower set not working ... stuck at 12 dBm

chrisvollo, 2018-04-22:
Hi, 

Thanks for putting this driver on a public place and all your work on it!

I am trying to adjust my 8192EU txpower level and it is stuck at 12 dBm ... I have tried many different commands and combinations ... any ideas on how I can set the txpower?

```
[root@alarm alarm]# iwconfig wlan0
wlan0     IEEE 802.11bgn  ESSID:off/any  
          Mode:Ad-Hoc  Frequency:2.462 GHz  Cell: Not-Associated   
          Tx-Power=12 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:on
          
[root@alarm alarm]# iw reg set BO
[root@alarm alarm]# iwconfig wlan0 txpower 20
[root@alarm alarm]# iwconfig wlan0
wlan0     IEEE 802.11bgn  ESSID:off/any  
          Mode:Ad-Hoc  Frequency:2.462 GHz  Cell: Not-Associated   
          Tx-Power=12 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:on

```

chrisvollo, 2018-04-23:
Ok, 

Looks like I may have found where it is being hardcoded to 12 dBm, and the function to set txpower is mostly #if "commented" out  ... why? How can we implement true txpower set?

./os_dep/linux/ioctl_cfg80211.c
```
static int cfg80211_rtw_set_txpower(struct wiphy *wiphy,
#if (LINUX_VERSION_CODE >= KERNEL_VERSION(3,8,0))
        struct wireless_dev *wdev,
#endif
#if (LINUX_VERSION_CODE >= KERNEL_VERSION(2,6,36)) || defined(COMPAT_KERNEL_RELEASE)
        enum nl80211_tx_power_setting type, int mbm)
#else
        enum tx_power_setting type, int dbm)
#endif
{
#if 0
        struct iwm_priv *iwm = wiphy_to_iwm(wiphy);
        int ret;

        switch (type) {
        case NL80211_TX_POWER_AUTOMATIC:
                return 0;
        case NL80211_TX_POWER_FIXED:
                if (mbm < 0 || (mbm % 100))
                        return -EOPNOTSUPP;

                if (!test_bit(IWM_STATUS_READY, &iwm->status))
                        return 0;

                ret = iwm_umac_set_config_fix(iwm, UMAC_PARAM_TBL_CFG_FIX,
                                              CFG_TX_PWR_LIMIT_USR,
                                              MBM_TO_DBM(mbm) * 2);
                if (ret < 0)
                        return ret;

                return iwm_tx_power_trigger(iwm);
        default:
                IWM_ERR(iwm, "Unsupported power type: %d\n", type);
                return -EOPNOTSUPP;
        }
#endif
        DBG_8192C("%s\n", __func__);
        return 0;
}

static int cfg80211_rtw_get_txpower(struct wiphy *wiphy,
#if (LINUX_VERSION_CODE >= KERNEL_VERSION(3,8,0))
        struct wireless_dev *wdev,
#endif
        int *dbm)
{
        DBG_8192C("%s\n", __func__);

        *dbm = (12);

        return 0;
}

```

chrisvollo, 2018-12-19:
> Hello. Has anyone managed to increase the rtl8192eu TX power? 12dBm is too
> weak from AP use.

Hi comdek,
I have not. If someone did, I would be thankful! Want to take a shot at it?
=-)

comdek, 2018-12-19:
I'll try because I made a purchase with several of this chipset hoping it would support to use as AP with 1000mw. the rtl8812au driver already supports adjusting the tx power. I think there are not many differences between these chipsets.

dotstal, 2019-04-12:
is there a movement? or all live with this TX power 12dBm ?

CGarces, 2019-09-05:
Hi  @dotstal  @comdek @chrisvollo 
Can you compile the driver with PR #136 and test it?

chrisvollo, 2019-09-08:
Did you add the tx power adjustment,?

commit 89a8c35a (2019-09-10): Merge pull request #136 from mbuzdalov/realtek-4.4.x

Enable power management.
Close #80
closed 2019-09-10 by CGarces