Mange/rtl8192eu-linux-driver #83: error: Error for wireless request "Set Mode" (8B06)

ricardofcostareis, 2018-06-07:
When I try use monitor mode issue below appears:

root@kali:~/Downloads/2# iwconfig wlan0 mode monitor
**Error for wireless request "Set Mode" (8B06) :
    SET failed on device wlan0 ; Operation not permitted.**

Configurations:

root@kali:~/Downloads/2# uname -r
4.15.0-kali3-amd64

root@kali:~/Downloads/2# iwconfig
lo        no wireless extensions.

wlan0     unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

eth0      no wireless extensions.

oot@kali:~/Downloads/2# iw list | grep "Supported interface modes" -A10
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * monitor
		 * P2P-client
		 * P2P-GO
	Band 1:
		Bitrates (non-HT):
			* 1.0 Mbps
			* 2.0 Mbps

brkuhgk, 2019-03-03:
did u found the solution ..

albaluz35, 2024-05-16:
In my case enabling monitor mode and doing packet injection WORKS! as follows:

sudo apt update
sudo apt upgrade
sudo apt-get install linux-headers-`uname -r` build-essential dkms
git clone https://github.com/Mange/rtl8192eu-linux-driver.git
cd rtl8192eu-linux
sed -i -E 's/CONFIG_WIFI_MONITOR\s*=\s*n/CONFIG_WIFI_MONITOR = y/' Makefile

sudo rmmod 8192eu rtl8xxxu r8188eu
echo "blacklist rtl8xxxu" >> ./blacklist-rtl8xxxu.conf
sudo mv ./blacklist-rtl8xxxu.conf /etc/modprobe.d/
sudo dkms add .
sudo dkms install rtl8192eu/1.0

-----------------------

**IF THERE ARE PROBLEMS WITH dkms, uninstall with dkms, then execute make commands manually:**
\# sudo make
\# sudo make install

-----------------

sudo modprobe 8192eu
sudo lsmod |grep 8192eu                                                     

!To enable Monitor mode and test packet injection:!
=================================================

WL=wlan0
sudo iwconfig
sudo ifconfig $WL down
sudo airmon-ng check kill
sudo iwconfig $WL mode monitor
sudo ifconfig $WL up
sudo iwconfig

\# Wait for a moment and retry if there are issues or reboot and try the "
!To enable Monitor mode and test packet injection:! " steps
sudo aireplay-ng --test $WL