Mange/rtl8192eu-linux-driver #88: Monitor mode doesn't work and when killed, all freezes

RetardedChease, 2018-06-29:
So, I bought new usb wireless card tl-wn822n v5 with Realtek chip. Apparently it uses rtl8192eu driver which i downloaded from github [](https://github.com/Mange/rtl8192eu-linux-driver). When installed and rebooted, card works normally, you can access wireless networks, connect and disconnect from them. But when I try starting monitor mode with airmon-ng and scanning with airodump-ng, it shows no wireless networks. Airmon-ng renames my card for monitor mode 'cause the name is too long. Normally, card's name is wlx503eaa3b1bbc and after starting airmon-ng it renames it to wlan0mon, where should be active monitor mode. When I try killing that monitor interface with airmon-ng stop wlan0mon it says;
`Found phy1 with no interfaces assigned, would you like to assign one to it? [y/n] n
PHY phy1 will remain lost.

PHY   Interface   Driver      Chipset

ethtool failed...
Only mac80211 devices on kernel 2.6.33 or higher are officially supported by airmon-ng.`
airmon-ng check kill does nothing, too
So, i can't use my wifi card until i reboot my laptop. And if I just plug the USB out, the next command that I run in terminal will freez everything. Trying to start monitor mode with iwconfig does pretty much the same thing. I tried downloading the original driver from tp link site, but that gives me error when I do make.
I know this card supports monitor mode 'cause before I installed this driver that I'm having now, I installed driver with the same name but from somewhere else. With that driver monitor mod was working normally just the signal strength was terrible so I had to remove that driver.
So, does anybody have something in mind?
I bought that card with intention to use it for monitoring network and sending some DeAuth packages.
And when Fluxion started, it does not offer selecting that card to use it. I know that Fluxion supports USB card because I have another USB wireless card that works normally.
Help?

DanStos, 2018-07-17:
This is also happening to me. I'm using a WN823N v6... After I send "iwconfig wlan0 mode monitor", the only status the terminal sends to me is "Killed". After that, I can't send any iw or ifconfig command. :'(

daedroza, 2018-10-26:
IMHO, you need a Atheros based WiFi adapter to change it to monitor mode & vice versa.

victorperin, 2019-03-26:
I'm having the same issues in here, with `TL-WN8200ND(UN) ver. 2.0`

CGarces, 2019-12-12:
Please try
`sudo iw dev wlan0 set type monitor` with the last version of the driver.

Reopen the issue if needed

closed 2019-12-12 by CGarces