Mange/rtl8192eu-linux-driver #89: Not support ARMv8 Arch?

CodeNemo, 2018-07-01:
Trying on Raspberry Pi 3B, ARMv8 Arch Linux (Linux 4.17.3) installed. Makefile modified, but both dkms install and make get the same error:
> ARCH=arm CROSS_COMPILE= -C /lib/modules/4.17.3-1-ARCH/build M=/var/lib/dkms/rtl8192eu/1.0/build  modules
> make[1]: Entering directory '/usr/lib/modules/4.17.3-1-ARCH/build'
> Makefile:637: arch/arm/Makefile: No such file or directory
> make[1]: *** No rule to make target 'arch/arm/Makefile'.  Stop.
> make[1]: Leaving directory '/usr/lib/modules/4.17.3-1-ARCH/build'
> make: *** [Makefile:1700: modules] Error 2
Same error on openSUSE Leap 15.0 (Linux 4.15). Any solution?

CGarces, 2018-07-01:
The code support ARM arch without problems, it's used on LibreElec builds.
 ¿Are you sure that you have the correct sources to compile the module?

CodeNemo, 2018-07-14:
Accidentally found that I should specify ARCH=arm64 but not arm!

closed 2018-07-14 by CodeNemo