Mange/rtl8192eu-linux-driver #90: Ubuntu 18.04 4.15.0-23-generic

rom1504, 2018-07-07:
Hi,
I upgraded to the last ubuntu (18.04) and the adapter got really slow (80Mb/s before to 10Mb/s after)
Do you know if there's something to upgrade about this here ?

daedroza, 2018-10-17:
I can confirm.
Is there a way around this?

daedroza, 2018-10-17:
I was able to solve this problem by blacklisting the rtl8xx2 module & forcing 8192eu module to start at boot.
Really thanks & grateful for this driver @CGarces

rom1504, 2018-10-20:
@daedroza could you provide instructions on how to do that ? it would really help !

daedroza, 2018-10-21:
### Install RTL8192EU Driver
`git clone https://github.com/Mange/rtl8192eu-linux-driver;`
`cd rtl8192eu-linux-driver;`
`sudo dkms add .;`
`sudo dkms install rtl8192eu/1.0;`

### Blacklist RTL8XXXU Driver
`echo "blacklist rtl8xxxu" | sudo tee /etc/modprobe.d/rtl8xxxu.conf;`

### Resolve Idle PM issues (Check issue 94)
`echo "options 8192eu rtw_power_mgnt=0 rtw_enusbss=0" | sudo tee /etc/modprobe.d/8192eu.conf;`

### Force RTL8192EU Driver to be active at boot
`echo -e "8192eu\n\nloop" | sudo tee /etc/modules;`

### Update changes (effective on next reboot)
`sudo update-grub; sudo update-initramfs -u;`

CGarces, 2018-10-21:
Thanks @daedroza, fell free to send a PR updating the readme with your instructions

closed 2018-10-21 by CGarces

rom1504, 2018-10-21:
@daedroza I followed your instructions and indeed it's now back to normal speeds. Thank you!