Mange/rtl8192eu-linux-driver #93: Can I run HostAPD with this driver?

CodeNemo, 2018-07-14:
Driver already installed, rtl8xxxu blacklisted.
> *-network:0               
>        description: Wireless interface
>        physical id: 2
>        bus info: usb@1:1.2
>        logical name: wlan0
>        serial: 
>        capabilities: ethernet physical wireless
>        configuration: broadcast=yes driver=rtl8192eu ip=10.69.1.1 multicast=yes wireless=unassociated
But when I tried to start HostAPD, I got the errors:
> nl80211: Could not configure driver mode
> nl80211: deinit ifname=wlan0 disabled_11b_rates=0
> hostapd[359]: nl80211 driver initialization failed.
> hostapd[359]: wlan0: interface state UNINITIALIZED->DISABLED
> hostapd[359]: wlan0: AP-DISABLED
Doesn't it supported by nl80211?

closed 2018-07-15 by CodeNemo

fcracker79, 2019-01-27:
Indeed I receive an error when using Adafruit's hostapd:

```
root@raspberrypi:~/rtl8192eu-linux-driver# sudo /usr/sbin/hostapd.adafruit /etc/hostapd/hostapd.conf
Configuration file: /etc/hostapd/hostapd.conf
Could not read interface wlan0 flags: No such device
ioctl(SIOCGIFINDEX): No such device
rtl871xdrv driver initialization failed.
root@raspberrypi:~/rtl8192eu-linux-driver# modprobe 8192cu
root@raspberrypi:~/rtl8192eu-linux-driver# sudo /usr/sbin/hostapd.adafruit /etc/hostapd/hostapd.conf
Configuration file: /etc/hostapd/hostapd.conf
Could not read interface wlan0 flags: No such device
ioctl(SIOCGIFINDEX): No such device
rtl871xdrv driver initialization failed.
root@raspberrypi:~/rtl8192eu-linux-driver# modprobe 8192
modprobe: FATAL: Module 8192 not found in directory /lib/modules/4.14.79+
root@raspberrypi:~/rtl8192eu-linux-driver# modprobe 8192eu
root@raspberrypi:~/rtl8192eu-linux-driver# sudo /usr/sbin/hostapd.adafruit /etc/hostapd/hostapd.conf
Configuration file: /etc/hostapd/hostapd.conf
drv->ifindex=4
l2_sock_recv==l2_sock_xmit=0x0x4e3a40
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Could not connect to kernel driver.
+rtl871x_sta_deauth_ops, ff:ff:ff:ff:ff:ff is deauth, reason=2
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Failed to set encryption.
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Failed to set encryption.
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Failed to set encryption.
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Failed to set encryption.
Using interface wlan0 with hwaddr 00:e1:b0:11:57:32 and ssid 'TOR'
rtl871x_set_wps_assoc_resp_ie
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Failed to set encryption.
rtl871x_set_beacon_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
rtl871x_set_hidden_ssid ignore_broadcast_ssid:0, TOR,3
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
rtl871x_set_acl
Failed to set beacon head/tail or DTIM period
^Cioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Could not connect to kernel driver.
+rtl871x_sta_deauth_ops, ff:ff:ff:ff:ff:ff is deauth, reason=2
ELOOP: remaining socket: sock=6 eloop_data=0x4e3ab0 user_data=(nil) handler=0x2a000
```

Any chances to make it work as an AP?
Thanks

udf, 2019-06-17:
It seems that the driver does support hostapd's `driver=nl80211` but you have to explicitly enable it in the makefile (it's enabled for several architectures, but left out for some like aarch64, not sure why).

You just have to put this line inside the `ifeq` block:

`EXTRA_CFLAGS += -DCONFIG_IOCTL_CFG80211 -DRTW_USE_CFG80211_STA_EVENT`

for example like this:
```diff
ifeq ($(CONFIG_PLATFORM_ARM_AARCH64), y)
EXTRA_CFLAGS += -DCONFIG_LITTLE_ENDIAN
+EXTRA_CFLAGS += -DCONFIG_IOCTL_CFG80211 -DRTW_USE_CFG80211_STA_EVENT
ARCH := arm64
CROSS_COMPILE :=
KVER  := $(shell uname -r)
KSRC ?= /lib/modules/$(KVER)/build
MODDESTDIR := /lib/modules/$(KVER)/kernel/drivers/net/wireless/
endif
```

over here:

https://github.com/Mange/rtl8192eu-linux-driver/blob/d6aa91a72870185f71d5956cddb6f333537fbfa9/Makefile#L947-L954