Mange/rtl8192eu-linux-driver #94: Device Need to unplug and replug to USB for connection to work on Ubuntu 18.04

devNaresh, 2018-07-19:
I am using TP-link TL823N V2 which is based on rtl8192eu wiresless adapter. Everytime I start the computer, the wifi is stopped from connecting with the message Connection Failed (Activation of network connection failed), even though it shows available networks. It connects after I unplug and replug the wifi adapter. The only problem I have that each time I start the pc, I have to unplug and replug my adapter from the back of the CPU Cabinet. Any idea

devNaresh, 2018-07-25:
Below command solved the issue for me.

`echo "options 8192eu rtw_power_mgnt=0 rtw_enusbss=0" | sudo tee /etc/modprobe.d/8192eu.conf
echo "blacklist rtl8xxxu" | sudo tee /etc/modprobe.d/rtl8xxxu.conf`

pdimh, 2018-08-29:
Thanks for the tip, i have the same device and had the same problem.

CGarces, 2018-08-30:
Issue solved by https://github.com/Mange/rtl8192eu-linux-driver/issues/94#issuecomment-407698097

closed 2018-08-30 by CGarces