Mange/rtl8192eu-linux-driver #95: WPA2 does not work (Debian9, )

RobTranquillo, 2018-07-20:
Hi and thank you very much for your repo. I know, you are not responsible for the driver, but maybe you know some help or somebody reading that and can help.

i compile it on two machines with
> Debian 9, intel Core i5 / i7
> Linux 4.9.0-7-amd64 #1 SMP Debian 4.9.110-1 (2018-07-05) x86_64 GNU/Linux 

The compile and installation was quite simple and works like expected. But connecting to a WPA2 secured network deas not work. Connecting to a unsecured network is no problem and it works normal.

michaelweghorn, 2018-07-20:
In case you're using NetworkManager (the default tool for network configuration): Can you please try whether the config change described e.g. in https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=839605#35 makes things work for you, i.e. disabling device-mac-randomization in NetworkManager?

CGarces, 2018-10-21:
Looks like a duplicate os

CGarces, 2018-10-21:
Looks like a duplicate  of #64
See fix of `/etc/NetworkManager/conf.d/30-mac-randomization.conf` inside the comments.

closed 2018-10-21 by CGarces