Mange/rtl8192eu-linux-driver #104: Frequent network traffic delays

soulseekah, 2018-10-24:
I have a D-Link DWA 131/E nano adapter.
Kernel version 4.15.0-36
Xubuntu 16.04.5

configuration: broadcast=yes driver=rtl8192eu ip=192.168.0.103 multicast=yes wireless=IEEE 802.11bgn

I am experiencing very frequent weird global network hangs that can be seen via ping:

```
64 bytes from ya.ru (87.250.250.242): icmp_seq=139 ttl=53 time=18.4 ms
64 bytes from ya.ru (87.250.250.242): icmp_seq=140 ttl=53 time=17.4 ms
64 bytes from ya.ru (87.250.250.242): icmp_seq=141 ttl=53 time=19.0 ms
64 bytes from ya.ru (87.250.250.242): icmp_seq=142 ttl=53 time=24.2 ms
64 bytes from ya.ru (87.250.250.242): icmp_seq=143 ttl=53 time=16.8 ms
64 bytes from ya.ru (87.250.250.242): icmp_seq=145 ttl=53 time=11598 ms
64 bytes from ya.ru (87.250.250.242): icmp_seq=147 ttl=53 time=9552 ms
64 bytes from ya.ru (87.250.250.242): icmp_seq=144 ttl=53 time=12622 ms
64 bytes from ya.ru (87.250.250.242): icmp_seq=146 ttl=53 time=10578 ms
64 bytes from ya.ru (87.250.250.242): icmp_seq=149 ttl=53 time=7509 ms
64 bytes from ya.ru (87.250.250.242): icmp_seq=152 ttl=53 time=4438 ms
64 bytes from ya.ru (87.250.250.242): icmp_seq=154 ttl=53 time=2390 ms
64 bytes from ya.ru (87.250.250.242): icmp_seq=155 ttl=53 time=1368 ms
64 bytes from ya.ru (87.250.250.242): icmp_seq=148 ttl=53 time=8536 ms
64 bytes from ya.ru (87.250.250.242): icmp_seq=150 ttl=53 time=6489 ms
64 bytes from ya.ru (87.250.250.242): icmp_seq=151 ttl=53 time=5471 ms
64 bytes from ya.ru (87.250.250.242): icmp_seq=153 ttl=53 time=3423 ms
64 bytes from ya.ru (87.250.250.242): icmp_seq=156 ttl=53 time=19.0 ms
64 bytes from ya.ru (87.250.250.242): icmp_seq=157 ttl=53 time=17.3 ms
64 bytes from ya.ru (87.250.250.242): icmp_seq=158 ttl=53 time=17.4 ms
```

In the above particular case `ping` freezes for 11 seconds then outputs 11 responses in one go. It's sort of as if it kept on piling up, not getting responses and then once the first ICMP got the response the rest unblocked as well. How can this be? I was sure that `ping` would wait for a response before sending the next packet. This happens every 200 ICMP packets or so.

Here's tcpdump output for the above ping command:

```
15:44:09.409112 IP 192.168.0.103 > ya.ru: ICMP echo request, id 7528, seq 139, length 64
15:44:09.427531 IP ya.ru > 192.168.0.103: ICMP echo reply, id 7528, seq 139, length 64
15:44:10.410603 IP 192.168.0.103 > ya.ru: ICMP echo request, id 7528, seq 140, length 64
15:44:10.427990 IP ya.ru > 192.168.0.103: ICMP echo reply, id 7528, seq 140, length 64
15:44:11.412048 IP 192.168.0.103 > ya.ru: ICMP echo request, id 7528, seq 141, length 64
15:44:11.431081 IP ya.ru > 192.168.0.103: ICMP echo reply, id 7528, seq 141, length 64
15:44:12.413158 IP 192.168.0.103 > ya.ru: ICMP echo request, id 7528, seq 142, length 64
15:44:12.437364 IP ya.ru > 192.168.0.103: ICMP echo reply, id 7528, seq 142, length 64
15:44:13.414440 IP 192.168.0.103 > ya.ru: ICMP echo request, id 7528, seq 143, length 64
15:44:13.431237 IP ya.ru > 192.168.0.103: ICMP echo reply, id 7528, seq 143, length 64
15:44:14.416331 IP 192.168.0.103 > ya.ru: ICMP echo request, id 7528, seq 144, length 64
15:44:15.438216 IP 192.168.0.103 > ya.ru: ICMP echo request, id 7528, seq 145, length 64
15:44:16.462199 IP 192.168.0.103 > ya.ru: ICMP echo request, id 7528, seq 146, length 64
15:44:17.486193 IP 192.168.0.103 > ya.ru: ICMP echo request, id 7528, seq 147, length 64
15:44:18.510335 IP 192.168.0.103 > ya.ru: ICMP echo request, id 7528, seq 148, length 64
15:44:19.534215 IP 192.168.0.103 > ya.ru: ICMP echo request, id 7528, seq 149, length 64
15:44:20.558217 IP 192.168.0.103 > ya.ru: ICMP echo request, id 7528, seq 150, length 64
15:44:21.582585 IP 192.168.0.103 > ya.ru: ICMP echo request, id 7528, seq 151, length 64
15:44:22.606193 IP 192.168.0.103 > ya.ru: ICMP echo request, id 7528, seq 152, length 64
15:44:23.630224 IP 192.168.0.103 > ya.ru: ICMP echo request, id 7528, seq 153, length 64
15:44:24.654352 IP 192.168.0.103 > ya.ru: ICMP echo request, id 7528, seq 154, length 64
15:44:25.678218 IP 192.168.0.103 > ya.ru: ICMP echo request, id 7528, seq 155, length 64
15:44:27.036888 IP ya.ru > 192.168.0.103: ICMP echo reply, id 7528, seq 145, length 64
15:44:27.039031 IP ya.ru > 192.168.0.103: ICMP echo reply, id 7528, seq 147, length 64
15:44:27.039033 IP ya.ru > 192.168.0.103: ICMP echo reply, id 7528, seq 144, length 64
15:44:27.041170 IP ya.ru > 192.168.0.103: ICMP echo reply, id 7528, seq 146, length 64
15:44:27.043730 IP ya.ru > 192.168.0.103: ICMP echo reply, id 7528, seq 149, length 64
15:44:27.045044 IP ya.ru > 192.168.0.103: ICMP echo reply, id 7528, seq 152, length 64
15:44:27.045046 IP ya.ru > 192.168.0.103: ICMP echo reply, id 7528, seq 154, length 64
15:44:27.046325 IP ya.ru > 192.168.0.103: ICMP echo reply, id 7528, seq 155, length 64
15:44:27.046327 IP ya.ru > 192.168.0.103: ICMP echo reply, id 7528, seq 148, length 64
15:44:27.047897 IP ya.ru > 192.168.0.103: ICMP echo reply, id 7528, seq 150, length 64
15:44:27.054023 IP ya.ru > 192.168.0.103: ICMP echo reply, id 7528, seq 151, length 64
15:44:27.054025 IP ya.ru > 192.168.0.103: ICMP echo reply, id 7528, seq 153, length 64
[... 32 lines cut ...]
ntrollen=0, msg_flags=0}, 0) = 64
recvmsg(3, 0x7ffc4fea9580, 0)           = -1 EAGAIN (Resource temporarily unavailable)
sendmsg(3, {msg_name(16)={sa_family=AF_INET, sin_port=htons(0), sin_addr=inet_addr("87.250.250.242")}, msg_iov(1)=[{"\10\0v\370F6\0;\213j\320[\0\0\0\0\21\375\16\0\0\0\0\0\20\21\22\23\24\25\26\27"..., 64}], msg_con
trollen=0, msg_flags=0}, 0) = 64
recvmsg(3, 0x7ffc4fea9580, 0)           = -1 EAGAIN (Resource temporarily unavailable)
sendmsg(3, {msg_name(16)={sa_family=AF_INET, sin_port=htons(0), sin_addr=inet_addr("87.250.250.242")}, msg_iov(1)=[{"\10\0004\334F6\0<\215j\320[\0\0\0\0`\30\0\0\0\0\0\0\20\21\22\23\24\25\26\27"..., 64}], msg_contr
ollen=0, msg_flags=0}, 0) = 64
recvmsg(3, 0x7ffc4fea9580, 0)           = -1 EAGAIN (Resource temporarily unavailable)
sendmsg(3, {msg_name(16)={sa_family=AF_INET, sin_port=htons(0), sin_addr=inet_addr("87.250.250.242")}, msg_iov(1)=[{"\10\0002mF6\0=\216j\320[\0\0\0\0a\206\0\0\0\0\0\0\20\21\22\23\24\25\26\27"..., 64}], msg_control
len=0, msg_flags=0}, 0) = 64
recvmsg(3, 0x7ffc4fea9580, 0)           = -1 EAGAIN (Resource temporarily unavailable)
sendmsg(3, {msg_name(16)={sa_family=AF_INET, sin_port=htons(0), sin_addr=inet_addr("87.250.250.242")}, msg_iov(1)=[{"\10\0 \36F6\0>\217j\320[\0\0\0\0r\324\0\0\0\0\0\0\20\21\22\23\24\25\26\27"..., 64}], msg_control
len=0, msg_flags=0}, 0) = 64
recvmsg(3, 0x7ffc4fea9580, 0)           = -1 EAGAIN (Resource temporarily unavailable)
sendmsg(3, {msg_name(16)={sa_family=AF_INET, sin_port=htons(0), sin_addr=inet_addr("87.250.250.242")}, msg_iov(1)=[{"\10\0\357\277F6\0?\220j\320[\0\0\0\0\2411\1\0\0\0\0\0\20\21\22\23\24\25\26\27"..., 64}], msg_con
trollen=0, msg_flags=0}, 0) = 64
recvmsg(3, 0x7ffc4fea9580, 0)           = -1 EAGAIN (Resource temporarily unavailable)
sendmsg(3, {msg_name(16)={sa_family=AF_INET, sin_port=htons(0), sin_addr=inet_addr("87.250.250.242")}, msg_iov(1)=[{"\10\0\233`F6\0@\221j\320[\0\0\0\0\364\217\1\0\0\0\0\0\20\21\22\23\24\25\26\27"..., 64}], msg_con
trollen=0, msg_flags=0}, 0) = 64
recvmsg(3, {msg_name(16)={sa_family=AF_INET, sin_port=htons(0), sin_addr=inet_addr("87.250.250.242")}, msg_iov(1)=[{"E\0\0T\206\311\0\0005\1\352\343W\372\372\362\300\250\0g\0\0\272sF6\0007\207j\320["..., 192}], ms
g_controllen=32, [{cmsg_len=32, cmsg_level=SOL_SOCKET, cmsg_type=0x1d /* SCM_??? */}], msg_flags=0}, 0) = 84
write(1, "64 bytes from ya.ru (87.250.250."..., 7064 bytes from ya.ru (87.250.250.242): icmp_seq=55 ttl=53 time=9998 ms
```

Looks like `EAGAIN (Resource temporarily unavailable)` during socket reads is the underlying symptom. There's nothing in dmesg or syslog during this phenomenon. How do I provide more debug information at a lower level?

CGarces, 2018-10-24:
I'm not sure if is related to  #90 
Can you please follow the instruction [there](https://github.com/Mange/rtl8192eu-linux-driver/issues/90#issuecomment-431636643) and tell me if force RTL8192EU to be active solve your problems?.

soulseekah, 2018-10-24:
Thanks for your response, @CGarces.
I had the issue described in #90 before I updated to this driver.
I followed all the instructions to blacklist the xxx module and force use rtl8192eu, all of those steps as in the README.md. The rtl8192eu module is definitely force-loaded and used.

soulseekah, 2018-10-24:
Here's a boot log:

```
[    1.433158] r8169 0000:21:00.0 eth0: RTL8168h/8111h at 0x        (ptrval), 30:9c:23:08:a8:b0, XID 14100800 IRQ 69
[    5.727411] RTL871X: module init start
[    5.727413] RTL871X: rtl8192eu v4.4.1_17696.20160509_BTCOEX20160412-0042
[    5.727413] RTL871X: build time: Oct 23 2018 17:47:06
[    5.727413] RTL871X: rtl8192eu BT-Coex version = BTCOEX20160412-0042
[    9.572965] RTL871X: hal_com_config_channel_plan chplan:0x21
[    9.657262] RTL871X: Wrong rate section: (CCK,2T)
[    9.657263] RTL871X: Wrong rate section: (CCK,2T)
[    9.657264] RTL871X: Wrong rate section: (CCK,2T)
[    9.657264] RTL871X: Wrong rate section: (CCK,2T)
[    9.657265] RTL871X: Wrong rate section: (CCK,2T)
[    9.657265] RTL871X: Wrong rate section: (CCK,2T)
[    9.657266] RTL871X: Wrong rate section: (CCK,2T)
[    9.657266] RTL871X: Wrong rate section: (CCK,2T)
[    9.657267] RTL871X: Wrong rate section: (CCK,2T)
[    9.657267] RTL871X: Wrong rate section: (CCK,2T)
[    9.657267] RTL871X: Wrong rate section: (CCK,2T)
[    9.657268] RTL871X: Wrong rate section: (CCK,2T)
[    9.657268] RTL871X: Wrong rate section: (CCK,2T)
[    9.657269] RTL871X: Wrong rate section: (CCK,2T)
[    9.657269] RTL871X: Wrong rate section: (CCK,2T)
[    9.657269] RTL871X: Wrong rate section: (CCK,2T)
[    9.657270] RTL871X: Wrong rate section: (CCK,2T)
[    9.657270] RTL871X: Wrong rate section: (CCK,2T)
[    9.657271] RTL871X: Wrong rate section: (CCK,2T)
[    9.657271] RTL871X: Wrong rate section: (CCK,2T)
[    9.657271] RTL871X: Wrong rate section: (CCK,2T)
[    9.657272] RTL871X: Wrong rate section: (CCK,2T)
[    9.657272] RTL871X: Wrong rate section: (CCK,2T)
[    9.657273] RTL871X: Wrong rate section: (CCK,2T)
[    9.657273] RTL871X: Wrong rate section: (CCK,2T)
[    9.657273] RTL871X: Wrong rate section: (CCK,2T)
[    9.657274] RTL871X: Wrong rate section: (CCK,2T)
[    9.657274] RTL871X: Wrong rate section: (CCK,2T)
[    9.657275] RTL871X: Wrong rate section: (CCK,2T)
[    9.657275] RTL871X: Wrong rate section: (CCK,2T)
[    9.657276] RTL871X: Wrong rate section: (CCK,2T)
[    9.657276] RTL871X: Wrong rate section: (CCK,2T)
[    9.657276] RTL871X: Wrong rate section: (CCK,2T)
[    9.657277] RTL871X: Wrong rate section: (CCK,2T)
[    9.657277] RTL871X: Wrong rate section: (CCK,2T)
[    9.657278] RTL871X: Wrong rate section: (CCK,2T)
[    9.657278] RTL871X: Wrong rate section: (CCK,2T)
[    9.657278] RTL871X: Wrong rate section: (CCK,2T)
[    9.657279] RTL871X: Wrong rate section: (CCK,2T)
[    9.657279] RTL871X: Wrong rate section: (CCK,2T)
[    9.657280] RTL871X: Wrong rate section: (CCK,2T)
[    9.657280] RTL871X: Wrong rate section: (CCK,2T)
[    9.657284] RTL871X: Wrong rate section: (OFDM,2T)
[    9.657284] RTL871X: Wrong rate section: (OFDM,2T)
[    9.657285] RTL871X: Wrong rate section: (OFDM,2T)
[    9.657285] RTL871X: Wrong rate section: (OFDM,2T)
[    9.657285] RTL871X: Wrong rate section: (OFDM,2T)
[    9.657286] RTL871X: Wrong rate section: (OFDM,2T)
[    9.657286] RTL871X: Wrong rate section: (OFDM,2T)
[    9.657287] RTL871X: Wrong rate section: (OFDM,2T)
[    9.657287] RTL871X: Wrong rate section: (OFDM,2T)
[    9.657287] RTL871X: Wrong rate section: (OFDM,2T)
[    9.657288] RTL871X: Wrong rate section: (OFDM,2T)
[    9.657288] RTL871X: Wrong rate section: (OFDM,2T)
[    9.657289] RTL871X: Wrong rate section: (OFDM,2T)
[    9.657289] RTL871X: Wrong rate section: (OFDM,2T)
[    9.657289] RTL871X: Wrong rate section: (OFDM,2T)
[    9.657290] RTL871X: Wrong rate section: (OFDM,2T)
[    9.657290] RTL871X: Wrong rate section: (OFDM,2T)
[    9.657291] RTL871X: Wrong rate section: (OFDM,2T)
[    9.657291] RTL871X: Wrong rate section: (OFDM,2T)
[... 93 lines cut ...]
[   51.539150] RTL871X: rtw_set_802_11_connect(wlx180f7609e54e)  fw_state=0x00000008
[   58.553872] RTL871X: rtw_set_802_11_connect(wlx180f7609e54e)  fw_state=0x00000808
[   59.495609] RTL871X: start auth
[   59.498121] RTL871X: auth success, start assoc
[   59.507003] RTL871X: rtw_cfg80211_indicate_connect(wlx180f7609e54e) BSS not found !!
[   59.507009] RTL871X: assoc success
[   59.607344] RTL871X: recv eapol packet
[   59.607526] RTL871X: send eapol packet
[   59.613682] RTL871X: recv eapol packet
[   59.613741] RTL871X: send eapol packet
[  105.057574] RTL871X: OnDeAuth(wlx180f7609e54e) reason=6, ta=b0:4e:26:a2:2f:48, ignore=0
[  113.128289] RTL871X: rtw_set_802_11_connect(wlx180f7609e54e)  fw_state=0x00000008
[  114.276459] RTL871X: start auth
[  114.279393] RTL871X: auth success, start assoc
[  114.286922] RTL871X: rtw_cfg80211_indicate_connect(wlx180f7609e54e) BSS not found !!
[  114.286932] RTL871X: assoc success
[  114.385279] RTL871X: recv eapol packet
[  114.385473] RTL871X: send eapol packet
[  115.385122] RTL871X: recv eapol packet
[  115.385257] RTL871X: send eapol packet
[  115.391607] RTL871X: recv eapol packet
[  115.391694] RTL871X: send eapol packet
[  115.391751] RTL871X: set pairwise key camid:4, addr:b0:4e:26:a2:2f:48, kid:0, type:AES
[  115.445512] RTL871X: set group key camid:5, addr:b0:4e:26:a2:2f:48, kid:2, type:TKIP
```

daedroza, 2018-10-26:
@soulseekah : 

Boot your computer & disconnect/connect WiFi adapter.
I'm still finding why WiFi adapter misbehaves on Power Management (leave it as is for 3-4 hours, WiFi adapter connects/disconnects constantly to WiFi SSID).

I'm searching online for other drivers, found this,
EXTRA_CFLAGS += -DCONFIG_IOCTL_CFG80211 -DRTW_USE_CFG80211_STA_EVENT

We need to simply enable CONFIG_IOCTL_CFG80211 & RTW_USE_CFG80211_STA_EVENT, meaning you should comment out the above line like this

#EXTRA_CFLAGS += -DCONFIG_IOCTL_CFG80211 -DRTW_USE_CFG80211_STA_EVENT

Reference: https://github.com/abperiasamy/rtl8812AU_8821AU_linux/issues/144

daedroza, 2018-10-26:
Okay, here is the symptoms I've:

1. Let your PC/Adapter idle for some time (30-60 minutes).
During this, the Adapter has this in the dmesg:
`[    9.544298] RTL871X: auth success, start assoc
[    9.551727] RTL871X: rtw_cfg80211_indicate_connect(wlan0) BSS not found !!
[    9.551742] RTL871X: assoc success
[    9.652503] RTL871X: recv eapol packet
[    9.652715] RTL871X: send eapol packet
[    9.675324] RTL871X: recv eapol packet
[    9.675472] RTL871X: send eapol packet
[    9.675545] RTL871X: set pairwise key camid:4, addr:xxxxx:84:ae, kid:0, type:AES
[    9.676281] RTL871X: set group key camid:5, addr:xxxxx:84:ae, kid:1, type:AES
`
2. Go to http://www.speedtest.net/
3. It should FAIL. 
During this, the Adapter has this in the dmesg:
`[ 1530.543422] RTL871X: recv eapol packet
[ 1530.543967] RTL871X: set group key camid:6, addr:xxxxx:84:ae, kid:2, type:AES
[ 1530.543969] RTL871X: send eapol packet
[ 5142.279089] RTL871X: recv eapol packet
[ 5142.279582] RTL871X: set group key camid:5, addr:xxxxx:84:ae, kid:1, type:AES
[ 5142.279584] RTL871X: send eapol packet
`
4. Reconnect WiFi Adapter.
`[ 5231.972918] RTL871X: start auth
[ 5231.976291] RTL871X: auth success, start assoc
[ 5231.983824] RTL871X: rtw_cfg80211_indicate_connect(wlan0) BSS not found !!
[ 5231.983834] RTL871X: assoc success
[ 5232.084068] RTL871X: recv eapol packet
[ 5232.084648] RTL871X: send eapol packet
[ 5232.090459] RTL871X: recv eapol packet
[ 5232.090735] RTL871X: send eapol packet
[ 5232.090901] RTL871X: set pairwise key camid:4, addr:xxxxx::84:ae, kid:0, type:AES
[ 5232.091807] RTL871X: set group key camid:5, addr:xxxxx::84:ae, kid:1, type:AES`
`

The difference is kid:0, kid:1 which is when I was able to test speed.
After idling, it is kid:2, kid:1 which failed miserably on speed test.
After reconnecting WiFi adapter, it was set again to kid:0, kid:1

I will try debugging further, see what I can get.

soulseekah, 2018-10-26:
Thanks for helping me look into this. The issue happens all the time, without idling, every couple of minutes it all sockets seem to stop being able to read with EAGAIN for about 10 seconds, then bam the buffered data is available again.

I'll try recompiling without those flags later today. I only have one workstation to test on and I'm always busy on it, unfortunately. Again, I appreciate your help.

CGarces, 2019-12-11:
Please test the last version and reopen the issue if needed

closed 2019-12-11 by CGarces