Mange/rtl8192eu-linux-driver #120: monitor mode not working after some time

respz, 2019-03-26:
I am trying to use airodump-ng. It works for a few minutes but then it stops capturing packages. If I unplug and plug the adapter, it s
```
# lsusb -s 1:3
Bus 001 Device 003: ID 2357:0108 TP-Link TL-WN822N Version 4 RTL8192EU
```
```
# uname -a
Linux kali 4.19.0-kali3-amd64 #1 SMP Debian 4.19.20-1kali1 (2019-02-14) x86_64 GNU/Linux
```

There are some messages from dmesg:
```
[  768.607908] RTL871X: Wrong rate section: (OFDM,2T)
[  768.607911] RTL871X: Wrong rate section: (OFDM,2T)
[  768.607913] RTL871X: Wrong rate section: (OFDM,2T)
[  768.607915] RTL871X: Wrong rate section: (OFDM,2T)
[  768.607917] RTL871X: Wrong rate section: (OFDM,2T)
[  768.607920] RTL871X: Wrong rate section: (OFDM,2T)
[  768.607922] RTL871X: Wrong rate section: (OFDM,2T)
[  768.607924] RTL871X: Wrong rate section: (OFDM,2T)
[  768.607926] RTL871X: Wrong rate section: (OFDM,2T)
[  768.607929] RTL871X: Wrong rate section: (OFDM,2T)
[  768.607931] RTL871X: Wrong rate section: (OFDM,2T)
[  768.607933] RTL871X: Wrong rate section: (OFDM,2T)
[  768.607935] RTL871X: Wrong rate section: (OFDM,2T)
[  768.607968] RTL871X: duplicate tx power limit combination [band 2.4G][regulation FCC][bw 20M][rate section HT][rf path 2T][chnl 01]
[  768.607973] RTL871X: duplicate tx power limit combination [band 2.4G][regulation ETSI][bw 20M][rate section HT][rf path 2T][chnl 01]
[  768.607976] RTL871X: duplicate tx power limit combination [band 2.4G][regulation MKK][bw 20M][rate section HT][rf path 2T][chnl 01]
[  768.607980] RTL871X: duplicate tx power limit combination [band 2.4G][regulation FCC][bw 20M][rate section HT][rf path 2T][chnl 02]
[  768.607984] RTL871X: duplicate tx power limit combination [band 2.4G][regulation ETSI][bw 20M][rate section HT][rf path 2T][chnl 02]
[  768.607987] RTL871X: duplicate tx power limit combination [band 2.4G][regulation MKK][bw 20M][rate section HT][rf path 2T][chnl 02]
[  768.607991] RTL871X: duplicate tx power limit combination [band 2.4G][regulation FCC][bw 20M][rate section HT][rf path 2T][chnl 03]
[  768.607995] RTL871X: duplicate tx power limit combination [band 2.4G][regulation ETSI][bw 20M][rate section HT][rf path 2T][chnl 03]
[  768.607998] RTL871X: duplicate tx power limit combination [band 2.4G][regulation MKK][bw 20M][rate section HT][rf path 2T][chnl 03]
[  768.608002] RTL871X: duplicate tx power limit combination [band 2.4G][regulation FCC][bw 20M][rate section HT][rf path 2T][chnl 04]
[  768.608005] RTL871X: duplicate tx power limit combination [band 2.4G][regulation ETSI][bw 20M][rate section HT][rf path 2T][chnl 04]
[  768.608009] RTL871X: duplicate tx power limit combination [band 2.4G][regulation MKK][bw 20M][rate section HT][rf path 2T][chnl 04]
[  768.608013] RTL871X: duplicate tx power limit combination [band 2.4G][regulation FCC][bw 20M][rate section HT][rf path 2T][chnl 05]

```

victorperin, 2019-03-26:
I have the same problem. Also when I run `aireplay-ng --test $MY_INTERFACE` it freezes my system.

CGarces, 2019-12-12:
Please try 
`sudo iw dev wlan0 set type monitor` with the last version of the driver.

Reopen the issue if needed

closed 2019-12-12 by CGarces

jcardenasy0810, 2020-03-05:
hi!
I can not make put the device on MONITOR using the driver, on fresh install on my machine, using  Kali Rolling. I've got this report:
`root@kali:/home/h4x0r# airmon-ng check kill

Killing these processes:

    PID Name
    722 wpa_supplicant

root@kali:/home/h4x0r# airmon-ng start wlan0

PHY     Interface       Driver          Chipset

phy0    wlan0           rtl8192eu       TP-Link 802.11n NIC

Newly created monitor mode interface wlan0mon is *NOT* in monitor mode.
Removing non-monitor wlan0mon interface...

WARNING: unable to start monitor mode, please run "airmon-ng check kill"
`
Please. Help Needed.
Thank you!
JCardenas

SatanicRaw, 2020-03-06:
Hey. This is not a driver problem, the problem is that you are connected to some kind of access point, so airmon-ng says that you need to kill wpa_supplicant. In general, on such issues you need to contact the Kali linux forum.

jcardenasy0810, 2020-03-11:
> Hey. This is not a driver problem, the problem is that you are connected to some kind of access point, so airmon-ng says that you need to kill wpa_supplicant. In general, on such issues you need to contact the Kali linux forum.
thanks!, SatanicRaw. The problem was solved. I was misunderstandig the procedure: I just need to do:
ifconfig wlan0 down
iwconfig wlan0 mode monitor

all fine, and thanks for the help!
Regards!