Mange/rtl8192eu-linux-driver #122: Keep throwing exception ERROR HalPwrSeqCmdParsing

yamng94, 2019-04-01:
When i try to connect to the wifi, following error emerged for executing command "dmesg".
Moreover, my linux kernal is 4.19.0-kali4-amd64
[image]
May i ask what is the root cause of the issue ? Thanks in advance

CGarces, 2019-11-09:
I have no idea about how fix this problem
¿Can you please test this branch?
https://github.com/CGarces/rtl8192eu-linux-driver/commits/realtek-5.x.x
It contains the driver v5.6.4_35685.20191108_COEX20171113-0047 from Realtek FTP plus all the patches from 4.4.x branch.

CGarces, 2019-11-20:
Version  v5.6.4 merged at #149 
Please test again an reopen the issue if needed

closed 2019-11-20 by CGarces