Mange/rtl8192eu-linux-driver #123: Building for kernel 5 and access_ok macro

w1ck3dg0ph3r, 2019-04-08:
Building for kernel 5 fails, apparently because kernel 5 dropped the type argument to access_ok():
https://lkml.org/lkml/2019/1/4/418

```
/var/lib/dkms/8192eu/54.6315416/build/os_dep/linux/rtw_android.c: In function ‘rtw_android_priv_cmd’:
/var/lib/dkms/8192eu/54.6315416/build/os_dep/linux/rtw_android.c:629:62: error: macro "access_ok" passed 3 arguments, but takes just 2
  if (!access_ok(VERIFY_READ, priv_cmd.buf, priv_cmd.total_len)){
                                                              ^
/var/lib/dkms/8192eu/54.6315416/build/os_dep/linux/rtw_android.c:629:7: error: ‘access_ok’ undeclared (first use in this function)
  if (!access_ok(VERIFY_READ, priv_cmd.buf, priv_cmd.total_len)){
       ^~~~~~~~~
/var/lib/dkms/8192eu/54.6315416/build/os_dep/linux/rtw_android.c:629:7: note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [scripts/Makefile.build:277: /var/lib/dkms/8192eu/54.6315416/build/os_dep/linux/rtw_android.o] Error 1
make[1]: *** [Makefile:1553: _module_/var/lib/dkms/8192eu/54.6315416/build] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.0.5-1-MANJARO/build'
make: *** [Makefile:1700: modules] Error 2
```

CGarces, 2019-04-08:
It's fixed on #110 
Are you ussing the branch `realtek-4.4.x`?
As you can see at https://github.com/Mange/rtl8192eu-linux-driver/pull/110/files#diff-5e6b610756277e382ebefac15e6996c3R632
the line at fails on your logs not apply on kernels < 5.0

w1ck3dg0ph3r, 2019-04-08:
Thanks! My bad, sorry, this branch works fine. I was just using package from AUR, which hadn't been updated.

closed 2019-04-08 by w1ck3dg0ph3r

CGarces, 2019-04-08:
I see 3º AUR packages
https://aur.archlinux.org/packages/8192eu-dkms/
https://aur.archlinux.org/packages/rtl8192eu/
https://aur.archlinux.org/packages/rtl8192eu-git/

I think that https://aur.archlinux.org/packages/rtl8192eu/ should work in theory the pkg is compiled when new kernels are released.