Mange/rtl8192eu-linux-driver #124: How to install in Jetson Nano (ARM A57)

jirolee, 2019-04-09:
Dears,
I have git clone this repo, but when I Make it, it will appear error. 
I found the problem may cause from 
MakeFile, but I don’t know which platform I should modify to y.
(I.e. for raspberry, CONFIG_PLATFORM_ARM_RPI = y)

CGarces, 2019-04-09:
Hi.

Please try
 `make V=1 ARCH=arm64`

jirolee, 2019-04-10:
Hi CGraces, 
It's work, thanks.

closed 2019-04-10 by jirolee

C-monC, 2019-11-24:
Hey jirolee,
what did you set as your platform in the makefile?

CGarces, 2019-11-24:
Hi @C-monC 

As described in the [readme.md](https://github.com/Mange/rtl8192eu-linux-driver/blob/realtek-4.4.x/README.md)

You can change the platform in the makefile. 

```
...
CONFIG_PLATFORM_I386_PC = n
...
CONFIG_PLATFORM_ARM_AARCH64 = y

```
Also you can pass the parameters at command line, or specify the compilations parameters by hand without a platform:

`make ARCH=arm64`

niyazFattahov, 2023-07-02:
Hello, could you please help me. I built for jetson nano this module, using command 'sudo modprobe -v 8723bu' I see in dmesg:
```
[   49.162786] 8723bu: loading out-of-tree module taints kernel.
[   49.197995] RTL871X: module init start
[   49.198000] RTL871X: rtl8723bu v4.3.6.11_12942.20141204_BTCOEX20140507-4E40
[   49.198003] RTL871X: rtl8723bu BT-Coex version = BTCOEX20140507-4E40
[   49.198117] usbcore: registered new interface driver rtl8723bu
[   49.198120] RTL871X: module init ret=0
```
But iwconfig shows nothing

```
dummy0    no wireless extensions.

l4tbr0    no wireless extensions.

eth0      no wireless extensions.

usb0      no wireless extensions.

lo        no wireless extensions.

docker0   no wireless extensions.

rndis0    no wireless extensions.
```
I followed this guide - https://github.com/lwfinger/rtl8723bu/blob/master/README.md and bulit '**master**' version

CGarces, 2023-07-02:
Try "ip" and "iw" instead of "ifconfig" and "iwconfig"