Mange/rtl8192eu-linux-driver #126: Fail at DKMS make

RAZORRZR0, 2019-05-27:
rzr@mac:~/rtl8192eu-linux-driver$ sudo dkms install rtl8192eu/1.0

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=5.0.0-15-generic........(bad exit status: 2)
ERROR (dkms apport): binary package for rtl8192eu: 1.0 not found
Error! Bad return status for module build on kernel: 5.0.0-15-generic (x86_64)
Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.

I have Ubuntu 19.04 and i am facing this issue while installing the driver

RAZORRZR0, 2019-05-27:
[make.log](https://github.com/Mange/rtl8192eu-linux-driver/files/3222385/make.log)
heres the make.log

CGarces, 2019-10-05:
@RAZORRZR0 have you solved the issue? can be closed?

closed 2020-02-09 by CGarces