Mange/rtl8192eu-linux-driver #129: 'Wrong rate section' after kernel update

sguk12, 2019-06-22:
I've been using you fabulous drivers for 6 months. From time to time I have to reinstall your drivers after a system update. I did a system update yesterday, saw no WIFI and executed the usual reinstall routine (full uninstall, than install again as per your read.me file). No WIFI still.

Because the dmesg snippet mentions FCC and ETSI, I'm guessing the new kernel (after the system update) confused your driver so it cannot find the correct power settings for my country (Australia)

serg@former-i:~> uname -a
Linux former-i 4.12.14-lp150.12.64-default #1 SMP Mon Jun 17 16:53:50 UTC 2019 (3edfd41) x86_64 x86_64 x86_64 GNU/Linux

dmesg snippet after the dongle insertion:
[   29.868410] usb 4-1.7: new high-speed USB device number 7 using ehci-pci
[   29.977511] usb 4-1.7: New USB device found, idVendor=2357, idProduct=0109
[   29.977514] usb 4-1.7: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   29.977516] usb 4-1.7: Product: 802.11n NIC 
[   29.977517] usb 4-1.7: Manufacturer: Realtek 
[   29.977519] usb 4-1.7: SerialNumber: 00e04c000001
[   30.007766] 8192eu: loading out-of-tree module taints kernel.
[   30.018780] RTL871X: module init start
[   30.018782] RTL871X: rtl8192eu v4.4.1_17696.20160509_BTCOEX20160412-0042
[   30.018782] RTL871X: build time: Jun 22 2019 07:09:36
[   30.018783] RTL871X: rtl8192eu BT-Coex version = BTCOEX20160412-0042
[   30.165381] RTL871X: hal_com_config_channel_plan chplan:0x22
[   30.169135] RTL871X: Wrong rate section: (CCK,2T)
[   30.169135] RTL871X: Wrong rate section: (CCK,2T)
[   30.169136] RTL871X: Wrong rate section: (CCK,2T)
[   30.169137] RTL871X: Wrong rate section: (CCK,2T)
[   30.169137] RTL871X: Wrong rate section: (CCK,2T)
[   30.169137] RTL871X: Wrong rate section: (CCK,2T)
[   30.169138] RTL871X: Wrong rate section: (CCK,2T)
[   30.169138] RTL871X: Wrong rate section: (CCK,2T)
[   30.169139] RTL871X: Wrong rate section: (CCK,2T)
[   30.169139] RTL871X: Wrong rate section: (CCK,2T)
----------8<---------------------8<---------------------------------
[   30.169173] RTL871X: Wrong rate section: (OFDM,2T)
[   30.169174] RTL871X: Wrong rate section: (OFDM,2T)
[   30.169174] RTL871X: Wrong rate section: (OFDM,2T)
[   30.169175] RTL871X: Wrong rate section: (OFDM,2T)
[   30.169175] RTL871X: Wrong rate section: (OFDM,2T)
[   30.169176] RTL871X: Wrong rate section: (OFDM,2T)
[   30.169176] RTL871X: Wrong rate section: (OFDM,2T)
[   30.169182] RTL871X: duplicate tx power limit combination [band 2.4G][regulation FCC][bw 20M][rate section HT][rf path 2T][chnl 01]
[   30.169183] RTL871X: duplicate tx power limit combination [band 2.4G][regulation ETSI][bw 20M][rate section HT][rf path 2T][chnl 01]
[   30.169184] RTL871X: duplicate tx power limit combination [band 2.4G][regulation MKK][bw 20M][rate section HT][rf path 2T][chnl 01]
[   30.169185] RTL871X: duplicate tx power limit combination [band 2.4G][regulation FCC][bw 20M][rate section HT][rf path 2T][chnl 02]
[   30.169186] RTL871X: duplicate tx power limit combination [band 2.4G][regulation ETSI][bw 20M][rate section HT][rf path 2T][chnl 02]
[   30.169186] RTL871X: duplicate tx power limit combination [band 2.4G][regulation MKK][bw 20M][rate section HT][rf path 2T][chnl 02]
[   30.169187] RTL871X: duplicate tx power limit combination [band 2.4G][regulation FCC][bw 20M][rate section HT][rf path 2T][chnl 03]
[   30.169188] RTL871X: duplicate tx power limit combination [band 2.4G][regulation ETSI][bw 20M][rate section HT][rf path 2T][chnl 03]
[   30.169189] RTL871X: duplicate tx power limit combination [band 2.4G][regulation MKK][bw 20M][rate section HT][rf path 2T][chnl 03]
[   30.169189] RTL871X: duplicate tx power limit combination [band 2.4G][regulation FCC][bw 20M][rate section HT][rf path 2T][chnl 04]
[   30.169190] RTL871X: duplicate tx power limit combination [band 2.4G][regulation ETSI][bw 20M][rate section HT][rf path 2T][chnl 04]
[   30.169191] RTL871X: duplicate tx power limit combination [band 2.4G][regulation MKK][bw 20M][rate section HT][rf path 2T][chnl 04]
[   30.169192] RTL871X: duplicate tx power limit combination [band 2.4G][regulation FCC][bw 20M][rate section HT][rf path 2T][chnl 05]
[   30.169193] RTL871X: duplicate tx power limit combination [band 2.4G][regulation ETSI][bw 20M][rate section HT][rf path 2T][chnl 05]
[   30.169193] RTL871X: duplicate tx power limit combination [band 2.4G][regulation MKK][bw 20M][rate section HT][rf path 2T][chnl 05]
[   30.169194] RTL871X: duplicate tx power limit combination [band 2.4G][regulation FCC][bw 20M][rate section HT][rf path 2T][chnl 06]
[   30.169195] RTL871X: duplicate tx power limit combination [band 2.4G][regulation ETSI][bw 20M][rate section HT][rf path 2T][chnl 06]
[   30.169196] RTL871X: duplicate tx power limit combination [band 2.4G][regulation MKK][bw 20M][rate section HT][rf path 2T][chnl 06]
[   30.169197] RTL871X: duplicate tx power limit combination [band 2.4G][regulation FCC][bw 20M][rate section HT][rf path 2T][chnl 07]
---------------------8<------------------------------8<----------------------------
[   30.169236] RTL871X: duplicate tx power limit combination [band 2.4G][regulation MKK][bw 40M][rate section HT][rf path 2T][chnl 10]
[   30.169237] RTL871X: duplicate tx power limit combination [band 2.4G][regulation FCC][bw 40M][rate section HT][rf path 2T][chnl 11]
[   30.169237] RTL871X: duplicate tx power limit combination [band 2.4G][regulation ETSI][bw 40M][rate section HT][rf path 2T][chnl 11]
[   30.169238] RTL871X: duplicate tx power limit combination [band 2.4G][regulation MKK][bw 40M][rate section HT][rf path 2T][chnl 11]
[   30.169377] RTL871X: rtw_ndev_init(wlan0) if1 mac_addr=50:3e:aa:ba:20:96
[   30.169561] usbcore: registered new interface driver rtl8192eu
[   30.169562] RTL871X: module init ret=0
[   30.174403] rtl8192eu 4-1.7:1.0 wlan1: renamed from wlan0
[   30.201461] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[   31.887263] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[   31.891167] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[   31.898761] NET: Registered protocol family 17
[   31.936927] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[   33.328413] RTL871X: nolinked power save enter
[   36.688974] RTL871X: nolinked power save leave
[   38.080465] RTL871X: nolinked power save enter

By the way,  #116 looks similar to my issue

marcovc, 2019-07-10:
I have this problem as well. Have you found a solution yet?

TheJags, 2019-08-05:
Well, I too am getting these errors, after I upgraded the kernel to 5.2.6-050206-generic. Though WiFi works ok so far.

Aug 06 04:36:44 um kernel: 8192eu: loading out-of-tree module taints kernel.
Aug 06 04:36:44 um kernel: 8192eu: module verification failed: signature and/or required key missing - tainting kernel
Aug 06 04:36:44 um kernel: RTL871X: module init start
Aug 06 04:36:44 um kernel: RTL871X: rtl8192eu v4.4.1_17696.20160509_BTCOEX20160412-0042
Aug 06 04:36:44 um kernel: RTL871X: build time: Aug  5 2019 23:22:17
Aug 06 04:36:44 um kernel: RTL871X: rtl8192eu BT-Coex version = BTCOEX20160412-0042
Aug 06 04:36:44 um kernel: i5500_temp 0000:00:14.3: Sensor seems to be disabled
Aug 06 04:36:44 um kernel: EDAC i7core: Device not found: dev 00.0 PCI ID 8086:2c70
Aug 06 04:36:44 um kernel: RTL871X: hal_com_config_channel_plan chplan:0x21
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (CCK,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (OFDM,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (OFDM,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (OFDM,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (OFDM,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (OFDM,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (OFDM,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (OFDM,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (OFDM,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (OFDM,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (OFDM,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (OFDM,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (OFDM,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (OFDM,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (OFDM,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (OFDM,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (OFDM,2T)
Aug 06 04:36:44 um kernel: RTL871X: Wrong rate section: (OFDM,2T)
[... 69 lines cut ...]
Aug 06 04:36:44 um kernel: RTL871X: duplicate tx power limit combination [band 2.4G][regulation ETSI][bw 40M][rate section HT][rf path 2T][chnl 05]
Aug 06 04:36:44 um kernel: RTL871X: duplicate tx power limit combination [band 2.4G][regulation MKK][bw 40M][rate section HT][rf path 2T][chnl 05]
Aug 06 04:36:44 um kernel: RTL871X: duplicate tx power limit combination [band 2.4G][regulation FCC][bw 40M][rate section HT][rf path 2T][chnl 06]
Aug 06 04:36:44 um kernel: RTL871X: duplicate tx power limit combination [band 2.4G][regulation ETSI][bw 40M][rate section HT][rf path 2T][chnl 06]
Aug 06 04:36:44 um kernel: RTL871X: duplicate tx power limit combination [band 2.4G][regulation MKK][bw 40M][rate section HT][rf path 2T][chnl 06]
[... cut ...]
Aug 06 04:36:44 um kernel: RTL871X: duplicate tx power limit combination [band 2.4G][regulation ETSI][bw 40M][rate section HT][rf path 2T][chnl 11]
Aug 06 04:36:44 um kernel: RTL871X: duplicate tx power limit combination [band 2.4G][regulation MKK][bw 40M][rate section HT][rf path 2T][chnl 11]
Aug 06 04:36:44 um kernel: RTL871X: rtw_ndev_init(wlan0) if1 mac_addr=98:de:d0:0b:03:a5
Aug 06 04:36:44 um kernel: usbcore: registered new interface driver rtl8192eu
Aug 06 04:36:44 um kernel: RTL871X: module init ret=0

- Thanks.

daedroza, 2019-08-20:
Well, do you've a drop in your speeds?
My download speeds are now 7 times less. I am trying to investigate the issue, could you please give output of the following -
1. lsmod
2. sudo lshw -c network

sguk12, 2019-08-21:
After a few kernel updates and re-installation of the driver, my WIFI started working again.
dmesg still shows the above errors, which now seems to be a red herring.
This is my kernel if anyone is interested:
serg@former-i:~> uname -a
Linux former-i 4.12.14-lp150.12.70-default #1 SMP Wed Aug 7 07:03:51 UTC 2019 (3643f11) x86_64 x86_64 x86_64 GNU/Linux

closed 2019-08-21 by sguk12