Mange/rtl8192eu-linux-driver #131: "No rule to make target modules". Stop

kasparov1997, 2019-07-04:
I have TPLink -WN823N V2 and OS - ARCH LINUX 
Kernel version - 4.19.34-1-lts .
I have install dkms - 2.7.1-1 , linux-headers - 5.1.15.arch-1

My WiFi is not able to connect and led is also not blinking .
When I try to manually build the driver it says
"Entering directory '/usr/lib/modules/4.19.34-1-lts/build"
***No rule to make target 'modules'. Stop .
Leaving directory '/usr/lib/modules/4.19.34-1-lts/build
If type command "dkms install rtl8192eu/1.0"
Error ! echo
Your kernel headers for kernel 4.19.34-1-lts cannot be found at /usr/lib/modules/4.19.34-1-lts/build or /usr/lib/modules/4.19.34-1-lts/source
[image]

CGarces, 2019-07-04:
You must download the headers of your Kernel.

From [README.md](https://github.com/Mange/rtl8192eu-linux-driver/blob/realtek-4.4.x/README.md)

> 
> Install DKMS and other required tools
> 
> for normal Linux systems
> $ sudo apt-get install git linux-headers-generic build-essential dkms;

kasparov1997, 2019-07-04:
First of all I use pacman, second thing is I have installed linux header 5.1.15 .
Is linux-header and linux-headers-generic build-esential different package??

CGarces, 2019-07-04:
it's trying to compile over 4.19.34

Can you check your current kernel version with `uname -r`?

closed 2019-09-05 by CGarces