Mange/rtl8192eu-linux-driver #135: AP mode not working Raspberry Pi / TP-Link TL-WN821N v6

AlbertJP, 2019-07-27:
I am trying to use AP mode on the TL-WN821N v6 with a Raspberry Pi 3 model B+. (USB ID 2357:0107)

In the past, this worked (kernel 4.9 on a 3 model B) but I needed to upgrade my kernel for the new Pi model B+, and now I get an error in the kernel log when trying to use AP mode:

    ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported

I have tried both 4.14.98 and 4.19.60 kernels from https://github.com/Hexxeh/rpi-firmware. (The kernel 4.9 which I used on the model B also came from there.)

Other modes than AP seem to work, I can at least see the nearby wifi networks.

yagoplx, 2019-11-19:
Make sure that hostapd is configured to use nl80211 driver.

Then, run:
`sudo modprobe -r r8xxxu;
sudo modprobe -r rtl8xxxu;
sudo modprobe -r 8192eu; 
sleep 3;
sudo modprobe 8192eu
`
After that try running hostapd again. It works for me, but in my case I prefer using create_ap to handle hostapd.

henriquefchaves, 2021-03-18:
Sorry, im a beat lost, can you tell me where i find how to configure hostapd to use nl80211?
and if you using this equipment on RPi need do have usb power hub?